svpcom/wfb-ng #4: some times, fec recovery wrong packet

openserli, 2017-05-16:
hi, some times, after do fec, 
the recoveryed packet is wrong, and the log  : fprintf(stderr, "corrupted packet %u\n", seq);
is dumpt.

void LocalAggregator::send_packet(int ring_idx, int fragment_idx)
{
    wpacket_hdr_t* packet_hdr = (wpacket_hdr_t*)(rx_ring[ring_idx].fragments[fragment_idx]);
    uint8_t *payload = (rx_ring[ring_idx].fragments[fragment_idx]) + sizeof(wpacket_hdr_t);

    if (packet_hdr->seq > seq + 1)
    {
        fprintf(stderr, "%u packets lost\n", packet_hdr->seq - seq - 1);
    }

    seq = packet_hdr->seq;

    if(packet_hdr->packet_size > MAX_PAYLOAD_SIZE)
    {
        fprintf(stderr, "corrupted packet %u\n", seq);
    }else{
        send(sockfd, payload, packet_hdr->packet_size, 0);
    }
}

svpcom, 2017-05-16:
You got corrupted packet from HW. Its size lager than expected

svpcom, 2017-05-16:
What version do you use? Last version use authenticated encryption and corrupted packets should not pass to FEC layer.

closed 2017-10-12 by svpcom