svpcom/wfb-ng #5: Why is there a need for the "TX rate lock" patch ?

Buanderie, 2017-06-06:
I understand that in AP mode, the minstrel_ht (or the Atheros one) rate selection algorithm will change the rate/modulation dynamically based on the STA RSSI. But in monitor/injection mode, why should the rate change dynamically ? Shouldn't calling "iw dev wlan0 set bitrates... ht-mcs-5 ..." be enough to set the bitrates and MCS ? Also I have some difficulties finding out what the patch is really doing.

svpcom, 2017-06-06:
There is no "tx rate lock patch".
There are patches:
1. tx power lock.
2. tx rate for injected packets
3. tx timeslot patch.

[1] is needed to fix tx power at max supported level and disable CRDA restrictions
[2] enables API for specify bitrate when injecting packets. But for current ieee80211 stack is not sufficient to specify rate in injected header and you also need to set it via ``iw dev $WLAN set bitrates ht-mcs-5 1 sgi-5`` before tx start
[3] removes most of tx timeslot guard intervals to maximize throughput. We use only one-directional broadcast and tx wouldn't listen for ACK's.  https://en.wikipedia.org/wiki/Extended_interframe_space

ez-wifibroadcast-1.4-kernel-4.4-patches.diff implements [1] and [3]
mac80211-radiotap-bitrate_mcs_rtscts.linux-4.4.patch implements [2]

Also there are AR9271 patches, but I've never try it because I don't have any atheros hardware. My HW setup is RT3572 (rt2800 driver) in 5GHz band.

Buanderie, 2017-06-07:
Thank you for the information...
Well, concerning [1], I have some additional questions. Isn't there a risk that the card will try to output at power exceeding its nominal value (btw, have you ever seen a case where it caused HW damage ?).
I also don't quite get this part: 
`
// CHANGES: set 58 for every data rate
for (i = 0; i < Ar5416RateSize; i++) {
    // ratesArray[i] -= AR5416_PWR_TABLE_OFFSET_DB * 2;
    ratesArray[i] = 58;
}
`
What does setting this to 58 do ? Also, why 58 ?

Regarding [3], I can't really use that since I need to be able to switch back and from AP mode.
I couldn't get [2] to work, yet but it's not really a priority :)

svpcom, 2017-06-07:
I've didn't test any ARxxxx patches due to lack of hardware. They was taken from https://github.com/bortek/EZ-WifiBroadcast as is. In case of TX powes limit - card select minumum of its supported power (stored in EEPROM) and CRDA power limit, so you shouldn't got HW damage.

 If you have recent mainline kernel (4.x) you don't need [2] (it already here). [1] and [3] are optional and needed only to increase transmit range and channel bandwidth.

svpcom, 2017-07-05:
Moved to wiki

closed 2017-07-05 by svpcom