svpcom/wfb-ng #6: failed to build

terial, 2017-07-04:
following the instructions on https://dev.px4.io/en/qgc/video_streaming_wifi_broadcast.html i cant seem to get the build successful. 

I got an error about sodium.h so i installed libsodium-dev but when trying to make this is the output

cc -c -o fec.o fec.c -Wall -g
fec.c:175:31: error: expected ‘;’, ‘,’ or ‘)’ before ‘dst’
 _addmul1(register gf*restrict dst, const register gf*restrict src, gf c, size_t sz) {
                               ^
fec.c: In function ‘_invert_mat’:
fec.c:250:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for (size_t i = 0; i < k; i++)
     ^
fec.c:250:5: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
fec.c:253:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for (size_t col = 0; col < k; col++) {
     ^
fec.c:264:9: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
         for (size_t row = 0; row < k; row++) {
         ^
fec.c:266:17: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
                 for (size_t ix = 0; ix < k; ix++) {
                 ^
fec.c:286:13: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
             for (size_t ix = 0; ix < k; ix++)
             ^
fec.c:300:13: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
             for (size_t ix = 0; ix < k; ix++)
             ^
fec.c:313:13: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
             for (size_t ix = 0; ix < k; ix++, p += k) {
             ^
fec.c:317:21: warning: implicit declaration of function ‘_addmul1’ [-Wimplicit-function-declaration]
                     addmul (p, pivot_row, c, k);
                     ^
fec.c:323:17: error: redefinition of ‘col’
     for (size_t col = k; col > 0; col--)
                 ^
fec.c:253:17: note: previous definition of ‘col’ was here
     for (size_t col = 0; col < k; col++) {
                 ^
fec.c:323:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for (size_t col = k; col > 0; col--)
     ^
fec.c:325:13: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
             for (size_t row = 0; row < k; row++)
             ^
fec.c: At top level:
fec.c:499:53: error: expected ‘;’, ‘,’ or ‘)’ before ‘const’
 build_decode_matrix_into_space(const fec_t*restrict const code, const unsigned*const restrict index, const unsigned k, gf*restrict const matrix) {
                                                     ^
fec.c: In function ‘fec_decode’:
fec.c:519:5: warning: implicit declaration of function ‘build_decode_matrix_into_space’ [-Wimplicit-function-declaration]
     build_decode_matrix_into_space(code, index, code->k, m_dec);
     ^
fec.c: At top level:
fec.c:236:1: warning: ‘_invert_mat’ defined but not used [-Wunused-function]
 _invert_mat(gf* src, size_t k) {
 ^
Makefile:8: recipe for target 'fec.o' failed
make: *** [fec.o] Error 1

svpcom, 2017-07-04:
What toolchain (gcc version, arm or x86) do you use?

terial, 2017-07-04:
trying to compile on a raspberry pi 3

pi@raspberrypi:~ $ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 4.9.2-10' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.9.2 (Raspbian 4.9.2-10)

svpcom, 2017-07-05:
@terial Fixed

closed 2017-07-05 by svpcom

svpcom, 2017-07-05:
@terial 
Article in PX4 wiki is outdated (I'll fix it later). See https://github.com/svpcom/wifibroadcast/wiki for details

commit 89fff12e (2017-07-05): Fix #6