svpcom/wfb-ng #7: Ralink RT3572 support?

terial, 2017-07-06:
Im using two AWUS052NH v2 usb adapters but cant seem to get them to connect. I got the the tx, rx and keygen from the build ok and created the tx.key and rx.key and copied both to my desktop running ubuntu 16.04 and a RPI3. I have tried running tx and rx on both but never seem to get a connection. When they are connected with no video streaming should i still see text being passed from the command line?

iwconfig outputs this on both devices:
wlan1     IEEE 802.11abgn  Mode:Monitor  Frequency:5.745 GHz  Tx-Power=30 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off

svpcom, 2017-07-06:
1. Run ``scripts/tx_standalone.sh wlan1`` on TX host
2. Run ``scripts/rx_standalone.sh wlan1`` on RX host
3. Run ``nc -u 5600`` on RX host
4. Run ``echo test | nc -u localhost 5600`` on TX host

If you got ``test`` message - debug video pipeline.
If you don't see ``test`` -- run tcpdump on TX ``tcpdump -i wlan1 -n -p`` and repeat test. If card successfully tx - repeat tcpdump test on RX host.

terial, 2017-07-07:
Hey @svpcom , still struggling but thank you for all of your help so far. Tried using both the desktop (ubuntu 16.04) and RPI3 (latest raspbian lite) as tx/rx but no luck seeing the test message.

on the RPI3 tcpdump -i wlan1 -n -p output:
admin@fca1:~/wifibroadcast $ sudo tcpdump -i wlan1 -n -p
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlan1, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
19:15:07.832459 6.0 Mb/s 5745 MHz 11a -69dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:15:08.242061 6.0 Mb/s 5745 MHz 11a -69dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:15:08.344452 6.0 Mb/s 5745 MHz 11a -69dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:15:08.549239 6.0 Mb/s 5745 MHz 11a -69dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:15:08.651642 6.0 Mb/s 5745 MHz 11a -69dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:15:09.061251 6.0 Mb/s 5745 MHz 11a -69dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:15:09.675602 6.0 Mb/s 5745 MHz 11a -69dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:15:09.982802 6.0 Mb/s 5745 MHz 11a -69dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
^X19:15:11.211618 6.0 Mb/s 5745 MHz 11a -69dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:15:11.314007 6.0 Mb/s 5745 MHz 11a -69dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:15:11.416388 6.0 Mb/s 5745 MHz 11a -67dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:15:11.723598 6.0 Mb/s 5745 MHz 11a -67dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:15:12.338027 6.0 Mb/s 5745 MHz 11a -69dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:15:12.440419 6.0 Mb/s 5745 MHz 11a -69dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
^C
14 packets captured
14 packets received by filter
0 packets dropped by kernel

On the desktop tcpdump -i wlx00c0ca91ef5a -n -p output:
nkariger@nkariger-ubuntu:~/Desktop/wifibroadcast$ sudo tcpdump -i wlx00c0ca91ef5a -n -p
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlx00c0ca91ef5a, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
19:18:21.373837 12.0 Mb/s 5745 MHz 11a -65dBm signal antenna 1 Probe Request () [6.0 9.0 12.0 18.0 24.0 36.0 48.0 54.0 Mbit]
19:18:39.631842 6.0 Mb/s 5745 MHz 11a -31dBm signal antenna 1 Probe Request () [6.0 9.0 12.0 18.0 24.0 36.0 48.0 54.0 Mbit]
19:18:39.642013 6.0 Mb/s 5745 MHz 11a -29dBm signal antenna 1 Probe Request () [6.0 9.0 12.0 18.0 24.0 36.0 48.0 54.0 Mbit]
19:18:41.188320 6.0 Mb/s 5745 MHz 11a -67dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:18:42.929612 6.0 Mb/s 5745 MHz 11a -65dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:18:46.719312 6.0 Mb/s 5745 MHz 11a -69dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
19:18:49.792184 6.0 Mb/s 5745 MHz 11a -67dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
^C
7 packets captured
7 packets received by filter
0 packets dropped by kernel

I noticed at one point that i saw the work "SONOS" appear on the command line which would be the wireless speaker on our network at home. I figured i would try to change the channel.... tried 148, 150, 155 but on both computers either as tx or rx this is the error i got

Setting wlan1 to channel 155
command failed: Invalid argument (-22)

i admit, advanced networking like this is beyond my skills so i am not sure where to go from here. I did notice though that the RPI tcpdump i always see each line with "6.0 Mb/s" while on the desktop i see "6.0 Mb/s and 12.0Mb/s" do you think there is something not correct with the ubuntu drivers for the RT3572 chipset? I also have no patched the kernel/firmware on either computers since I am trying to keep things simple to start as well as never having done that process. Thanks again for your help and hard work!

svpcom, 2017-07-07:
It is very strange that wifibroadcast doesn't inject any packets.
What kernel version do you use?
Try following steps:
1. Disable NetworkManager (systemctl stop NetworkManager)
2. open 3 consoles on your PC:
2.1 in first console run: ``$ sudo ./scripts/tx_standalone.sh wlan1`` where wlan1 is name of your wifi card
2.2 in second run: ``$ sudo tcpdump -i wlan1 -nn -p``
2.3 in third run: ``echo test | nc -u localhost 5600``

check for any error messages on the first console
on second console you should got something like:
```
listening on wlan1, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
14:54:16.209408 [bit 15] 13:22:33:44:55:01 Unknown SSAP 0xb4 > 13:22:33:44:55:01 802.1B I Supervisory, Reject, rcv seq 37, Flags [Poll], length 73
14:54:16.209449 [bit 15] 13:22:33:44:55:01 > 13:22:33:44:55:01 Null Information, send seq 0, rcv seq 0, Flags [Command], length 36
	0x0000:  0100 0000 0000 0000 0043 bc5e c377 8bf2  .........C.^.w..
	0x0010:  4c86 4e35 a4f6 21b6 2ee5 ee51 919d a5ee  L.N5..!....Q....
	0x0020:  8aef 8651                                ...Q
14:54:16.211518 [bit 15] 13:22:33:44:55:01 Unknown SSAP 0xb4 > 13:22:33:44:55:01 802.1B I Supervisory, Reject, rcv seq 37, Flags [Poll], length 73
14:54:16.211547 [bit 15] 13:22:33:44:55:01 > 13:22:33:44:55:01 Null Information, send seq 0, rcv seq 0, Flags [Command], length 36
	0x0000:  0100 0000 0000 0000 0043 bc5e c377 8bf2  .........C.^.w..
	0x0010:  4c86 4e35 a4f6 21b6 2ee5 ee51 919d a5ee  L.N5..!....Q....
	0x0020:  8aef 8651                                ...Q
```

terial, 2017-07-08:
ok, looks like NetworkManager on my desktop w/ Ubuntu might be causing part of the issue... console 2 with the tcpdump was just outputting the same messages as in my previous post until i sent the test message echo test | nc -u localhost 5600 on the third console which i output a quick stream of different messages:

18:51:41.691671 [bit 15] 13:22:33:44:55:01 Unknown SSAP 0x46 > 13:22:33:44:55:01 802.1B I Information, send seq 54, rcv seq 61, Flags [Final], length 73
	0x0000:  0247 6c7b a989 e1f2 8215 30cf df15 f386  .Gl{......0.....
	0x0010:  4ef1 41e0 d331 bb8f 0bd2 9ba6 9a08 8e52  N.A..1.........R
	0x0020:  1147 64c1 ed73 d965 66ef be08 2952 8474  .Gd..s.ef...)R.t
	0x0030:  1fac 3d03 ea75 3570 fdb5 e0ac 8f93 93fd  ..=..u5p........
	0x0040:  2b66 4701 d25f a4ef 5c                   +fG.._..\
18:51:41.691720 [bit 15] 13:22:33:44:55:01 802.1B I > 13:22:33:44:55:01 Null Information, send seq 0, rcv seq 0, Flags [Response], length 36
	0x0000:  0103 0000 0000 0000 00d4 256f c450 e1f3  ..........%o.P..
	0x0010:  63d2 318d 740a 1801 2826 1723 e5cc 6931  c.1.t...(&.#..i1
	0x0020:  6dd5 5cb2                                m.\.
18:51:41.693817 [bit 15] 13:22:33:44:55:01 Unknown SSAP 0x46 > 13:22:33:44:55:01 802.1B I Information, send seq 54, rcv seq 61, Flags [Final], length 73
	0x0000:  0247 6c7b a989 e1f2 8215 30cf df15 f386  .Gl{......0.....
	0x0010:  4ef1 41e0 d331 bb8f 0bd2 9ba6 9a08 8e52  N.A..1.........R
	0x0020:  1147 64c1 ed73 d965 66ef be08 2952 8474  .Gd..s.ef...)R.t
	0x0030:  1fac 3d03 ea75 3570 fdb5 e0ac 8f93 93fd  ..=..u5p........
	0x0040:  2b66 4701 d25f a4ef 5c                   +fG.._..\
18:51:41.693862 [bit 15] 13:22:33:44:55:01 802.1B I > 13:22:33:44:55:01 Null Information, send seq 0, rcv seq 0, Flags [Response], length 36
	0x0000:  0103 0000 0000 0000 00d4 256f c450 e1f3  ..........%o.P..
	0x0010:  63d2 318d 740a 1801 2826 1723 e5cc 6931  c.1.t...(&.#..i1
	0x0020:  6dd5 5cb2                                m.\.
18:51:41.788197 6.0 Mb/s 5745 MHz 11a -65dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
 
I tried to do the same on the RPI3 but ended up being pretty difficult for me since i needed to use 3 terminals so i ended up learning how to use screen. I started tx_standalong.sh in one screen called wifi,
started sudo tcpdump -i wlan1 -nn -p in a screen called tcp  and had it logged and while both were detached ran echo test | nc -u localhost 5600 a few times and paused tcpdump and checked the screen and both the logs, never saw anything like what you had above or when running it on my desktop.

uname -a on RPI3 is,
Linux fca1 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux

svpcom, 2017-07-17:
Ok, now need to ensure that TX side send anything to air:

On Raspberry PI:
1. Disable NetworkManager (systemctl stop NetworkManager)
2. open 2 consoles:
  2.1 in first run: ``sudo ./scripts/rx_standalone.sh wlan1`` where wlan1 is name of your wifi
  2.2 in second run: ``sudo tcpdump  -i wlan1 -nn -p``

On PC:
3. Disable NetworkManager (systemctl stop NetworkManager)
4. open 2 consoles:
    4.1 in first console run: ``sudo ./scripts/tx_standalone.sh wlan1`` where wlan1 is name of your wifi card
    4.2 in second run: ``echo test | nc -u localhost 5600``

You should got some packets on tcpdump console

svpcom, 2017-07-17:
You could use ``screen``, ``tmux`` or just several ssh sessions

terial, 2017-07-19:
Gave it another try, On the RPI3 as rx output of sudo tcpdump -i wlan1 -nn -p:

05:52:00.928941 6.0 Mb/s 5745 MHz 11a -65dBm signal antenna 1 Beacon () ESS, PRIVACY
05:52:01.031342 6.0 Mb/s 5745 MHz 11a -67dBm signal antenna 1 Beacon () ESS, PRIVACY
05:52:01.133750 6.0 Mb/s 5745 MHz 11a -63dBm signal antenna 1 Beacon () ESS, PRIVACY
05:52:01.168337 6.0 Mb/s 5745 MHz 11a -67dBm signal antenna 1 Beacon (TG1672GF2-5G) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 149, PRIVACY
05:52:01.236145 6.0 Mb/s 5745 MHz 11a -67dBm signal antenna 1 Beacon () ESS, PRIVACY
05:52:01.338548 6.0 Mb/s 5745 MHz 11a -65dBm signal antenna 1 Beacon () ESS, PRIVACY
05:52:01.543354 6.0 Mb/s 5745 MHz 11a -67dBm signal antenna 1 Beacon () ESS, PRIVACY
05:52:01.748158 6.0 Mb/s 5745 MHz 11a -65dBm signal antenna 1 Beacon () ESS, PRIVACY
05:52:01.850557 6.0 Mb/s 5745 MHz 11a -67dBm signal antenna 1 Beacon () ESS, PRIVACY
05:52:01.952966 6.0 Mb/s 5745 MHz 11a -63dBm signal antenna 1 Beacon () ESS, PRIVACY

and on the desktop running echo test | nc -u localhost 5600 the output on the RPI3 became:

05:52:03.968947 5745 MHz 11n -17dBm signal antenna 1 30.0 Mb/s MCS 1 40 MHz short GI 13:22:33:44:55:01 Unknown SSAP 0x3c > 13:22:33:44:55:01 802.1B I Supervisory, Receiver Ready, rcv seq 73, Flags [Command], length 73
05:52:03.969454 5745 MHz 11n -17dBm signal antenna 1 30.0 Mb/s MCS 1 40 MHz short GI 13:22:33:44:55:01 802.1B I > 13:22:33:44:55:01 Null Information, send seq 0, rcv seq 0, Flags [Command], length 36
	0x0000:  0102 0000 0000 0000 007b aaf3 8258 23de  .........{...X#.
	0x0010:  d3fe 002d 3621 408e 285b cff1 99eb ae02  ...-6!@.([......
	0x0020:  c57b bac9                                .{..

I tried "echo test | nc -u localhost 5600" a few times on the desktop and each time the RPI3 ssh screen changed so I am assuming it is transmitting and receiving ok.

svpcom, 2017-07-19:
Ok, now instead of ``tcpdump`` run ``nc -u -l 5600`` on RPI3. You should get ``test`` message

terial, 2017-07-20:
ok, I got the test message on the RPI3 when it was running as rx. Still can not get anything when I try the RPI3 as tx though. Do you think this is because of a power issue?

svpcom, 2017-07-20:
What is kernel version on RPI3?

terial, 2017-07-21:
Its just regular raspbian jessie lite so it should be kernel 4.9, 2017-07-05

terial, 2017-08-06:
Hey @svpcom , any ideas about the issue i am having? Thanks

svpcom, 2017-08-06:
On which step transmit failed if transmitter is RPI3?
Does local (on RPI3) tcpdump show anything?

OneStromberg, 2017-08-06:
Hi guys! Let me join to this conversation.
@svpcom  I have same issue
**command failed: Invalid argument (-22)**
 with
`sudo iw dev wlan0 set bitrates ht-mcs-2.3 1 sgi-5`

As I find in iw --help:

`Usage:	iw [options] dev <devname> set bitrates [legacy-<2.4|5> <legacy rate in Mbps>*] [ht-mcs-<2.4|5> <MCS index>*] [vht-mcs-<2.4|5> <NSS:MCSx,MCSy... | NSS:MCSx-MCSy>*] [sgi-2.4|lgi-2.4] [sgi-5|lgi-5]`

Here _ht-mcs-<2.4|5>_ it says that we cant set **5** sharp.
Probably I'm using different version of kernel
**Linux 4.4.0-89-generic #112-Ubuntu SMP Mon Jul 31 19:38:41 UTC 2017 x86_64**

svpcom, 2017-08-07:
Does your wifi hardware 5Ghz capable?

OneStromberg, 2017-08-08:
@svpcom I'm using **D-Link DWA-137**. It has _Ralink RT5372_ chipset, so, according the [list of supported devices](https://github.com/bortek/EZ-WifiBroadcast/wiki/List-of-Wifi-cards-and-doungles) it should work.

Answering on your questions I've got that It doesn't 5Ghz capable.

I'm trying to establish wifibroadcasting using Ubuntu 16.04, with **TP-Link TL-WN822N** and _raspberry pi3_ with **D-Link DWA-137** on board.

I've made everything via [instructions](https://dev.px4.io/en/qgc/video_streaming_wifi_broadcast.html) but have some issues. To test my environment I've ran your [advice](https://github.com/svpcom/wifibroadcast/issues/7#issuecomment-313427280) and have had **-22** error.

I think that this is right conversation for mine question.

Can you advice how can I change _scripts_ from `scripts` to run test?

After running `./rx` on mine _Ubuntu_ with  I see test line with message in _terminal_ **TP-Link TL-WN822N**, but when I'm running `./tx` on _Raspberry Pi3_ with I don't see any messages, just silence in _terminal_.

I've not patched my kernel with patches from `patches` _directory_. I hope this is not the reason of this issue.

svpcom, 2017-08-08:
If you want to use wifibroadcast in 2.4GHz band you need to select "2G" band in tx and rx scripts

OneStromberg, 2017-08-11:
@svpcom thanx for your support! :+1: 
I pulled from repo, selected mine band.

Now I have error on *rx* side on
`command failed: Input/output error (-5)`

I tried all lines from _rx_standalone.sh_ and get that error is from:
`sudo iw dev $WLAN set bitrates ht-mcs-2.4 1 sgi-2.4`
As I understand that is because of mine *TP-Link TL-WN822N*

Ignoring the error I run scripts on both side and tried to send message with
`echo test | nc -u localhost 5600`
But I hadn't got it on *tx* side(TP-Link TL-WN822N).

Then I decided to use raspberry with D-link(DWA-137) as a *transmitter* and PC with TP-Link as *receiver* and it worked.

Next I decided that my TP-Link doesn't work as receiver and switch it with D-Link.
And it hadn't any results, I still don't see massages from raspberry.

Now I can say that issue is: "I can't see messages that comes from my Raspberry to PC, but can see them to the other side -> from Raspberry to PC".

svpcom, 2017-08-11:
Only some cards can be TX (supports packets injection), but most of the cards (except few bad chipsets) supports monitor mode (i.e. can receive arbitrary packets).

First of all let's localize the problem.

1. Can you connect both network adapters to PC with linux?
2. It will be wlan1 and wlan2 (or wlan0 and wlan1 if you don't have builtin wifi).
3. WN822N has Atheros AR7010 or AR9287 (according to wikidevi.com) it is 2.4GHz only and I've never tried it as TX (injection mode). Lets it will be RX (receiver).
4.  D-link DWA-137 is a Ralink RT5372 - same 2.4GHz. Lets use it in injection (TX) mode.
5. Edit rx_standalone and add ``-u 5601`` to rx arguments.
6. Run rx_standalone and tx_standalone scripts with appropriate wlan cards.
If there are some error messages, rerun scripts with ``bash -x`` to localize which command fails.
7. If previous steps were succeed run ``nc -u -l 5601`` and ``echo test | nc -u localhost 5600``

terial, 2017-09-17:
Hey svpcom, sorry it has been so long since i got back to you. I think that my issue was power related. I have my TX wired to a separate power source and got EZ-wifibroadcast working. I still want to experiment with your version as well as the OSD. when i run tx_standalone.sh i dont get any errors but I am not sure it is actually transmitting. When EZ-wifibroadcast runs i can hear the network card buzz and it will get warm. Neither of those two things happen when I run tx_standalone.sh

svpcom, 2017-09-18:
@terial  What wifi channel and country (crda) do you use with EZ-Wifibroadcast? Try to use same (fix tx_standalone.sh and rx_standalone.sh) with wifibroadcast. CRDA can disable transmit if channel is disabled in your country.

terial, 2017-09-19:
Thanks I will check that out and get back to you. I did do the kernel patches that are in Thebes wifibroadcast repository

maniranjanojha, 2017-12-19:
Hi,

I am facing a totally different issue.
I am getting data on RX console.
Now, the issue is its showing unable to decrypt the packet.
Please help in this case.

output of echo test | nc -u localhost 5600 is 
unable to decrypt session key.
unable to decrypt packet #0x2402d04

svpcom, 2017-12-19:
Have you generate tx.key and rx.key (on one node) and put tx.key to TX node and rx.key to rx node?

maniranjanojha, 2017-12-19:
yes I did that still can you please tell me the location where I should paste them?

svpcom, 2017-12-19:
You should execute keygen only on **one** machine, because in generate a *keypair* - i.e. tx.key and rx.key are linked together. After keygen copy tx.key to TX host and rx.key to RX host. Default search path for tx and rx keys are current directory, but you can override it via ``-K /path/to/your.key``

closed 2018-01-22 by svpcom