svpcom/wfb-ng #8: premade images

geofrancis, 2017-10-15:
could you provide some pre made SD images to make trying your system easer?

svpcom, 2017-10-16:
I don't have any SD images now, because there are a lot of possible configurations that I'll need to support. For quick start you can use image from https://github.com/bortek/EZ-WifiBroadcast/wiki. 
It already have all needed kernel patches. You need to disable original wifibroadcast tx and rx and upload my version. Also you need to setup gstreamer1.0 tools (using apt-get install).  At the last step your need to generate a pair of keys for encryption using ``keygen`` command.

On raspberry pi run:
--------------------------
To start capture RTP stream from raspberry pi camera run:
```
raspivid --nopreview --awb auto -ih -t 0 -w 1920 -h 1080 -fps 30 -b 4000000 -g 30 -pf high -o - | gst-launch-1.0 fdsrc ! h264parse !  rtph264pay !  udpsink host=127.0.0.1 port=5600
```

To broadcast RTP stream:
```
./scripts/tx_standalone.sh wlan1   # where wlan1 is your wifi interface
```

On your host machine run:
-----------------------------------
To receive RTP stream:
```
./scripts/rx_standalone.sh wlan1  # your wifi interface for rx
```

To play RTP stream:
```
gst-launch-1.0 udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' ! rtph264depay ! avdec_h264 ! xvimagesink sync=false
```

dagar, 2017-10-17:
Would this be a good use for docker?

svpcom, 2017-10-18:
@dagar 
Yes, it is a good idea to use docker for automatically cross-build wifibroadcast packages. I've look into travis-ci, but it seems doesn't support native cross build for raspbian. Also there is a problem how to sign and upload packages to any external storage because all keys and api tokens should be in public git repo

dagar, 2017-10-18:
You might want to try semaphoreci for docker builds. I believe it will take care of your authentication with docker hub outside of the repo.

svpcom, 2017-10-20:
Prebuilt binaries for raspberry pi:
https://github.com/svpcom/wifibroadcast/releases/tag/wifibroadcast-17.10.1

Yes21, 2018-06-04:
Hi, I want to test your version of wifibroadcast with low latency.

< _For quick start you can use image from https://github.com/bortek/EZ-WifiBroadcast/wiki.
It already have all needed kernel patches. You need to disable original wifibroadcast tx and rx and upload my version_ >

But I don't know what to do with your prebuilt binaries. I can't find where to replace them in borteks image !

I need a little help, please ...

ps : have you ever think of sharing your code with bortek ? Perhaps he could include it for next updates.

(I'm sad when I see people working each on its own side ...)

svpcom, 2018-06-08:
You can only get modified kernel (from /boot) from these images. And use clear image with this kernel and wifibroadcast binaries. Prebuild wifibroadcast binaries for raspberry pi are in "Releases" section. Or you can build it manually if you have linux with docker installed. Run ``make build_rpi``.  I don't have patched kernel for Raspberry PI because I use wifibroadcast on NanoPI NEO2.

Tyrrx, 2018-07-12:
I tested my setup but on the groundstation I get the error: erroneus pipline: no element "avdec_h264" after trying to play the RTP stream. Another problem is that it says "unable to decode package #xxxxxx"
How do i fix these problems?

svpcom, 2018-07-12:
What is your pipeline?
Have you generated keypair (using keygen utility)? If you see "unable to decode" only a few seconds during startup it is normal. Rx daemon unable to decode stream before it get encrypted session key from TX (which is broadcasted every second).

Tyrrx, 2018-07-12:
I have did it like here: https://dev.px4.io/en/qgc/video_streaming_wifi_broadcast.html
Yes I used the keygen but it remains much longer than just a few secounds.

svpcom, 2018-07-12:
1. What linux disto and what HW do you use for ground station? Have you installed gstreamer1-libav package?
2. Need more info: what wifi card do you use, wifibroadcast output, etc.

Tyrrx, 2018-07-17:
1142666	PKT	0:0:0:0:0:0
1143667	PKT	0:0:0:0:0:0
1144669	PKT	0:0:0:0:0:0
1145670	PKT	0:0:0:0:0:0
1146671	PKT	0:0:0:0:0:0
New session detected
1147671	ANT	0	75:197:203:206
1147671	PKT	76:0:76:7:0:0
1148671	ANT	0	90:203:204:206
1148671	PKT	90:0:90:0:0:0
1149671	ANT	0	88:195:203:207
1149671	PKT	89:0:89:0:0:0
unable to decrypt packet #0x1a08
8 packets lost
1150671	ANT	0	83:202:206:211
1150671	PKT	85:1:84:8:8:0
unable to decrypt packet #0x2204
4 packets lost
1151671	ANT	0	81:209:210:211
1151671	PKT	83:1:82:14:4:0
unable to decrypt packet #0x2a01
7 packets lost
1152671	ANT	0	81:208:209:211
1152671	PKT	83:1:82:8:7:0
unable to decrypt packet #0x3200
8 packets lost
1153671	ANT	0	83:207:208:210
1153671	PKT	85:1:84:11:8:0
unable to decrypt packet #0x3a00
8 packets lost
1154671	ANT	0	79:207:208:209
1154671	PKT	81:1:80:8:8:0
unable to decrypt packet #0x4107
9 packets lost
1155671	ANT	0	80:208:209:210
1155671	PKT	82:1:81:9:9:0
unable to decrypt packet #0x4903
8 packets lost
1156671	ANT	0	80:208:209:211
1156671	PKT	82:1:81:16:8:0
unable to decrypt packet #0x5103
5 packets lost
1157671	ANT	0	78:208:209:211
1157671	PKT	80:1:79:16:5:0
unable to decrypt packet #0x5906
12 packets lost
1158671	ANT	0	81:162:208:210
1158671	PKT	83:1:82:21:12:0
unable to decrypt packet #0x600b
8 packets lost
1159671	ANT	0	82:208:209:210
1159671	PKT	84:1:83:8:8:0
unable to decrypt packet #0x6900
8 packets lost
1160671	ANT	0	81:201:208:211
1160671	PKT	83:1:82:8:8:0
1161671	ANT	0	81:197:203:209
1161671	PKT	82:0:82:0:0:0
unable to decrypt packet #0x770b
8 packets lost
1162671	ANT	0	83:201:206:209
1162671	PKT	85:1:84:12:8:0
unable to decrypt packet #0x7f06
10 packets lost
1163671	ANT	0	81:200:206:212
1163671	PKT	83:1:82:17:10:0
unable to decrypt packet #0x8706
10 packets lost
1164671	ANT	0	83:203:206:211
1164671	PKT	85:1:84:8:10:0
unable to decrypt packet #0x8f09
16 packets lost
1165671	ANT	0	77:188:200:210
1165671	PKT	79:1:78:12:16:0
unable to decrypt packet #0x9704
12 packets lost
1166671	ANT	0	86:208:210:214
1166671	PKT	88:1:87:8:12:0
unable to decrypt packet #0x9f07
9 packets lost
1167671	ANT	0	81:180:210:212
1167671	PKT	83:1:82:14:9:0
unable to decrypt packet #0xa70b
8 packets lost
1168671	ANT	0	82:209:210:212
1168671	PKT	84:1:83:20:8:0
unable to decrypt packet #0xaf0a
8 packets lost
1169671	ANT	0	80:209:210:212
1169671	PKT	82:1:81:16:8:0
unable to decrypt packet #0xb708
8 packets lost
1170671	ANT	0	83:209:210:211
1170671	PKT	85:1:84:16:8:0
unable to decrypt packet #0xbf0b
8 packets lost
1171671	ANT	0	83:209:210:211
1171671	PKT	85:1:84:8:8:0
1172671	ANT	0	85:209:209:211
1172671	PKT	86:0:86:0:0:0
unable to decrypt packet #0xce0b
8 packets lost
1173671	ANT	0	81:179:209:211
1173671	PKT	83:1:82:20:8:0
unable to decrypt packet #0xd700
8 packets lost
1174671	ANT	0	80:208:209:210
1174671	PKT	82:1:81:18:8:0
unable to decrypt packet #0xde02
unable to decrypt packet #0xde0b
6 packets lost
1175671	ANT	0	77:204:206:210
1175671	PKT	80:2:78:14:6:0
unable to decrypt packet #0xe601
7 packets lost
1176671	ANT	0	81:205:206:209
1176671	PKT	83:1:82:8:7:0
unable to decrypt packet #0xee00
8 packets lost
1177671	ANT	0	82:205:208:211
1177671	PKT	84:1:83:12:8:0
unable to decrypt packet #0xf605
11 packets lost
1178671	ANT	0	82:205:207:212
1178671	PKT	84:1:83:8:11:0
unable to decrypt packet #0xfe04
4 packets lost
1179671	ANT	0	84:207:207:210
1179671	PKT	86:1:85:8:4:0
unable to decrypt packet #0x10609
8 packets lost
1180671	ANT	0	84:208:209:211
1180671	PKT	86:1:85:8:8:0
unable to decrypt packet #0x10e04
unable to decrypt packet #0x10f02
12 packets lost
1181671	ANT	0	83:208:210:212
1181671	PKT	86:2:84:8:12:0
unable to decrypt packet #0x1160a
8 packets lost
1182671	ANT	0	84:208:209:211
1182671	PKT	86:1:85:8:8:0
unable to decrypt packet #0x12002
1183671	ANT	0	84:210:210:212
1183671	PKT	86:1:85:0:0:0
6 packets lost
unable to decrypt packet #0x12802
1184671	ANT	0	84:209:210:212
1184671	PKT	86:1:85:9:6:0
6 packets lost
unable to decrypt packet #0x13002
1185671	ANT	0	82:209:210:212
1185671	PKT	84:1:83:14:6:0
6 packets lost
unable to decrypt packet #0x13801
1186671	ANT	0	82:208:211:213

geofrancis, 2018-07-17:
why dont you just make a image that can be put straight onto a sd card rather than making people deal with all these patches.

svpcom, 2018-07-17:
Becase there are a lot of different hardware with multiple linux distributions on it. I use NanoPI NEO2 and have patches for its latest kernel (attached to message above)

geofrancis, 2018-07-19:
yes but few people are going to be running a configuration like that, even if its just compatible with raspberry pi boards it would cover 90%+ of people wanting to try your version of wifibroadcast.

geofrancis, 2018-07-23:
but are you not still needing to use a pi to encode the video?

geofrancis, 2018-07-29:
whats the advantage to doing it like that if the pi can do everything anyway.

geofrancis, 2018-07-30:
why cant the video encoding be done on the neo2?

svpcom, 2018-07-30:
1. it doesn't have drivers for HW H264 encoder (there are some blobs for very old kernels)
2. it doesn't have high-speed camera bus (like CSI on Pi)
So only solution for NEO2 is to use external usb2 camera with h264 encoder inside. The choices are Logitech C920 or Pi zero + Pi camera.

Tyrrx, 2018-08-28:
Okay finally it works even without kernel patches.
I am using the Logitech c920 + Nanopi Neo2 + Alfa AWUS55036NEH (1w output and very small) on the tx side and as rx I am using a HP power pavilion with ubuntu 14.04 and 3 TL-WN722N (Old version).

QiXuanWang, 2019-01-03:
So, what's the working process in case I have two RPIs and using PiCamera,  and have 18.12.1 image downloaded.
Questions:
1. Can I use it like EZWIFI? Meaning: burn the image into SD card and plug and play?
2. If not, do I install normal Raspbian image and execute the commands as described in Release 18.12.1?
3. Will it work with QGC (I'm using andoid phone) out of box? Meaning no setup in QGC. Or do I need to setup as the link described in PX4 "Long Range ... system" link?

svpcom, 2019-01-03:
1. No. There are no plug-and-play images.
2. Yes, you need to install WFB over clear image as described in release notes.
3. QGC listen for incoming video and telemetry stream out-of-box. You need to configure WFB to properly forward data to it. See https://github.com/svpcom/wifibroadcast/wiki/Setup-HOWTO. 
4. Yes. Link to patched driver is on wiki

QiXuanWang, 2019-01-03:
Cool. One more question, since in your post of this thread, you have two commands to make Tx work. So I'm thinking is it possible to make it background processes so I can run other programms too, in case I used a RPI3B+ or something powerful. 

I think it's not possible for other forks as far as I understand.

svpcom, 2019-01-03:
It run in background by default as systemd service (wifibroadcast@drone or wifibroadcast@gs)

QiXuanWang, 2019-01-03:
It's really the correct software I'm looking for. Will try it soon.

Seeelefant, 2019-09-10:
Hi @svpcom , maybe it´s interesting to take a look at https://github.com/HD-Fpv/Open.HD_Image_Builder . That´s a collecting of scripts which create a fresh Open.HD image.
My other remark is on the camera, wouldn´t an IP camera  like https://www.aliexpress.com/item/32931677880.html?spm=a2g0s.9042311.0.0.27424c4dAeolAF
 be another option? As far as I understand it deliveres an H.264/H.265 steram.

svpcom, 2019-09-16:
I know about OpenHD, but it just a collection of bad-structured shell scripts around RPI+pi camera only. The problem of most security ip-cameras is a high latency (it can be up to several seconds). For FPV you need machine-vision cameras.

Seeelefant, 2019-09-17:
Well, even if the Image builder is not structured to the highest level it is at least working. Maybe you could create some structured shell scripts? I like your solution, but takes quite some steps to get it running. Just for my understanding: Which component inside an IP camera takes that long (encoder, network etc...)?

svpcom, 2019-09-17:
@Seeelefant  I prefer to follow classic UNIX way - make small services but not monolithic system. WFB can be build by `make deb` or `make rpm` and `apt-get install` on any linux system, not only for Raspbian on Raspberry PI as OpenHD does.  

Most of security cameras are proprietary SoC, so even if you know what part is bad you unable to fix it. They don't have any public docs, so I don't want to invest my time to hacking device which can be EOL on the next year.

Yes21, 2020-03-10:
Hi to all,
I've written a very detailed guide for installing WFB on a raspberry Pi :
https://github.com/svpcom/wifibroadcast/issues/53

I'm now able to build an image which should work on both drone and gs side with a few modifications on each side.
But unfortunately this is not working :(

My setup : 

- drone :
Pi2B + Pi cam + AWUS036ACH (rtl8812au)

- gs :
Pi3B + 7 " HDMI screen + AWUS1900 (rtl8814au) or ASUS AC56 (rtl8812au)

Both Pis are ethernet connected with my desktop, and I can ssh to them.

The 2 Pis are booting perfectly, and the 2 wifi boards are up and in monitor mode.

But using "wfb-cli gs" on the gs Pi, I can see that all 3 connections failed (refused by the other side)

So if someone has time to read my guide and find what is wrong or missing, I will share my image as soon as it will work.

Thanks for your help !!

closed 2021-02-06 by svpcom