svpcom/wfb-ng #11: name clash

fnoop, 2018-04-23:
Hi, great to see this project take up from and improve on the old befinitiv project.  It's a bit confusing to have the same name, particularly for systems that want to offer both as an option.   Might be an idea to rename the project to avoid confusion?

svpcom, 2018-06-08:
Original wifibroadcast (https://befinitiv.wordpress.com/wifibroadcast-analog-like-transmission-of-live-video-data/) IMHO dead. So now only live ez-wifibroadcast and my version.

fnoop, 2018-06-08:
It's just people associate 'wifibroadcast' with befinitiv, and that code is still used and alive by people even if not developed.  But your choice of course, thanks for your new version :)

closed 2018-06-08 by svpcom