svpcom/wfb-ng #16: Difference between this implementation and Ez-wifibroadcast

myeabhishek, 2018-09-19:
Hello Developers,

Can you please tell me what is the difference between your version and Ez-wifibroadcast implementations?
I want to use for longer range and need encryption as well. As per my understanding Ez-wifibroadcast is widely tested for range of over 20 Kms? Can that be achieved with your implementation as well?

Please specify the differences between two implementations. It would be of great help. Thanks!

svpcom, 2018-09-19:
https://github.com/svpcom/wifibroadcast/wiki
```
Q: What is a difference from original wifibroadcast?

A: Original version of wifibroadcast use a byte-stream as input and splits it to packets of fixed size (1024 by default). If radio packet was lost and this is not corrected by FEC you'll got a hole at random (unexpected) place of stream. This is especially bad if data protocol is not resistent to (was not desired for) such random erasures. So i've rewrite it to use UDP as data source and pack one source UDP packet into one radio packet. Radio packets now have variable size depends on payload size. This is reduces a video latency a lot.
```
There are no difference in HW usage (but it depends what wifi speed you will set), so you will get the same range as with EzWFB

myeabhishek, 2018-09-19:
How about the video quality? Which one will provide with better image quality?

myeabhishek, 2018-09-19:
Hello Svpcom, 
Thankyou for answering my queries. I have one more question. If I want AES encryption with your implementation, how can I go about it?

svpcom, 2018-09-19:
https://download.libsodium.org/doc/secret-key_cryptography/aead/aes-256-gcm
Replace  crypto_aead_chacha20poly1305* with crypto_aead_aes256gcm* (functions and variable types)
But aes is slow compared to chacha20poly1305

svpcom, 2018-09-19:
Be aware - it will not work on ARM:
```
Limitations

The current implementation of this construction is hardware-accelerated and requires the Intel SSSE3 extensions, as well as the aesni and pclmul instructions.

Intel Westmere processors (introduced in 2010) and newer meet the requirements.

There are no plans to support non hardware-accelerated implementations of AES-GCM. If portability is a concern, use ChaCha20-Poly1305 instead.
```

closed 2018-12-23 by svpcom