svpcom/wfb-ng #21: There's no data in RX computer

lucaspenna00, 2018-11-12:
Hello everyone,

I'm developing a drone and i'm working on image transmission. I'm trying to solve this problem using the wifibroadcast, as recommended by PX4 Developer Guide. However, the PX4 Wifibroadcast Tutorial is not much understandable in my opinion and I have some doubts.

> On TX I did the following steps:

```
$ gst-launch-1.0 uvch264src device=/dev/video0 initial-bitrate=6000000 average-bitrate=6000000 iframe-period=1000 name=src auto-start=true \ src.vidsrc ! queue ! video/x-h264,width=1920,height=1080,framerate=30/1 ! h264parse ! rtph264pay ! udpsink host=localhost port=5600
```

```
$ git clone https://github.com/svpcom/wifibroadcast
$ cd wifibroadcast
$ make
$ ./keygen
##Then, i copied the rx.key file to the RX computer
$ sudo ./scripts/tx_standalone.sh wlx00c0ca4aaeec  # where wlx00c0ca4aaeec is your WiFi TX interface
```

The output on TX is:

Setting wlx00c0ca4aaeec to channel 6

> On RX I did the following steps:

```
$ git clone https://github.com/svpcom/wifibroadcast
$ cd wifibroadcast
$ make
##I pasted here the rx.key generated on UAV computer
$ sudo ./scripts/rx_standalone.sh wlx803f5d2217c9  # where wlx803f5d2217c9 is your WiFi TX interface
```
The output on RX is:

Setting wlx803f5d2217c9 to channel 6
wlx803f5d2217c9 has DLT_IEEE802_11_RADIO Encap
17753362	PKT	0:0:0:0:0:0
17754363	PKT	0:0:0:0:0:0
17755364	PKT	0:0:0:0:0:0
17756365	PKT	0:0:0:0:0:0
17757366	PKT	0:0:0:0:0:0
17758367	PKT	0:0:0:0:0:0
17759368	PKT	0:0:0:0:0:0
17760369	PKT	0:0:0:0:0:0
17761370	PKT	0:0:0:0:0:0
17762371	PKT	0:0:0:0:0:0
17763372	PKT	0:0:0:0:0:0
17764373	PKT	0:0:0:0:0:0
17765374	PKT	0:0:0:0:0:0
17766375	PKT	0:0:0:0:0:0
17767376	PKT	0:0:0:0:0:0 (...)

Finally, due to the RX output I suppose i'm not receiving data from UAV. Could you explain me how I must proceed in order to solve this? The PX4 Developer Guide was not enough didactic for me. 

The Hardware Setup is:

On TX:
- Alfa AWUSO36NH (2.4Ghz) - Note: in both scripts (rx_standalone and tx_standalone) I changed the band option: 5G to 2G
- Logitech C920 Camera
- Dragonboard 410c (UAV computer) (Debian 10.5)

On RX:
- Norion Wirelles N (2.4GHz)
- Computer running Ubuntu 16.04 

If there's missing information, please tell me.
Thanks for the help!

svpcom, 2018-11-13:
Try these steps: https://github.com/svpcom/wifibroadcast/wiki/Troubleshooting

lucaspenna00, 2018-11-13:
Hello svpcom, thanks for the reply.

I followed these steps and it produced these outputs:

> Run scripts/tx_standalone.sh wlx00c0ca4aaeec  on TX host
Setting wlx00c0ca4aaeec to channel 6

>Run scripts/rx_standalone.sh wlx803f5d2217c9 on RX host
Setting wlx803f5d2217c9 to channel 6
wlx803f5d2217c9 has DLT_IEEE802_11_RADIO Encap
17753362	PKT	0:0:0:0:0:0
17754363	PKT	0:0:0:0:0:0
17755364	PKT	0:0:0:0:0:0
17756365	PKT	0:0:0:0:0:0
17757366	PKT	0:0:0:0:0:0
17758367	PKT	0:0:0:0:0:0
17759368	PKT	0:0:0:0:0:0
17760369	PKT	0:0:0:0:0:0
17761370	PKT	0:0:0:0:0:0
17762371	PKT	0:0:0:0:0:0
17763372	PKT	0:0:0:0:0:0
17764373	PKT	0:0:0:0:0:0
17765374	PKT	0:0:0:0:0:0
17766375	PKT	0:0:0:0:0:0
17767376	PKT	0:0:0:0:0:0 (...)

> On RX: $ nc -u localhost 5600
there was no output

> On TX: $ echo test | nc -u localhost 5600
there was no output too

> On TX: $ tcpdump -i wlx00c0ca4aaeec -n -p

tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlx00c0ca4aaeec, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
14:01:06.904129 1.0 Mb/s 2437 MHz 11b -77dBm signal antenna 1 Beacon (caverna) [1.0* 2.0* 5.5* 11.0* 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
14:01:06.927395 2.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Probe Response (eduroam) [2.0* 5.5 11.0 18.0 24.0 36.0 48.0 54.0 Mbit] CH: 6, PRIVACY
14:01:06.943661 1.0 Mb/s 2437 MHz 11b -81dBm signal antenna 1 Beacon (LMO02) [1.0 2.0 5.5 11.0 9.0 18.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
14:01:06.946564 1.0 Mb/s 2437 MHz 11b -45dBm signal antenna 1 Beacon (SICAM) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
14:01:06.960596 2.0 Mb/s 2437 MHz 11b -73dBm signal antenna 1 Beacon (eduroam) [2.0* 5.5 11.0 18.0 24.0 36.0 48.0 54.0 Mbit] ESS CH: 6, PRIVACY
14:01:06.961438 2.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Probe Response (eduroam) [2.0* 5.5 11.0 18.0 24.0 36.0 48.0 54.0 Mbit] CH: 6, PRIVACY
14:01:06.971277 11.0 Mb/s 2437 MHz 11b -67dBm signal antenna 1 Data IV:78d8 Pad 20 KeyID 1
14:01:06.971752 11.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 Data IV:178d8 Pad 20 KeyID 1
14:01:07.004398 2.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 EAP packet (0) v1, len 5
14:01:07.006667 1.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Beacon (caverna) [1.0* 2.0* 5.5* 11.0* 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
14:01:07.047654 2.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Probe Response (eduroam) [2.0* 5.5 11.0 18.0 24.0 36.0 48.0 54.0 Mbit] CH: 6, PRIVACY
14:01:07.048730 2.0 Mb/s 2437 MHz 11b -77dBm signal antenna 1 Probe Response (eduroam) [2.0* 5.5 11.0 18.0 24.0 36.0 48.0 54.0 Mbit] CH: 6, PRIVACY
14:01:07.056670 1.0 Mb/s 2437 MHz 11b -45dBm signal antenna 1 Beacon (SICAM) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
14:01:07.061677 2.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Beacon (eduroam) [2.0* 5.5 11.0 18.0 24.0 36.0 48.0 54.0 Mbit] ESS CH: 6, PRIVACY
14:01:07.072832 1.0 Mb/s 2437 MHz 11b -73dBm signal antenna 1 Beacon (Citi) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY (...)

> On RX: $ tcpdump -i wlx803f5d2217c9 -n -p

tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlx803f5d2217c9, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
12:06:35.674747 11.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Data IV:9774f4 Pad 20 KeyID 1
12:06:35.675244 11.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Data IV:9874f4 Pad 20 KeyID 1
12:06:35.675628 11.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Data IV:9974f4 Pad 20 KeyID 1
12:06:35.688940 2.0 Mb/s 2437 MHz 11b -77dBm signal antenna 1 Data IV: 36 Pad 20 KeyID 1
12:06:35.726474 1.0 Mb/s 2437 MHz 11b -85dBm signal antenna 1 Beacon (caverna) [1.0* 2.0* 5.5* 11.0* 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
12:06:35.767735 1.0 Mb/s 2437 MHz 11b -55dBm signal antenna 1 Beacon (SICAM) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
12:06:35.774026 1.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Beacon (Citi) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
12:06:35.774599 11.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Data IV:9a74f4 Pad 20 KeyID 1
12:06:35.774989 11.0 Mb/s 2437 MHz 11b -81dBm signal antenna 1 Data IV:9b74f4 Pad 20 KeyID 1
12:06:35.775254 11.0 Mb/s 2437 MHz 11b -81dBm signal antenna 1 Data IV:9c74f4 Pad 20 KeyID 1
12:06:35.828950 1.0 Mb/s 2437 MHz 11b -83dBm signal antenna 1 Beacon (caverna) [1.0* 2.0* 5.5* 11.0* 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
12:06:35.833821 1.0 Mb/s 2437 MHz 11b -83dBm signal antenna 1 Probe Request (eduroam) [1.0 2.0 5.5 11.0 Mbit]
12:06:35.866004 1.0 Mb/s 2437 MHz 11b -55dBm signal antenna 1 Beacon (SICAM) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
12:06:35.876324 1.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Beacon (Citi) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
12:06:35.878263 11.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Data IV:a074f4 Pad 20 KeyID 1
12:06:35.878687 11.0 Mb/s 2437 MHz 11b -81dBm signal antenna 1 Data IV:a174f4 Pad 20 KeyID 1
12:06:35.931348 1.0 Mb/s 2437 MHz 11b -85dBm signal antenna 1 Beacon (caverna) [1.0* 2.0* 5.5* 11.0* 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
12:06:35.967237 1.0 Mb/s 2437 MHz 11b -55dBm signal antenna 1 Beacon (SICAM) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
[... cut ...]
12:06:36.081971 11.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Data IV:a374f4 Pad 20 KeyID 1
12:06:36.122696 1.0 Mb/s 2437 MHz 11b -81dBm signal antenna 1 Probe Response (Citi) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] CH: 6, PRIVACY

Finally, I configured the QGroundControl (QGC) Video as the following:

Video Source: UDP Video Stream
UDP Port: 5600
Aspect Ratio: 1.777777

But there's no video. The video output on QGC is: "Waiting for video".

Please, tell me if there is any error or wrong procedure.

Thanks!

svpcom, 2018-11-13:
It seems that nothing is injected. Try to see tcpdump on TX side while gstreamer is running. Also check that NetworkManager is disabled or 
```
[keyfile]
unmanaged-devices=interface-name:wlx00c0ca4aaeec
```
is added on /etc/NetworkManager/NetworkManager.conf on TX (also add to RX with its card name)

lucaspenna00, 2018-11-14:
Svpcom, I disabled the NetworkManager by this command: 

```
$ sudo systemctl stop NetworkManager
#both RX and TX

```
Then, I repeated the steps (as described on my post) and I`m still not receiving data on QGroundControl.

The output produced by tcpdump (on TX) (while gstreamer is running on TX and NetworkManager is disabled) is:
cpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlx00c0ca4aaeec, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
13:10:56.846333 1.0 Mb/s 2437 MHz 11b -53dBm signal antenna 1 Beacon (SICAM) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
13:10:56.871115 1.0 Mb/s 2437 MHz 11b -75dBm signal antenna 1 Beacon (caverna) [1.0* 2.0* 5.5* 11.0* 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
13:10:56.902843 1.0 Mb/s 2437 MHz 11b -69dBm signal antenna 1 Probe Response (Citi) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] CH: 6, PRIVACY
13:10:56.909708 1.0 Mb/s 2437 MHz 11b -73dBm signal antenna 1 Probe Response (Citi) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] CH: 6, PRIVACY
13:10:56.913038 1.0 Mb/s 2437 MHz 11b -73dBm signal antenna 1 Probe Response (Citi) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] CH: 6, PRIVACY
13:10:56.925994 2437 MHz 11n -65dBm signal antenna 1 43.3 Mb/s MCS 4 20 MHz short GI Data IV: 9c Pad 20 KeyID 0
13:10:56.927080 9.0 Mb/s 2437 MHz 11g -77dBm signal antenna 1 Data IV: ae Pad 20 KeyID 0
13:10:56.937648 1.0 Mb/s 2437 MHz 11b -73dBm signal antenna 1 Beacon (Citi) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
13:10:56.937770 11.0 Mb/s 2437 MHz 11b -75dBm signal antenna 1 Data IV:4d6848 Pad 20 KeyID 1
13:10:56.938388 11.0 Mb/s 2437 MHz 11b -73dBm signal antenna 1 Data IV:4e6848 Pad 20 KeyID 1
13:10:56.944717 1.0 Mb/s 2437 MHz 11b -53dBm signal antenna 1 Beacon (SICAM) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
13:10:56.949108 1.0 Mb/s 2437 MHz 11b -67dBm signal antenna 1 Beacon (PoliSemFio) [1.0* 2.0* 5.5 11.0 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6
13:10:56.951012 1.0 Mb/s 2437 MHz 11b -57dBm signal antenna 1 Beacon (eduroam) [1.0* 2.0* 5.5 11.0 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
13:10:56.952895 1.0 Mb/s 2437 MHz 11b -59dBm signal antenna 1 Beacon (DEV) [1.0* 2.0* 5.5 11.0 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
13:10:56.973552 1.0 Mb/s 2437 MHz 11b -75dBm signal antenna 1 Beacon (caverna) [1.0* 2.0* 5.5* 11.0* 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
13:10:57.043491 1.0 Mb/s 2437 MHz 11b -51dBm signal antenna 1 Beacon (SICAM) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
13:10:57.052119 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 Beacon (PoliSemFio) [1.0* 2.0* 5.5 11.0 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6
13:10:57.054807 1.0 Mb/s 2437 MHz 11b -59dBm signal antenna 1 Beacon (eduroam) [1.0* 2.0* 5.5 11.0 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
13:10:57.056694 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 Beacon (DEV) [1.0* 2.0* 5.5 11.0 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
13:10:57.075779 1.0 Mb/s 2437 MHz 11b -77dBm signal antenna 1 Beacon (caverna) [1.0* 2.0* 5.5* 11.0* 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
13:10:57.101075 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 
13:10:57.101509 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 
13:10:57.104009 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 
13:10:57.104647 1.0 Mb/s 2437 MHz 11b -67dBm signal antenna 1 
13:10:57.105249 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 
13:10:57.106000 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 
13:10:57.106620 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 
13:10:57.125887 1.0 Mb/s 2437 MHz 11b -79dBm signal antenna 1 Beacon (PoliSemFio) [1.0* 2.0* 5.5 11.0 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6
13:10:57.142363 1.0 Mb/s 2437 MHz 11b -73dBm signal antenna 1 Beacon (Citi) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
13:10:57.142427 11.0 Mb/s 2437 MHz 11b -73dBm signal antenna 1 Data IV:4f6848 Pad 20 KeyID 1
13:10:57.153474 1.0 Mb/s 2437 MHz 11b -51dBm signal antenna 1 Beacon (SICAM) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
13:10:57.160525 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 Beacon (PoliSemFio) [1.0* 2.0* 5.5 11.0 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6
13:10:57.162426 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 Beacon (eduroam) [1.0* 2.0* 5.5 11.0 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
13:10:57.164438 1.0 Mb/s 2437 MHz 11b -57dBm signal antenna 1 Beacon (DEV) [1.0* 2.0* 5.5 11.0 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY
13:10:57.244887 1.0 Mb/s 2437 MHz 11b -73dBm signal antenna 1 Beacon (Citi) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 6, PRIVACY
13:10:57.259536 1.0 Mb/s 2437 MHz 11b -57dBm signal antenna 1 Beacon (PoliSemFio) [1.0* 2.0* 5.5 11.0 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6                                                    
13:10:57.261530 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 Beacon (eduroam) [1.0* 2.0* 5.5 11.0 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY                                              
13:10:57.263486 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 Beacon (DEV) [1.0* 2.0* 5.5 11.0 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY                                                  
13:10:57.280918 1.0 Mb/s 2437 MHz 11b -77dBm signal antenna 1 Beacon (caverna) [1.0* 2.0* 5.5* 11.0* 18.0 24.0 36.0 54.0 Mbit] ESS CH: 6, PRIVACY                                            
[... cut ...]
13:10:57.314292 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 Action (84:d4:7e:06:53:21): BA ADDBA Request                                                                                   
13:10:57.315038 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 Action (84:d4:7e:06:53:21): BA ADDBA Request                                                                                   
13:10:57.315660 1.0 Mb/s 2437 MHz 11b -65dBm signal antenna 1 Action (84:d4:7e:06:53:21): BA ADDBA Request                                                                                   
13:10:57.316418 1.0 Mb/s 2437 MHz 11b -67dBm signal antenna 1 Action (84:d4:7e:06:53:21): BA ADDBA Request

lucaspenna00, 2018-11-14:
In order to check if there`s iptable rules dropping these udp packets, I ran the command:
$ sudo iptables -S 
the output is:
-P INPUT ACCEPT
-P FORWARD ACCEPT
-P OUTPUT ACCEPT

So, I suppose there`s no iptables dropping the packets.

I have no idea how I can strace the TX process in order to confirm if it`s receiving udp packets from gstreamer. Could you recommend any command? 

Thank you

svpcom, 2018-11-14:
```
strace -ff -p PID
```
where PID is a pid of tx process:  `ps uaxw | grep tx`

lucaspenna00, 2018-11-14:
These commands produced this output: 

[image]

svpcom, 2018-11-14:
Please check your gstreamer settings. It seems that udp packets doesn't reach localhost:5600

lucaspenna00, 2018-11-14:
Which settings do you recommend to check? And where exactly? Sorry, I`m newbie as you can see haha.

svpcom, 2018-11-16:
Try to use `echo test | nc -u localhost 5600` while strace is running. Trace `./tx -p1` process, not a bash script

lucaspenna00, 2018-12-05:
On TX side:

`linaro@linaro-alip:~$ ps uaxw | grep tx`

root      3671  0.0  0.3   6876  3172 pts/1    S+   16:55   0:00 sudo ./scripts/tx_standalone.sh wlx00c0ca4aaeec
root      3672  0.0  0.2   4892  2808 pts/1    S+   16:55   0:00 /bin/bash ./scripts/tx_standalone.sh wlx00c0ca4aaeec
root      3683  0.0  0.1   7252  1600 pts/1    S+   16:55   0:00 ./tx -p 1 -u 5600 wlx00c0ca4aaeec
root      3684  0.0  0.1   7252  1532 pts/1    S+   16:55   0:00 ./tx -m 100 -p 2 -u 14550 wlx00c0ca4aaeec
linaro    4001  0.0  0.0   4396   572 pts/2    S+   17:04   0:00 grep tx

`linaro@linaro-alip:~$ sudo strace -ff -p 3683`

strace: Process 3683 attached
recvfrom(3, "test\n", 1446, 0, NULL, NULL) = 5
sendto(4, "\0\0\r\0\0\200\10\0\10\0\7\5\1\10\1\0\0\377\377\377\377\377\377\23\"3DU\1\23\"3"..., 110, 0, NULL, 0) = 110
sendto(4, "\0\0\r\0\0\200\10\0\10\0\7\5\1\10\1\0\0\377\377\377\377\377\377\23\"3DU\1\23\"3"..., 69, 0, NULL, 0) = 69
recvfrom(3, 

(While strace is running)

`linaro@linaro-alip:~$ echo test | nc -u localhost 5600` 

Then, this command (echo test) produced the above output. In addition, I received an output on RX side.

On RX side:

When I ran the command "echo test" on TX side, it produced this output** (on RX side):

.
.
.
45626512	PKT	0:0:0:0:0:0
45627512	PKT	0:0:0:0:0:0
45628513	PKT	0:0:0:0:0:0
45629514	PKT	0:0:0:0:0:0
45630515	PKT	0:0:0:0:0:0
45631516	PKT	0:0:0:0:0:0
**45632516	ANT	1	1:-21:-21:-21
45632516	PKT	1:0:1:0:0:0
45633517	PKT	0:0:0:0:0:0
45634518	PKT	0:0:0:0:0:0
45635519	PKT	0:0:0:0:0:0
45636520	PKT	0:0:0:0:0:0
45637521	PKT	0:0:0:0:0:0
45638522	PKT	0:0:0:0:0:0
.
.  
.

Thus, I think the connection between the two antennas is working well. It's correct? But I didn't receive any image data on QGroundControl.

lucaspenna00, 2018-12-06:
Hello svpcom. I solved the problem replacing the "localhost" argument on GStreamer by "127.0.0.1". Now, the QGroundControl is decoding the video. Thank you for help me.

closed 2018-12-07 by svpcom