svpcom/wfb-ng #22: security issue

dposea, 2018-11-13:
The following warning points to a security issue with printf in linux. Google printf exploit for details.

wifibroadcast.hpp: In instantiation of ‘std::__cxx11::string string_format(const char*, Args ...) [with Args = {}; std::__cxx11::string = std::__cxx11::basic_string<char>]’:
tx.cpp:137:68:   required from here
wifibroadcast.hpp:46:27: warning: format not a string literal and no format arguments [-Wformat-security]
     size_t size = snprintf(nullptr, 0, format, args ...) + 1; // Extra space fo
                           ^
wifibroadcast.hpp:46:27: warning: format not a string literal and no format arguments [-Wformat-security]
wifibroadcast.hpp:48:13: warning: format not a string literal and no format arguments [-Wformat-security]
     snprintf(buf.get(), size, format, args ...);
             ^
wifibroadcast.hpp:48:13: warning: format not a string literal and no format arguments [-Wformat-security]

svpcom, 2018-11-13:
This is not a bug but a compiler warning. `string_format` function is line `snprintf` but outputs to a STL string. Old gcc is not smart enough and generate a warning due to non-literal format string and varargs as argument list

dposea, 2018-11-14:
Vasily, I realize it's a warning but points to a condition that can be exploited by a hacker. It's simple to fix the warning by adding a cast to the format function. There are lots of articles on printf exploits and how they create a security issue. It's not a really big concern on a raspberry pi but something I am now aware of and try to pass along to other developers. I had no idea until about 6 months ago that such a big security hole existed. Nearly every program I have written has a printf in it.

svpcom, 2018-11-14:
I've rewrote this function

closed 2018-11-16 by svpcom