svpcom/wfb-ng #24: Failed to build wifibroadcast-18.12

avgorbi, 2018-12-20:
```
pi@raspberrypi:~/wifibroadcast $ make
PYTHONPATH=`pwd` trial telemetry.tests
telemetry
  tests
    test_proxy ...                                                      [ERROR]

===============================================================================
[ERROR]
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/twisted/trial/runner.py", line 498, in loadPackage
    module = modinfo.load()
  File "/usr/lib/python2.7/dist-packages/twisted/python/modules.py", line 383, in load
    return self.pathEntry.pythonPath.moduleLoader(self.name)
  File "/usr/lib/python2.7/dist-packages/twisted/python/reflect.py", line 303, in namedAny
    topLevelPackage = _importAndCheckStack(trialname)
  File "/usr/lib/python2.7/dist-packages/twisted/python/reflect.py", line 250, in _importAndCheckStack
    reraise(excValue, excTraceback)
  File "/home/pi/wifibroadcast/telemetry/tests/test_proxy.py", line 8, in <module>
    from telemetry.proxy import UDPProxyProtocol
  File "/home/pi/wifibroadcast/telemetry/proxy.py", line 21, in <module>
    import mavlink
  File "/home/pi/wifibroadcast/telemetry/mavlink.py", line 9, in <module>
    from builtins import range
exceptions.ImportError: No module named builtins

telemetry.tests.test_proxy
-------------------------------------------------------------------------------
Ran 1 tests in 0.023s

FAILED (errors=1)
Makefile:43: recipe for target 'test' failed
make: *** [test] Error 1
```

I am new to python, so I don’t know what to do with it.
C ++ part of the project is building fine.

svpcom, 2018-12-20:
apt-get install python-twisted python-future

avgorbi, 2018-12-20:
```
pi@raspberrypi:~/wifibroadcast $ sudo apt-get install pyton-twisted python-future
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package pyton-twisted
E: Unable to locate package python-future
```

It's look like wrong python version
```
pi@raspberrypi:~/wifibroadcast $ python -V
Python 2.7.9
```
It;s correct?

avgorbi, 2018-12-20:
This help: 
```
sudo pip install future
```

svpcom, 2018-12-20:
Also need to install python-virtualenv

avgorbi, 2018-12-20:
Thanks. I do it. All building success.

avgorbi, 2018-12-20:
Last question. Can I use only wfb_tx & wfb_rx for 2 way udp transmition?

svpcom, 2018-12-20:
Yes, but in case of mavlink you will receive packets on one udp port (which listen wfb_tx) and send from another (which use wfb_rx). This may confuse mavlink router. Python daemon acts as proxy to use single port and route packets to/from wfb_tx/rx daemons.

svpcom, 2018-12-20:
Also python daemon do mavlink aggregation (collect multiple mavlink packets to a batch) and TX diversity (make decision which wifi card use for TX based in RX RSSI)

svpcom, 2018-12-20:
If you need C-only code and don't need TX diversity - you can backport "mavlink mode" (packet aggregation) from previous release. It was moved to python side in this version.

avgorbi, 2018-12-20:
Thank you. Ok.

closed 2018-12-23 by svpcom