svpcom/wfb-ng #25: Support of BananaPi

QiXuanWang, 2019-01-02:
I remember I saw you listed hardware comparison but I couldn't find it right now. The impression is that this project doesn't really care too much about computer hardware?

I was wondering if you could support BPI? You'll have some advanced/expensive choice regarding computing power which could be better fit when needed.

svpcom, 2019-01-02:
If bananapi supports modern kernel (4.x.x) and has usb ports it should works. If kernel is old (3.x.x) then you need to apply additional patches to enable packet injection.

QiXuanWang, 2019-01-02:
I see. I think it could same system as RPI3. Just a FYI so in case your project evolved into that stage that a strong hardware would be helpful then there is an option. I'll probably try it out when I fully understand your process.

svpcom, 2019-01-02:
My primary hw setup is NanoPI NEO2

QiXuanWang, 2019-01-03:
No problem. Please close this since it's not an issue.

closed 2019-01-03 by svpcom