svpcom/wfb-ng #29: can't decrypt error

QiXuanWang, 2019-01-24:
I was able to start the process but somehow RX reported error as:
...
379062	PKT	0:0:0:0:0:0
380063	PKT	0:0:0:0:0:0
381064	PKT	0:0:0:0:0:0
382065	PKT	0:0:0:0:0:0
unable to decrypt packet #0xd801
unable to decrypt packet #0xd802
unable to decrypt packet #0xd803
unable to decrypt packet #0xd804
unable to decrypt packet #0xd805
unable to decrypt packet #0xd806
383065	PKT	6:6:0:0:0:0
384067	PKT	0:0:0:0:0:0
385068	PKT	0:0:0:0:0:0
386069	PKT	0:0:0:0:0:0
387070	PKT	0:0:0:0:0:0
unable to decrypt packet #0x1c30a
unable to decrypt packet #0x1c30b
unable to decrypt packet #0x1c400
unable to decrypt packet #0x1c401
unable to decrypt packet #0x1c402
unable to decrypt packet #0x1c403
unable to decrypt packet #0x1c404
388070	PKT	7:7:0:0:0:0
389071	PKT	0:0:0:0:0:0
390073	PKT	0:0:0:0:0:0
391074	PKT	0:0:0:0:0:0
392075	PKT	0:0:0:0:0:0
393076	PKT	0:0:0:0:0:0
394077	PKT	0:0:0:0:0:0
395078	PKT	0:0:0:0:0:0
396079	PKT	0:0:0:0:0:0
...
I'm using the both keys generated on TX (copied gs.key from TX to RX)
So how can I debug for next step?

svpcom, 2019-01-24:
Check that gs.key and drone.key was generated in the same run of wfb_keygen program.
Small amount of `decrypt error` after startup can be normal because encrypted session key is announced every second and while RX doesn't get an encrypted session key it unable to descript packet stream. See https://github.com/svpcom/wifibroadcast/wiki/Encryption

QiXuanWang, 2019-01-24:
Thanks. I,m no expert on this.  Now I'm trying to use VLC to view the RTP/UDP stream?
I tried on both TX and RX with rtp://@:5600 but failed. I'm not running advanced setup for telemetry yet. I followed the PX4+WifiBroadcast instructions.

svpcom, 2019-01-24:
PX4 + WFB instruction is quite old. I've filled a pull request https://github.com/PX4/Devguide/pull/723 , but it not still applied.  Use gstreamer or QGroundControl instead of vlc. VLC is unable to show raw RTP stream without dirty hacks.

QiXuanWang, 2019-01-25:
Thanks. Just to update what I had now by following instructions of PX4 version.
It worked, with extremely slow video. Actually, it's static indeed. I don;t know if it's RPI3 setting problem or WFB software problem. But it won't work compared with EZWifibroadcast.  The gst-launch did open up a video window but the content never changed...

I'll try to follow the advanced setup guide you posted today and see if I can get better results.

There are several points I don't really understand about that guide after reading it over and over again :)
But one particular thing is, mavlink_router should only be used on TX right?

svpcom, 2019-01-25:
Yes, mavlink_router is needed only if you want to transmit teletemtry on the drone. If you want to use video only you don't need to configure it

svpcom, 2019-01-25:
yes. You could use gstremer on the ground to decode video

QiXuanWang, 2019-01-25:
It seems my problem is that RX can't receive data. Either it can't decrypt or it fails to receive at all. Have tried both 5G and 2G but no difference. There must be some steps that I missed.  I'll try the trouble-shooting solution.
A few more questions:
1. For telemetry, can I use port=5600 and 14550 for both TX and RX, for wifibroadcast.cfg?
2. When  I run ./script/tx_standalone,sh wlan1, will wifibroadcast.cfg work? From what I can tell, it won't but just to confirm.
3. for TX, 
 "raspivid --nopreview --awb auto -ih -t 0 -w 640 -h 480 -fps 30 -b 4000000 -g 30 -pf high -o - | gst-launch-1.0 fdsrc ! h264parse ! rtph264pay !  udpsink host=127.0.0.1 port=5600"
Is the "4000000" correct or should I use "4000"?

QiXuanWang, 2019-01-25:
This is some sample RX infor:

override block 0x701f with 0 fragments
override block 0x7020 with 0 fragments
override block 0x7021 with 0 fragments
override block 0x7022 with 0 fragments
override block 0x7023 with 0 fragments
override block 0x7024 with 0 fragments
override block 0x7025 with 0 fragments
override block 0x7026 with 0 fragments
override block 0x7027 with 0 fragments
override block 0x7028 with 5 fragments
3600 packets lost
87223632	ANT	1	11:-22:-20:-20
87223632	ANT	0	11:-16:-14:-14
87223632	PKT	11:0:11:8:3600:0
87224633	PKT	0:0:0:0:0:0
87225634	PKT	0:0:0:0:0:0
87226635	PKT	0:0:0:0:0:0

Does it indicate failing to receive or failing to decrypt??

svpcom, 2019-01-25:
1. Yes
2. No, wifibroadcast.cfg is only for python daemon. wfb_tx and wfb_rx utilities use only command-line parameters.
3. My config for Pi camera: ``gst-launch-1.0 rpicamsrc annotation-mode=0xc bitrate=4000000 keyframe-interval=49 inline-headers=true sensor-mode=5 drc=0 preview=false fullscreen=false iso=0 awb-mode=sunlight metering-mode=matrix ! video/x-h264,profile=high,width=1280,height=720,framerate=49/1,stream-format=byte-stream !  h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=127.0.0.1 port=5600``

Check you RPI CPU usage. It may be not enough cpu power for camera + wfb

QiXuanWang, 2019-01-28:
The decrypt issue should be related to wificard power issue. Close until I have further findings.

closed 2019-01-28 by QiXuanWang