svpcom/wfb-ng #30: telemetry data not available

QiXuanWang, 2019-01-28:
So now I have finished the testing with video and it works with manual debugging mode as well as daemon mode.
I moved to telemetry data. Notice I'm using 9271 cards and made necessary changes before I finally sold the 8821 cards. 
Currently video works but wfb-cli gs shows no telemetry data. I'm wondering if it's the port setting somewhere is not correct. Pasted the log here and some help needed:

TX:
-- The start-up result is done.
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] Log opened.
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] # iw reg set BO
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] # ifconfig wlan1 down
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] # iw dev wlan1 set monitor otherbss
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] # ifconfig wlan1 up
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] # iw dev wlan1 set channel 6 HT20
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] Listen for telem stream 1(RX), 2(TX) on 0.0.0.0:14550
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] UDPProxyProtocol starting on 14550
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] UDPProxyProtocol starting on 14700
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] UDPProxyProtocol starting on 34531
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] Telem RX: /usr/bin/wfb_rx -p 1 -u 14700 -K /etc/drone.key wlan1
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] Telem TX: /usr/bin/wfb_tx -p 2 -u 14701 -K /etc/drone.key -B 20 -G long -S 1 -M 1
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] Started telem rx
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] Started telem tx
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] Listen for video stream 3 on 0.0.0.0:5600
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] Started video tx
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] Video: /usr/bin/wfb_tx -p 3 -u 5600 -K /etc/drone.key -B 20 -G long -S 1 -M 1 wla
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] telem tx: Listen on 14701 for wlan1
Jan 28 18:22:25 raspberrypi python[14687]: 2019-01-28 18:22:25+0800 [-] video tx: Listen on 5600 for wlan1
Jan 28 18:22:26 raspberrypi python[14687]: 2019-01-28 18:22:26+0800 [-] telem rx: wlan1 has DLT_IEEE802_11_RADIO Encap

TX Mavlink Router:
Open UDP [4] 127.0.0.1:14550  
Open TCP [5] 0.0.0.0:5790 *

RX:
-- The start-up result is done.
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] Log opened.
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] # iw reg set BO
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] # ifconfig wlan1 down
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] # iw dev wlan1 set monitor otherbss
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] # ifconfig wlan1 up
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] # iw dev wlan1 set channel 6 HT20
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] Connect telem stream 2(RX), 1(TX) to 127.0.0.1:14550
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] UDPProxyProtocol starting on 54177
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] UDPProxyProtocol starting on 14600
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] UDPProxyProtocol starting on 35354
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] Telem RX: /usr/bin/wfb_rx -p 2 -u 14600 -K /etc/gs.key wlan1
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] Telem TX: /usr/bin/wfb_tx -p 1 -u 14601 -K /etc/gs.key -B 20 -G long -S 1 -M 1 wlan1
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] AntennaFactory starting on 8001
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] Started telem rx
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] Started telem tx
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] Send video stream 3 to 127.0.0.1:5600
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] AntennaFactory starting on 8002
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] Started video rx
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] Video: /usr/bin/wfb_rx -p 3 -c 127.0.0.1 -u 5600 -K /etc/gs.key wlan1
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] telem tx: Listen on 14601 for wlan1
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] video rx: wlan1 has DLT_IEEE802_11_RADIO Encap
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] video rx: unable to decrypt packet #0x7b00
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] video rx: unable to decrypt packet #0x7b01
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] video rx: unable to decrypt packet #0x7b02
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] telem rx: wlan1 has DLT_IEEE802_11_RADIO Encap
Jan 28 18:22:42 raspberrypi python[23438]: 2019-01-28 18:22:42+0800 [-] video rx: unable to decrypt packet #0x7b03

RX wfb-cli:
──[video]─────────────────────────────────────────────────────────────────────┐
│NO TX                                                                         │
│RX PKT:   recv 187015 d_ok 186988 fec_r 5460 lost 1028 d_err 27 bad 0         │
│RX PKT/s: recv 87 d_ok 87 fec_r 6 lost 0 d_err 0 bad 0                        │
│                                                                              │
│0000: 86 pkt/s, rssi -38 < -36 < -35                                          │
│                                                                              │
│                                                                              │
│                                                                              │
│                                                                              │
└──────────────────────────────────────────────────────────────────────────────┘
┌──[telem]─────────────────────────────────────────────────────────────────────┐
│TX ANT 00**                                                                   │
│RX PKT:   recv 0 d_ok 0 fec_r 0 lost 0 d_err 0 bad 0                          │
│RX PKT/s: recv 0 d_ok 0 fec_r 0 lost 0 d_err 0 bad 0                          │
│                                                                              │
│Link lost!             

Any idea which setting is wrong?

svpcom, 2019-01-28:
1. What is your /etc/wifibrodcast.cfg?
2. How you feed telemetry to WFB?

QiXuanWang, 2019-01-28:
I'm using the example you provided:
[drone_mavlink]
 peer = 'listen://0.0.0.0:14550'   # incoming connection
 # peer = 'connect://127.0.0.1:14550'  # outgoing connection
 [drone_video]
 peer = 'listen://0.0.0.0:5600'  # listen for video stream (gstreamer on drone)

2. This is what I didn't quite understand. Isn't it automatically done when mavlink-router is up?

QiXuanWang, 2019-01-29:
I added ttyAMA0 config.
It seems working by observing wfb-cli output:

│TX ANT 00**                                                                   │
│RX PKT:   recv 280 d_ok 280 fec_r 19 lost 135 d_err 0 bad 0                   │
│RX PKT/s: recv 2 d_ok 2 fec_r 0 lost 0 d_err 0 bad 0                          │
│                                                                              │
│0000: 1 pkt/s, rssi -34 < -34 < -34  

Is this output reasonable?

And, mavlink router kept reporting:
5 messages to unknown endpoints in the last 5 seconds
5 messages to unknown endpoints in the last 5 seconds
5 messages to unknown endpoints in the last 5 seconds
5 messages to unknown endpoints in the last 5 seconds
...

Any suggestion?

QiXuanWang, 2019-01-30:
1.5M .. How do you get this number? I thought 115200 was recomendded at ardupilot site. But I'll try that.
One thing is that I was trying MavProxy beforing trying this project. It worked but when I started wifibroadcast drone daemon, the link is down. Is it expected?
How can I get/send telemetry msg then?  Any suggestions to control mavlink msg in general? As you may guess that this is my main focus and also why your version is more appealing to me.

svpcom, 2019-01-30:
I use PX4 (not ardupilot). It supports up to 2Mb/s for telemetry UART on pixhawk hardware. Mavproxy is slow for arm hardware - use mavlink_router instead. 
```
[autopilot] <--(uart)-->[mavlink_router]-->(udp 14550)-->[wfb_drone]--...-(wifi)-...[wfb_gs]--->(udp 14550)-->[QGrondControl]
```
Your config file for malink-routerd should contains two endpoints:
1. for /dev/ttyXXX uart (with proper uart speed)
2. for udp 127.0.0.1:14550 (assuming that wfb_drone **listen** this port)

Also check that there are no firewall enabled and you add you added to /etc/NetworkManager/NetworkManager.conf:
```
[keyfile]
unmanaged-devices=interface-name:wlanXX,interface-name:wlanYY
```
for all wifi cards used by wifibroadcast

QiXuanWang, 2019-01-30:
Thanks for the quick reply. I used PX4 too as I double check. I remember ardupilot 3.6.x won't work for 115200 somehow with mavproxy.
For mavlink-router config, yes I did have them both. I currently set baudrate to 115200 but I'll try 1500000 too.
I don't have NetworkManager since my RPI is ruuning raspbian stretch I guess NM is removed by default.

The acutal question is how my program could get involved in this link:
[autopilot] <--(uart)-->[mavlink_router]-->(udp 14550)-->[wfb_drone]--...-(wifi)-...[wfb_gs]--->(udp 14550)-->[QGrondControl]
But as I double check mavlink-router I think I made some mistakes in understanding. I was hoping for some API to MavProxy but it seems I have to use Mavlink-router and build everthing on that myself?

Thanks again for your patience for all the questions. Great project and it's really amazing as I read through the codes.

svpcom, 2019-01-30:
MavProxy is too slow for 1.5M baudrate (it is pure python-based vs C-based mavlink router) 
To send/receive mavlink messages you can use python mavlink api  (pymavlink) https://mavlink.io/en/mavgen_python
To connect your program to mavlink router just add **another endpoint** for it. See mavlink router docs for reference

QiXuanWang, 2019-01-30:
Thanks. This solved my question exactly. 
Please close the issue as I have no further question regarding this. 

I feel the knowledge involved is quite broad. As a hobby it really takes a lot of time and efforts :) 
Amazing work again.

QiXuanWang, 2019-02-02:
It's not related to this project but I'm a little confused after setting things up:
mavlink-router config:
[UartEndpoint PX4]
Device = /dev/ttyAMA0
Baud = 921600
[UdpEndpoint MyApp]
Mode = Normal
Address = 127.0.0.1
Port = 11000

I tried to connect to UAV(PX4) and do some tweaking with below code:
```
mav = mavutil.mavlink_connection('udpin:127.0.0.1:11000') 
mav.wait_heartbeat()
# below fails
mav.set_mode(mode_id)
```
some how the set_mode doesn't work. I tried several variants and none works and mavlink router reported:
"1 messages to unknown endpoints in the last 5 seconds"
As I searched, some one mentioned the correct connection string should be something "udpout:0.0.0.0：some_port".

Any idea what should be done to fix this?
Thanks for any idea.

QiXuanWang, 2019-02-02:
I tried to use mavlink-router/examples/arm.py by "python arm.py 127.0.0.1:14550" on RX and same "1 messages to unknown endpoints in the last 5 seconds" printed on TX. 
It seems the message is not correctly parsed?

QiXuanWang, 2019-02-04:
Thanks. Later it seems using earlier version of arducopter works for me. Don't really know hy though.