svpcom/wfb-ng #31: Some trouble setting up the system

Davidsastresas, 2019-02-15:
Hello, I've read all the information I could find about setting up this project, and I have tried many times to set it up, from the prebuilt binaries, building them myself, but I am still doing something wrong, and I would really appreciate some help.

I am using 2 raspberry pi 3b hardware, with another pi for video encoding ( I plan to make some video processing on the encoding pi, that is why I separate them ). Those are the steps I am taking:

1-I take the /boot folder from ezwifibroadcast and merge it on to a fresh raspbian image. That should change the kernel with the appropriate drivers right? 

2-I install the python stuff, and untar the prebuilt binaries as stated on the release notes.

3- Then I generate encryption keys on my desktop ubuntu, and I place them as acorded on HOW TO page on the wiki, on /etc/ of each rpi3

4-I create /etc/wifibroadcast.cfg on each rpi3, according to the HOW TO page of the wiki.

5-I run systemctl daemon-reload and start each module on each pi, and seems to be all right. I put on .bashrc the comands for starting the modules on both pi.

6-wfb-cli throws this:
Usage: /usr/bin/wfb-cli <profile>
I don't know if it seems right. This should throw debug information about the working state of the system right?

Maybe what I am missing is trivial but my knowledge on linux embedded is not as good as I would like it to be.

I wouldn't mind to write a step by step guide for beginners for setting up this system, and upload some ready to burn images. I could as well complete the documentation of the project, I just need a few hints on how to set it up.

Thank you very much in advance.

Best regards.

svpcom, 2019-02-15:
1. You need to copy `/lib/modules/<kernel_name>` too.
5. You need to run `systemctl enable wifibroadcast@gs` or `systemctl enable wifibroadcast@drone`  only once and they will be started by systemd on every boot. Don't add them to .bashrc
6.  ``wfb-cli gs`` where gs is profile name for ground station. Run this on GS only for realtime monitoring.

svpcom, 2019-02-15:
Updated tutorial for PX4:  http://dev.px4.io/en/qgc/video_streaming_wifi_broadcast.html

Davidsastresas, 2019-02-15:
Thank you very much, I will try again next week.

Thanks again.

Best regards.

Davidsastresas, 2019-02-25:
I did it again as you told me, and wfb-cli gs is outputing

│Connection failed: Connection was refused by other side: 111: Connection refused.                      │

on both video and telem. Any hint on how can I find more information about the problem and debug? 

Thank you very much for your time.

closed 2019-04-23 by svpcom