svpcom/wfb-ng #33: Too many UDP packets dropped during the streaming

lucaspenna00, 2019-03-22:
Hello everyone,

I'm working on a image streaming project in my drone and I had a trouble:
too many packets dropped during the streaming. 
Unfortunately, I tested the same wifibroadcast setup on my friend PC and the streaming worked well (literally I copied the same wifibroadcast directory to my friend PC)
The image quality on my friend PC is a lot better than my PC. We noticed the quality was improved on his PC when we added the wlp3s0 internal antenna on the transmission script. But the same doesn't occur on my PC, when I add the internal antenna (wlp2s0) to the transmission (further the external antenna that I'm using to stream).

The HW setup is:

TX: alfa AWUS036NH with 9dB antenna
RX: alfa AWUS036NH with 9dB antenna

Both computers running Ubuntu 16.04

WiFi card specs on my PC:

                description: Wireless interface
                product: QCA6174 802.11ac Wireless Network Adapter
                vendor: Qualcomm Atheros
                physical id: 0
                bus info: pci@0000:02:00.0
                logical name: wlp2s0
                version: 32
                serial: d8:0f:99:bf:ee:dd
                width: 64 bits
                clock: 33MHz
                capabilities: bus_master cap_list logical wireless
                configuration: broadcast=yes driver=ath10k_pci driverversion=4.15.0-46-generic firmware=WLAN.RM.4.4.1-00079-QCARMSWPZ-1 latency=0 link=yes multicast=yes wireless=IEEE 802.11

WiFi card specs on my friend PC:

*-network
                description: Wireless interface
                product: QCA9377 802.11ac Wireless Network Adapter
                vendor: Qualcomm Atheros
                physical id: 0
                bus info: pci@0000:03:00.0
                logical name: wlp3s0
                version: 31
                serial: 98:83:89:94:46:31
                width: 64 bits
                clock: 33MHz
                capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
                configuration: broadcast=yes driver=ath10k_pci driverversion=4.15.0-46-generic firmware=WLAN.TF.1.0-00267-1 ip=192.168.42.28 latency=0 link=yes multicast=yes wireless=IEEE 802.11
                resources: irq:127 memory:ef200000-ef3fffff

Output during the streaming on my PC:

8 packets lost
13 packets lost
1086331	ANT	1	102:-55:-48:-45
1086331	PKT	103:0:103:50:44:0
13 packets lost
8 packets lost
12 packets lost
16 packets lost
1087331	ANT	1	136:-47:-45:-43
1087331	PKT	137:0:137:64:49:0
14 packets lost
8 packets lost
8 packets lost
8 packets lost
1088331	ANT	1	128:-47:-45:-45
1088331	PKT	128:0:128:74:38:0
30 packets lost
7 packets lost
10 packets lost
7 packets lost
2 packets lost
7 packets lost
1089331	ANT	1	132:-47:-45:-45
1089331	PKT	133:0:133:57:63:0
8 packets lost
8 packets lost
24 packets lost
16 packets lost
1090331	ANT	1	123:-47:-44:-43
1090331	PKT	124:0:124:55:56:0
15 packets lost
6 packets lost
16 packets lost
33 packets lost
1091331	ANT	1	105:-49:-46:-43
1091331	PKT	105:0:105:35:70:0
23 packets lost
8 packets lost
15 packets lost
1092331	ANT	1	113:-51:-47:-45
1092331	PKT	114:0:114:57:46:0
15 packets lost
7 packets lost
17 packets lost
16 packets lost
1093331	ANT	1	129:-49:-46:-45
1093331	PKT	130:0:130:47:55:0

Sorry for this generic question, but I didn't understand what is happening with my wifibroadcast streaming.

svpcom, 2019-03-22:
Do you use ath10k driver? As I know it doesn't support packet injection.  Check the following:
1. Disable any firewall
2. Disable NetworkManager or add your card to /etc/NetworkManager/NetworkManager.conf
```
[keyfile]
unmanaged-devices=interface-name:wlp2s0
```
3. Your cards are different: QCA6174 vs QCA9377. Try to use the same card on both PC to compare results.

lucaspenna00, 2019-03-27:
`sudo systemctl stop NetworkManager ## disabling wifi interferences
sudo ufw disable ## disabling firewall
sleep 2
cd wifibroadcast
sudo ./scripts/rx_standalone.sh wlx00c0ca4aafe0 wlp2s0
`
I followed your recommendation and I ran the transmission script that contains the above instructions, however the packets are being dropped still;

Furthermore, I added to NetworkManager.conf and tried to run the script. But it's still losing too many packets.

Then I tried to test run rx_standalone script just with wlp2s0 antenna, but I noticed the streaming didn't work (it just got 0).

Is it possible this is a hardware limitation problem or my Wi-Fi card doens't work well with wifibroadcast?

lucaspenna00, 2019-03-27:
I don't know if it helps to understand the problem, but I ran tcpdump and got this output:

16:28:28.820358 IP localhost > localhost: ICMP localhost udp port 5600 unreachable, length 556
16:28:28.840378 Null Supervisory, Receiver not Ready, rcv seq 0, Flags [Command], length 382
16:28:28.841359 Null Supervisory, Receiver not Ready, rcv seq 0, Flags [Command], length 691
16:28:28.841688 Null Supervisory, Receiver not Ready, rcv seq 0, Flags [Command], length 651
16:28:28.844356 IP localhost.41144 > localhost.5600: UDP, length 310
16:28:28.844398 IP localhost > localhost: ICMP localhost udp port 5600 unreachable, length 346
16:28:28.871064 Null Information, send seq 9, rcv seq 0, Flags [Command], length 289
	0x0000:  0000 1200 2e48 0000 0002 8a09 a000 c101  .....H..........
	0x0010:  0000 8000 0000 ffff ffff ffff 50c7 bf0f  ............P...
	0x0020:  d29b 50c7 bf0f d29b 6095 8041 e1e6 eb01  ..P.....`..A....
	0x0030:  0000 6400 3100 0004 4369 7469 0108 8284  ..d.1...Citi....
	0x0040:  8b96 0c12 1824 0301 0605 0400 0100 002a  .....$.........*
	0x0050:  0100 3204 3048 606c 2d1a ee11 1bff ffff  ..2.0H`l-.......
	0x0060:  0000 0000 0000 0000 0001 0000 0000 0000  ................
	0x0070:  0000 0000 3d16 0607 1700 0000 0000 0000  ....=...........
	0x0080:  0000 0000 0000 0000 0000 0000 4a0e 1400  ............J...
	0x0090:  0a00 2c01 c800 1400 0500 1900 7f08 0100  ..,.............
	0x00a0:  0000 0000 0040 dd18 0050 f202 0101 0000  .....@...P......
	0x00b0:  03a4 0000 27a4 0000 4243 5e00 6232 2f00  ....'...BC^.b2/.
	0x00c0:  dd09 0003 7f01 0100 00ff 7f30 1801 0000  ...........0....
	0x00d0:  0fac 0202 0000 0fac 0400 0fac 0201 0000  ................
	0x00e0:  0fac 0200 00dd 1a00 50f2 0101 0000 50f2  ........P.....P.
	0x00f0:  0202 0000 50f2 0400 50f2 0201 0000 50f2  ....P...P.....P.
	0x0100:  027f 0400 0000 02dd 1800 50f2 0410 4a00  ..........P...J.
	0x0110:  0110 1044 0001 0210 4900 0600 372a 0001  ...D....I...7*..
	0x0120:  20                                       .
16:28:28.881439 Null Supervisory, Receiver not Ready, rcv seq 0, Flags [Command], length 706
16:28:28.881736 Null Supervisory, Receiver not Ready, rcv seq 0, Flags [Command], length 712
16:28:28.882084 Null Supervisory, Receiver not Ready, rcv seq 0, Flags [Command], length 772
16:28:28.882466 Null Supervisory, Receiver not Ready, rcv seq 0, Flags [Command], length 623
16:28:28.882927 Null Supervisory, Receiver not Ready, rcv seq 0, Flags [Command], length 961
16:28:28.883453 Null Supervisory, Receiver not Ready, rcv seq 0, Flags [Command], length 961
16:28:28.883866 Null Supervisory, Receiver not Ready, rcv seq 0, Flags [Command], length 961
16:28:28.884370 Null Supervisory, Receiver not Ready, rcv seq 0, Flags [Command], length 961
16:28:28.884385 IP localhost.41144 > localhost.5600: UDP, length 619
16:28:28.884422 IP localhost > localhost: ICMP localhost udp port 5600 unreachable, length 556
16:28:28.884448 IP localhost.41144 > localhost.5600: UDP, length 634
16:28:28.884461 IP localhost > localhost: ICMP localhost udp port 5600 unreachable, length 556
16:28:28.884478 IP localhost.41144 > localhost.5600: UDP, length 700
16:28:28.884490 IP localhost > localhost: ICMP localhost udp port 5600 unreachable, length 556
^C
625 packets captured
901 packets received by filter
30 packets dropped by kernel

svpcom, 2019-03-28:
Try to change wifi channel and/or band (2.4 or 5.8). You need to run tcpdump on radio card on wlp2s0 not on localhost:  ``tcpdump -i wlp2s0 -nn -p``

closed 2021-02-06 by svpcom