svpcom/wfb-ng #34: Help setting up

Howie9600, 2019-03-29:
Hello

I have setup two raspberry pis with AWUS036ACH.  I have the cards working and in monitor mode.  When running the TX side shows this

● wifibroadcast@drone.service - WFB profile drone
   Loaded: loaded (/lib/systemd/system/wifibroadcast@.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-03-29 08:18:41 UTC; 19s ago
 Main PID: 1205 (python)
   CGroup: /system.slice/system-wifibroadcast.slice/wifibroadcast@drone.service
           ├─1205 /usr/bin/python -m telemetry.server drone wlx00c0ca961470
           ├─1269 /usr/bin/wfb_rx -p 1 -u 14601 -K /etc/drone.key wlx00c0ca961470
           ├─1271 /usr/bin/wfb_tx -p 2 -u 14600 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 wlx00c0ca961470
           └─1273 /usr/bin/wfb_tx -p 3 -u 5602 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 wlx00c0ca961470

Mar 29 08:18:45 navio python[1205]: 2019-03-29 08:18:45+0000 [-] Telem RX: /usr/bin/wfb_rx -p 1 -u 14601 -K /etc/drone.key wlx00c0ca961470
Mar 29 08:18:45 navio python[1205]: 2019-03-29 08:18:45+0000 [-] Telem TX: /usr/bin/wfb_tx -p 2 -u 14600 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 wlx00c0ca961470
Mar 29 08:18:45 navio python[1205]: 2019-03-29 08:18:45+0000 [-] Started telem rx
Mar 29 08:18:45 navio python[1205]: 2019-03-29 08:18:45+0000 [-] Started telem tx
Mar 29 08:18:45 navio python[1205]: 2019-03-29 08:18:45+0000 [-] Listen for video stream 3 on 0.0.0.0:5602
Mar 29 08:18:45 navio python[1205]: 2019-03-29 08:18:45+0000 [-] Started video tx
Mar 29 08:18:45 navio python[1205]: 2019-03-29 08:18:45+0000 [-] Video: /usr/bin/wfb_tx -p 3 -u 5602 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 wlx00c0ca961470
Mar 29 08:18:45 navio python[1205]: 2019-03-29 08:18:45+0000 [-] telem tx: Listen on 14600 for wlx00c0ca961470
Mar 29 08:18:45 navio python[1205]: 2019-03-29 08:18:45+0000 [-] video tx: Listen on 5602 for wlx00c0ca961470
Mar 29 08:18:45 navio python[1205]: 2019-03-29 08:18:45+0000 [-] telem rx: wlx00c0ca961470 has DLT_IEEE802_11_RADIO Encap

the RX side

● wifibroadcast@gs.service - WFB profile gs
   Loaded: loaded (/lib/systemd/system/wifibroadcast@.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-03-29 08:08:48 UTC; 15s ago
 Main PID: 1005 (python)
   CGroup: /system.slice/system-wifibroadcast.slice/wifibroadcast@gs.service
           ├─1005 /usr/bin/python -m telemetry.server gs wlx00c0ca96146c
           ├─1052 /usr/bin/wfb_rx -p 2 -u 14600 -K /etc/gs.key wlx00c0ca96146c
           ├─1054 /usr/bin/wfb_tx -p 1 -u 14601 -K /etc/gs.key -B 20 -G long -S 1 -L 0 -M 1 wlx00c0ca96146c
           └─1060 /usr/bin/wfb_rx -p 3 -c 127.0.0.1 -u 5600 -K /etc/gs.key wlx00c0ca96146c

Mar 29 08:08:51 navio python[1005]: 2019-03-29 08:08:51+0000 [-] AntennaFactory starting on 8001
Mar 29 08:08:51 navio python[1005]: 2019-03-29 08:08:51+0000 [-] Started telem rx
Mar 29 08:08:51 navio python[1005]: 2019-03-29 08:08:51+0000 [-] Started telem tx
Mar 29 08:08:51 navio python[1005]: 2019-03-29 08:08:51+0000 [-] Send video stream 3 to 127.0.0.1:5600
Mar 29 08:08:51 navio python[1005]: 2019-03-29 08:08:51+0000 [-] AntennaFactory starting on 8002
Mar 29 08:08:51 navio python[1005]: 2019-03-29 08:08:51+0000 [-] Started video rx
Mar 29 08:08:51 navio python[1005]: 2019-03-29 08:08:51+0000 [-] Video: /usr/bin/wfb_rx -p 3 -c 127.0.0.1 -u 5600 -K /etc/gs.key wlx00c0ca96
Mar 29 08:08:51 navio python[1005]: 2019-03-29 08:08:51+0000 [-] telem tx: Listen on 14601 for wlx00c0ca96146c
Mar 29 08:08:51 navio python[1005]: 2019-03-29 08:08:51+0000 [-] video rx: wlx00c0ca96146c has DLT_IEEE802_11_RADIO Encap
Mar 29 08:08:51 navio python[1005]: 2019-03-29 08:08:51+0000 [-] telem rx: wlx00c0ca96146c has DLT_IEEE802_11_RADIO Encap

i have data from each modem showing on Wireshark and kismet.  But they are not talking.  No data is flowing so to speak.

wfb-cli gs shows this 
xRX PKT:   recv 0 d_ok 0 fec_r 0 lost 0 d_err 0 bad 0                                                                                      
xRX PKT/s: recv 0 d_ok 0 fec_r 0 lost 0 d_err 0 bad 0 

I have the autopilot outputting data on 127.0.0.1:14550 its a navio 2 so outputs direct IP data.   

What am i doing wrong??

svpcom, 2019-04-22:
Try to use latest release which has support of IPoverWB. Then check connectivity via ping. If works - check mavlink settings.

Howie9600, 2019-04-26:
it might have been the units were drawing too much power and resetting.  I will try with lesser power units.

i know it maybe stupid to ask but once i compile and run are all the setting default correct to send the mavlink data?  or do i need to change the ip ranges?  sorry i am new to this and i need a basic step by step guide.

svpcom, 2019-04-26:
Please explain your setup. 
By default drone listen on 14560 (see telemetry/conf/master.cfg). To listen on 14550 add to /etc/wifibroadcast.cfg:
```
[drone_mavlink]
peer = 'listen://0.0.0.0:14550'   # incoming connection
```

closed 2021-02-06 by svpcom