svpcom/wfb-ng #35: distributed operations

brealin, 2019-04-09:
How to implement distributed operation, what is the module “wifibroadcast_agg” in the figure? There are no more detailed documentation. thank you for your reply!

svpcom, 2019-04-10:
https://github.com/svpcom/wifibroadcast/wiki/wfb_tx-and-wfb_rx-reference
You should setup ``wfb_rx`` as remote forwarder on nodes, that own wifi cards and setup ``wfb_rx`` as remote aggregator on central node. But in this setup all data transfers will be unidirectional, because current version of python middelware (which select tx adapter in bi-directional mode) supports only local wifi cards.

brealin, 2019-04-15:
thank you for your reply！
I found two problems in the test:
1. Regardless of whether I use HT20 or HT40, the UDP rate can never exceed 22Mbps (MCS5, MCS6, MCS7 rate does not change)
2.MCS8-MCS15 cannot be used (my module is 2*2 mimo wireleass ap)

I use iperf2.x test, connection method: 
PC1===>ap1(tx192.168.166.11)--------->ap2(rx192.168.166.12)====>PC2.

PC1: iperf -u -c 192.168.166.11 -p 5600 -b 50M -i 1
PC2: iperf -s -u -p 5600 -i 1 (AP2 forwards UDP to PC2)
Do you have any suggestions I can test again?

brealin, 2019-04-15:
By the way, when I connect two APs using AP+STA, their UDP rate can reach more than 95Mbps.

svpcom, 2019-04-23:
1. Iperf2 is not very good tool for udp performance test. You need to check that it really send UDP packets with size close to MTU. Also check that there are no collisions due to interference with other wifi stations.
2. MIMO is not suitable for WFB (too long to describe it here)

closed 2021-02-06 by svpcom