svpcom/wfb-ng #36: RTL8814 (AWUS1900) No link

chocol4te, 2019-05-13:
The only two adapters I have are a TP-Link T4U (RTL8812AU) and an Alfa AWUS1900 (RTL8814AU). I am testing on two Raspberry Pi 3B's, both using @svpcom's https://github.com/svpcom/rtl8812au driver. I was getting "No link" in `wfb-cli` so I followed the troubleshooting steps with no luck. `echo test | nc -u 127.0.0.1 5600` blocks on the transmitter, and `tcpdump` only seems to be picking up probe requests from nearby access points. I've tried both devices acting as transmitter and receiver, as well as both on the same machine and on different machines.

Do you have any additional steps I could try, or code modifications that might need to be done?

svpcom, 2019-05-13:
I've had a bad experience with  Alfa AWUS1900 (RTL8814AU). Try with both RTL8812AU. Also try yo use different band or channel (2.4 <-> 5.8 GHz)

chocol4te, 2019-05-15:
I'm now using two RTL8812AU's on one machine, with the rx_standalone script modified to output on port 5601.

I get this output, (5G is the exact same except on channel 149. Swapping wlan1 and wlan2 makes no difference):
```
$ sudo ./scripts/rx_standalone.sh wlan1
Setting wlan1 to channel 6
wlan1 has DLT_IEEE802_11_RADIO Encap
537821	PKT	0:0:0:0:0:0
538822	PKT	0:0:0:0:0:0
539823	PKT	0:0:0:0:0:0
540824	PKT	0:0:0:0:0:0
541825	PKT	0:0:0:0:0:0
542827	PKT	0:0:0:0:0:0
```

```
$ sudo ./scripts/tx_standalone.sh wlan2
Setting wlan2 to channel 6
Listen on 5600 for wlan2
```

No output on port 5601, and `echo test | nc -u 127.0.0.1 5600` blocks.

svpcom, 2019-05-15:
Try latest version from master. It support IP tunnels which send keepalive packets (you don't need to do any "echo | nc" calls for test).  Also use python services for first tests (sh scripts are obsolete)

chocol4te, 2019-05-15:
Done, I'm getting this output:

```
$ sudo ./wfb_tx -K tx.key wlan1
Listen on 5600 for wlan1
```
```
$ sudo ./wfb_rx -K rx.key wlan2
Error: unknown encapsulation on wlan2
```

(No difference when swapping interfaces)

Thanks for your help :)

svpcom, 2019-05-15:
What is shown by ``ethtool -i wlan1`` and ``ethtool -i wlan2`` ?

chocol4te, 2019-05-15:
```
$ ethtool -i wlan1
driver: rtl88xxau
version:
firmware-version:
expansion-rom-version:
bus-info: 1-1.5:1.0
supports-statistics: no
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

$ ethtool -i wlan2
driver: rtl88xxau
version:
firmware-version:
expansion-rom-version:
bus-info: 1-1.2:1.0
supports-statistics: no
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

svpcom, 2019-05-15:
What is version of rtl88xxau driver?  Do ``git branch`` to show

svpcom, 2019-05-15:
And ``modinfo ./88XXau.ko  | grep version``

chocol4te, 2019-05-15:
```
$ git branch
* v5.3.4
```

```
$ modinfo ./88XXau.ko | grep version
version:        v5.3.4_28226.20180614_COEX20171103-6c6d
srcversion:     DC30DE0EB3C235E62E5B42F
vermagic:       4.14.98-v7+ SMP mod_unload modversions ARMv7 p2v8
parm:           rtw_chip_version:int
```

svpcom, 2019-05-15:
This version is not compatible with WFB. Use v5.2.20 from  https://github.com/svpcom/rtl8812au

svpcom, 2019-05-15:
``git checkout v5.2.20``

chocol4te, 2019-05-15:
🤦‍♂ my bad, building now

chocol4te, 2019-05-15:
Okay, I've installed the correct version.

```
$ modinfo ./88XXau.ko | grep version
version:        v5.2.20.2_28373.20180619
srcversion:     2371CE7C983AA2C18EBE8C6
vermagic:       4.14.98-v7+ SMP mod_unload modversions ARMv7 p2v8
parm:           rtw_chip_version:int
```

```
$ sudo ethtool -i wlan1
driver: rtl88xxau
version:
firmware-version:
expansion-rom-version:
bus-info: 1-1.2:1.0
supports-statistics: no
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

but I'm getting the same error:

```
$ sudo ./wfb_rx -K gs.key wlan1
Error: unknown encapsulation on wlan1
```

svpcom, 2019-05-15:
Use rx_standalone.sh (or code from it to setup wifi card in monitor mode)

svpcom, 2019-05-15:
Or use  ``python -m telemetry.server gs wlan1`` and ``python -m telemetry.server drone wlan2`` (recommended)

chocol4te, 2019-05-15:
```
$ sudo python -m telemetry.server gs wlan1
2019-05-15 18:38:44+0100 [-] Log opened.
2019-05-15 18:38:44+0100 [-] # iw reg set BO
2019-05-15 18:38:44+0100 [-] # ifconfig wlan1 down
2019-05-15 18:38:44+0100 [-] # iw dev wlan1 set monitor otherbss
2019-05-15 18:38:44+0100 [-] # ifconfig wlan1 up
2019-05-15 18:38:45+0100 [-] # iw dev wlan1 set channel 161 HT20
2019-05-15 18:38:45+0100 [-] Connect telem stream 2(RX), 1(TX) to 127.0.0.1:14550
2019-05-15 18:38:45+0100 [-] UDPProxyProtocol starting on 35970
2019-05-15 18:38:45+0100 [-] UDPProxyProtocol starting on 14600
2019-05-15 18:38:45+0100 [-] UDPProxyProtocol starting on 55851
2019-05-15 18:38:45+0100 [-] Telem RX: ./wfb_rx -p 2 -u 14600 -K ./gs.key -k 1 -n 2 wlan1
2019-05-15 18:38:45+0100 [-] Telem TX: ./wfb_tx -p 1 -u 14601 -K ./gs.key -B 20 -G long -S 1 -L 0 -M 1 -k 1 -n 2 wlan1
2019-05-15 18:38:45+0100 [-] AntennaFactory starting on 8001
2019-05-15 18:38:45+0100 [-] Started telem rx
2019-05-15 18:38:45+0100 [-] Started telem tx
2019-05-15 18:38:45+0100 [-] Send video stream 3 to 127.0.0.1:5600
2019-05-15 18:38:45+0100 [-] AntennaFactory starting on 8002
2019-05-15 18:38:45+0100 [-] Started video rx
2019-05-15 18:38:45+0100 [-] Video: ./wfb_rx -p 3 -c 127.0.0.1 -u 5600 -K ./gs.key -k 8 -n 12 wlan1
2019-05-15 18:38:45+0100 [-] UDPProxyProtocol starting on 14800
2019-05-15 18:38:45+0100 [-] UDPProxyProtocol starting on 32941
2019-05-15 18:38:45+0100 [-] Tunnel RX: ./wfb_rx -p 5 -u 14800 -K ./gs.key -k 1 -n 2 wlan1
2019-05-15 18:38:45+0100 [-] Tunnel TX: ./wfb_tx -p 4 -u 14801 -K ./gs.key -B 20 -G long -S 1 -L 0 -M 1 -k 1 -n 2 wlan1
2019-05-15 18:38:45+0100 [-] AntennaFactory starting on 8003
2019-05-15 18:38:45+0100 [-] Started tunnel rx
2019-05-15 18:38:45+0100 [-] Started tunnel tx
2019-05-15 18:38:45+0100 [-] telem tx: Listen on 14601 for wlan1
2019-05-15 18:38:45+0100 [-] telem rx: wlan1 has DLT_IEEE802_11_RADIO Encap
2019-05-15 18:38:45+0100 [-] video rx: wlan1 has DLT_IEEE802_11_RADIO Encap
2019-05-15 18:38:45+0100 [-] tunnel tx: Listen on 14801 for wlan1
2019-05-15 18:38:45+0100 [-] tunnel rx: wlan1 has DLT_IEEE802_11_RADIO Encap
```

```
$ sudo python -m telemetry.server drone wlan2
2019-05-15 18:39:23+0100 [-] Log opened.
2019-05-15 18:39:23+0100 [-] # iw reg set BO
2019-05-15 18:39:23+0100 [-] # ifconfig wlan2 down
2019-05-15 18:39:23+0100 [-] # iw dev wlan2 set monitor otherbss
2019-05-15 18:39:23+0100 [-] # ifconfig wlan2 up
2019-05-15 18:39:23+0100 [-] # iw dev wlan2 set channel 161 HT20
2019-05-15 18:39:23+0100 [-] Listen for telem stream 1(RX), 2(TX) on 0.0.0.0:14560
2019-05-15 18:39:23+0100 [-] UDPProxyProtocol starting on 14560
2019-05-15 18:39:23+0100 [-] UDPProxyProtocol starting on 14700
2019-05-15 18:39:23+0100 [-] UDPProxyProtocol starting on 42107
2019-05-15 18:39:23+0100 [-] Telem RX: ./wfb_rx -p 1 -u 14700 -K ./drone.key -k 1 -n 2 wlan2
2019-05-15 18:39:23+0100 [-] Telem TX: ./wfb_tx -p 2 -u 14701 -K ./drone.key -B 20 -G long -S 1 -L 0 -M 1 -k 1 -n 2 wlan2
2019-05-15 18:39:23+0100 [-] Started telem rx
2019-05-15 18:39:23+0100 [-] Started telem tx
2019-05-15 18:39:23+0100 [-] Listen for video stream 3 on 0.0.0.0:5602
2019-05-15 18:39:23+0100 [-] Started video tx
2019-05-15 18:39:23+0100 [-] Video: ./wfb_tx -p 3 -u 5602 -K ./drone.key -B 20 -G long -S 1 -L 0 -M 1 -k 8 -n 12 wlan2
2019-05-15 18:39:23+0100 [-] UDPProxyProtocol starting on 14900
2019-05-15 18:39:23+0100 [-] UDPProxyProtocol starting on 58327
2019-05-15 18:39:23+0100 [-] Tunnel RX: ./wfb_rx -p 4 -u 14900 -K ./drone.key -k 1 -n 2 wlan2
2019-05-15 18:39:23+0100 [-] Tunnel TX: ./wfb_tx -p 5 -u 14901 -K ./drone.key -B 20 -G long -S 1 -L 0 -M 1 -k 1 -n 2 wlan2
2019-05-15 18:39:23+0100 [-] Started tunnel rx
2019-05-15 18:39:23+0100 [-] Started tunnel tx
2019-05-15 18:39:23+0100 [-] video tx: Listen on 5602 for wlan2
2019-05-15 18:39:23+0100 [-] telem tx: Listen on 14701 for wlan2
2019-05-15 18:39:23+0100 [-] telem rx: wlan2 has DLT_IEEE802_11_RADIO Encap
2019-05-15 18:39:23+0100 [-] tunnel tx: Listen on 14901 for wlan2
2019-05-15 18:39:23+0100 [-] tunnel rx: wlan2 has DLT_IEEE802_11_RADIO Encap
```

chocol4te, 2019-05-15:
Nothing happens when the other side is started, is it working?

svpcom, 2019-05-15:
Run `python -m telemetry.cli gs` in other console

svpcom, 2019-05-15:
Tunnel link should not be in state "link lost"

chocol4te, 2019-05-16:
[image]

Unfortunately it is :(

chocol4te, 2019-05-16:
I left it for a bit, and sometimes it flickers and a few packets are maybe sent?

[image]

svpcom, 2019-05-16:
It seems that it works (session detected) but something prevent normal operation. Try this:
1. Change `wifi_channel` from 165 to 1 in config file (some cards are incompatible with 5GHz band)
2. Check that NetworkManager is disabled or your wifi cards are blacklisted in its config. (``unmanaged-devices=interface-name:wlan1,interface-name:wlan2``)
3. Check that wifi cards have enough power (RPI has too weak usb ports for high-power wifi cards). Check them on linux PC or use external +5V (or powered USB hub) for wifi cards.

chocol4te, 2019-05-16:
I changed wifi_channel in `telemetry/conf/master.cfg`, still link lost.

NetworkManager is not installed, and neither wifi card is in /etc/network/interfaces

I'm trying it on my Ubuntu box now, but it doesn't look like it's working here either.

chocol4te, 2019-05-16:
Okay, now I'm having a few issues building the driver, may I ask what kernel version you are using?

svpcom, 2019-05-16:
I use 4.14.52 on NanoPI NEO2 and 4.18.20-100.fc27.x86_64 on PC

svpcom, 2019-05-16:
Try to use powered USB hub in case of RPI

chocol4te, 2019-05-16:
Progress!!

[image]

I can start both the gs and drone, and the tunnel works! However, when I listened to port 5600 and send "test" over port 5602, it successfully sent the text but then I got link lost everywhere. Restarting just the drone didn't help, I had to restart both before it would establish a tunnel again.

svpcom, 2019-05-16:
"Link lost" message shown if there are no packets in channel during 1 second. So you need to emit heartbeat packets if no traffic available. (This is only for video and mavlink streams, for tunnel hearbeart generation is builtin)

svpcom, 2019-05-16:
But this is strange that sending one packet into mavlink stream broke IP tunnel. This should not be happened

svpcom, 2019-05-16:
Don't forget to add
```
[keyfile]
unmanaged-devices=interface-name:wlan1,interface-name:wlan2
```
to /etc/NetworkManager/NetworkManager.conf

svpcom, 2019-05-16:
For test video you can use:
```
gst-launch-1.0 videotestsrc ! videoconvert ! video/x-raw,format=NV12,framerate=30/1,width=1280,height=720 ! clockoverlay ! x264enc tune=zerolatency bitrate=4000 ! rtph264pay ! udpsink port=5602
```
and
```
gst-launch-1.0 udpsrc do-timestamp=true port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' ! rtph264depay ! avdec_h264 ! xvimagesink
```

chocol4te, 2019-05-16:
I've added that, but I'm getting a lot of lost packets.

I can get semi-stable video if I reduce the resolution down to 240p, but the packet loss is still around 50%

[image]

svpcom, 2019-05-16:
Try on real hardware. It seems that is virtualbox issues

chocol4te, 2019-05-16:
I still haven't had success on real hardware, but I tried Ubuntu Desktop 18.04, and it works! Full 720p stream, no visual issues, and only 1 or 2 packets dropped every few seconds :D

[image]

closed 2019-05-16 by chocol4te