svpcom/wfb-ng #38: Limit time packet is held for CCA?

FLYBYME, 2019-06-04:
Any way to limit the time the packet is held in the cards buffer as it waited for CCA?

I would like to be able to set an upper limit say of 100ms and if the frame cant be sent in that time then it should be dropped(sender side). 

Im not 100% sure on how 802.11 deals with this but im sure that the card would drop packets if the buffer is full so maybe so maybe it can do it on a time limit.

I would rather drop the packet then have it come in late.

svpcom, 2019-06-05:
This is regulated by card driver and/or firmware. Some card (for example atheros ath9k) has ability to load custom (open) firmware and control this. For 8812au I disable packet retransmission in the driver if RADIOTAP_F_TX_NOACK flag set in radiotap header.
```
//core/rtw_xmit.c:
pattrib->retry_ctrl = (txflags & 0x08)?_FALSE:_TRUE;
```

FLYBYME, 2019-06-05:
Interesting I am running 8812au card with the aircrack driver so I will give it a try.

When you talk about re-transmission with WFB it should never receive an ACK so does the card re-transmit every packet?

svpcom, 2019-06-05:
There are a two types of retransmits - no ack and busy medium. There are no public datasheets for 88xxau  so I can only guess what is mean there

closed 2021-02-06 by svpcom