svpcom/wfb-ng #39: Base Image

Howie9600, 2019-06-12:
Hello.  I have been having real problems with making all of this work.

Can anyone who has made it work help with a rough guide.

What base image did you use?  What hardware?

I am having real trouble getting everything to work.  Can anyone help at all?

svpcom, 2019-06-12:
I use [NanoPI NEO2](https://www.friendlyarm.com/index.php?route=product/product&product_id=180) with [Armbian stretch](https://www.armbian.com/nanopi-neo-2/) and [h264 encoding camera](https://www.kurokesu.com/shop/CAMUSB1)
See installation instructions in the [WIKI](https://github.com/svpcom/wifibroadcast/wiki/Setup-HOWTO) and [px4-devguide](https://dev.px4.io/en/qgc/video_streaming_wifi_broadcast.html)

Howie9600, 2019-06-12:
i ordered two yesterday and they literally arrived 10 mins ago.

thank you i will try it all again now.

Howie9600, 2019-06-12:
i have followed all the instructions and i am getting the current error.  Any help would be great

root@nanopineocore2:/# systemctl status wifibroadcast@gs
* wifibroadcast@gs.service - WFB profile gs
   Loaded: loaded (/lib/systemd/system/wifibroadcast@.service; disabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-06-12 14:33:40 UTC; 11ms ago
 Main PID: 3115 (python)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/system-wifibroadcast.slice/wifibroadcast@gs.service
           `-3115 /usr/bin/python -m telemetry.server gs wlx24050ff7306a

Jun 12 14:33:40 nanopineocore2 systemd[1]: Started WFB profile gs.
root@nanopineocore2:/# systemctl status wifibroadcast@gs
* wifibroadcast@gs.service - WFB profile gs
   Loaded: loaded (/lib/systemd/system/wifibroadcast@.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2019-06-12 14:33:40 UTC; 37ms ago
  Process: 3115 ExecStart=/usr/bin/python -m telemetry.server gs ${WFB_NICS} (code=exited, status=1/FAILURE)
 Main PID: 3115 (code=exited, status=1/FAILURE)

svpcom, 2019-06-12:
journalctl -xu wifibroadcast@gs

Howie9600, 2019-06-12:
thank you so much.  i got to see the error and now it is working.  How did you get the rpi camera to send data through the link?  is it ok if i post a basic step by step guide?   well its what i did so it may help

svpcom, 2019-06-12:
Systemd service file for raspbian

fpv-camera.service:
```
[Unit]
Description=FPV camera
After=network-online.target

[Service]
ExecStart=/bin/sh -c "raspivid --nopreview --awb auto -ih -t 0 -w 1920 -h 1080 -fps 30 -b 4000000 -g 30 -pf high -o - | gst-launch-1.0 fdsrc ! h264parse !  rtph264pay !  udpsink host=YOUR_WFB_IP_ADDRESS port=YOUR_WFB_PORT"
Type=simple
Restart=always
RestartSec=1s
TimeoutStopSec=10s

[Install]
WantedBy=multi-user.target
```

svpcom, 2019-06-12:
Step-by-step guide will be ok - I'll add it to the wiki

Howie9600, 2019-06-12:
i will write it an send it to you.  

I had it working and rebooted and am now getting 

ImportError: cannot import name IPRoute

i have pyroute2 installed but it still gives this error

svpcom, 2019-06-12:
try to install pyroute2-python2 (or something like this).

svpcom, 2019-06-12:
python2 -m pydoc pyroute2

Howie9600, 2019-06-12:
so i have fixed that error and both tx and rx are running.  I try to test and get this error

telemetry.config_parser.ConfigError: 'ascii' codec can't decode byte 0xe2 in position 76: ordinal not in range(128)

is it something i have done wrong?

svpcom, 2019-06-13:
Remove non-ascii symbols from config file (for example unicode dash and quotes). Test it with `iconv -f ascii -t utf8 /etc/wifibroadcast.cfg`

Howie9600, 2019-06-13:
it is only while running the wfb_cli test.  the code doesnt have the issue normally

Howie9600, 2019-06-13:
ok.  so that fixed the issue but i am seeing link lost on all three connections.  i cant ping 10.5.0.2 or 10.5.0.1.

for the test to work is there anything i am missing?  both scripts are stable and the cards have been configure to right channel and power

svpcom, 2019-06-13:
Check frequency channel. Try to switch from 5.8 to 2.4 GHz. Upload your config files and logs (from journalctl)

svpcom, 2019-06-13:
What cards do you use on the both sides?

Howie9600, 2019-06-13:
hello we have a two realtek cards

-- Logs begin at Thu 2019-06-13 13:03:40 UTC, end at Thu 2019-06-13 14:39:52 UTC. --
Jun 13 14:39:44 nanopineocore2 systemd[1]: Started WFB profile drone.
-- Subject: Unit wifibroadcast@drone.service has finished start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit wifibroadcast@drone.service has finished starting up.
--
-- The start-up result is done.
Jun 13 14:39:49 nanopineocore2 python[2989]: 2019-06-13 14:39:49+0000 [-] Log opened.
Jun 13 14:39:49 nanopineocore2 python[2989]: 2019-06-13 14:39:49+0000 [-] # iw reg set BO
Jun 13 14:39:49 nanopineocore2 python[2989]: 2019-06-13 14:39:49+0000 [-] # ifconfig wlx24050ff7320c down
Jun 13 14:39:49 nanopineocore2 python[2989]: 2019-06-13 14:39:49+0000 [-] # iw dev wlx24050ff7320c set monitor otherbss
Jun 13 14:39:49 nanopineocore2 python[2989]: 2019-06-13 14:39:49+0000 [-] # ifconfig wlx24050ff7320c up
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] # iw dev wlx24050ff7320c set channel 11 HT20
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] # iw dev wlx24050ff7320c set txpower fixed 58
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] Listen for telem stream 1(RX), 2(TX) on 0.0.0.0:14550
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] UDPProxyProtocol starting on 14550
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] UDPProxyProtocol starting on 14700
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] UDPProxyProtocol starting on 56786
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] Telem RX: /usr/bin/wfb_rx -p 1 -u 14700 -K /etc/drone.key -k 1 -n 2 wlx2$
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] Telem TX: /usr/bin/wfb_tx -p 2 -u 14701 -K /etc/drone.key -B 20 -G long $
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] Started telem rx
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] Started telem tx
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] Listen for video stream 3 on 0.0.0.0:5602
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] Started video tx
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] Video: /usr/bin/wfb_tx -p 3 -u 5602 -K /etc/drone.key -B 20 -G long -S 1$
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] UDPProxyProtocol starting on 14900
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] UDPProxyProtocol starting on 43822
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] Tunnel RX: /usr/bin/wfb_rx -p 4 -u 14900 -K /etc/drone.key -k 1 -n 2 wlx$
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] Tunnel TX: /usr/bin/wfb_tx -p 5 -u 14901 -K /etc/drone.key -B 20 -G long$
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] Started tunnel rx
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] Started tunnel tx
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] telem tx: Listen on 14701 for wlx24050ff7320c
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] video tx: Listen on 5602 for wlx24050ff7320c
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] telem rx: wlx24050ff7320c has DLT_IEEE802_11_RADIO Encap
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] tunnel tx: Listen on 14901 for wlx24050ff7320c
Jun 13 14:39:50 nanopineocore2 python[2989]: 2019-06-13 14:39:50+0000 [-] tunnel rx: wlx24050ff7320c has DLT_IEEE802_11_RADIO Encap

-- Logs begin at Thu 2016-11-03 17:16:42 UTC, end at Thu 2019-06-13 14:35:01 UTC. --
Jun 13 12:41:34 nanopineocore2 systemd[1]: Started WFB profile gs.
-- Subject: Unit wifibroadcast@gs.service has finished start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit wifibroadcast@gs.service has finished starting up.
--
-- The start-up result is done.
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] Log opened.
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] # iw reg set BO
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] # ifconfig wlx24050ff7306a down
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] # iw dev wlx24050ff7306a set monitor otherbss
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] # ifconfig wlx24050ff7306a up
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] # iw dev wlx24050ff7306a set channel 11 HT20
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] # iw dev wlx24050ff7306a set txpower fixed 100
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] Connect telem stream 2(RX), 1(TX) to 127.0.0.1:14550
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] UDPProxyProtocol starting on 33214
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] UDPProxyProtocol starting on 14600
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] UDPProxyProtocol starting on 60088
[... cut ...]
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] telem rx: wlx24050ff7306a has DLT_IEEE802_11_RADIO Encap
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] video rx: wlx24050ff7306a has DLT_IEEE802_11_RADIO Encap
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] tunnel tx: Listen on 14801 for wlx24050ff7306a
Jun 13 12:41:38 nanopineocore2 python[1765]: 2019-06-13 12:41:38+0000 [-] tunnel rx: wlx24050ff7306a has DLT_IEEE802_11_RADIO Encap

svpcom, 2019-06-13:
What driver version do you use?

svpcom, 2019-06-13:
Use **v5.2.20** from https://github.com/svpcom/rtl8812au

jeng37, 2019-06-26:
Hi,

I got the same trouble for installing, so i wrote a little howto.
[Wifibroadcast-Setup.txt](https://github.com/svpcom/wifibroadcast/files/3328317/Wifibroadcast-Setup.txt)

Thanks goes to Vasily.

svpcom, 2019-06-26:
I've added it to the wiki: https://github.com/svpcom/wifibroadcast/wiki/Install-from-scratch

Howie9600, 2019-06-26:
That is excellent.   Just a question what base image did you use?  I always have issues with the drivers as they can never work with the kernel of what ever image I use.

jeng37, 2019-06-26:
What device are you using?

svpcom, 2019-06-26:
I use https://dl.armbian.com/nanopineo2/Debian_stretch_next.7z  but with custom kernel from FriendlyArm: http://wiki.friendlyarm.com/wiki/index.php/Building_U-boot_and_Linux_for_H5/H3/H2%2B
Patched drivers for 8812au is here: https://github.com/svpcom/rtl8812au

svpcom, 2019-06-26:
Builded kernel and modules (with patched 88XXau driver) for NEO2:
[linux.tgz.zip](https://github.com/svpcom/wifibroadcast/files/3329556/linux.tgz.zip)
You need to extract it to the image root and symlink Image and dtbs. The rebuild initramfs. See http://wiki.friendlyarm.com/wiki/index.php/Building_U-boot_and_Linux_for_H5/H3/H2%2B for details.

jeng37, 2019-06-26:
I use this to build my own.
https://github.com/armbian/build/blob/c7a9f59293b29020a0e9ed81ef9c28b230015ac0/README.md

Howie9600, 2019-06-26:
I have both pi's (all versions) and nanopineocore2

Tyrrx, 2019-07-31:
@svpcom 
Could you post how to symlink your Kernel and modules binaries and how to rebuild the initramfs, step by step?
I tried it but I dont know where I failed.

svpcom, 2019-07-31:
How to install kernel and modules:
```
cd / && tar xzvf /tmp/linux.tgz
cd /boot
ln -s Image-svpcom Image
ln -s dtb-svpcom dtb
update-initramfs -c -k 4.14.111
```

Tyrrx, 2019-07-31:
@svpcom 
Thank you very much.
Do I need to copy /linux/lib/modules to /lib/modules too?

svpcom, 2019-07-31:
`cd / && tar xzvf /tmp/linux.tgz` will extract modules to /lib/modules automatically

Tyrrx, 2019-07-31:
@svpcom 
I did ln -sf ..... becuase these symlinks already exist.
At the moment the neo2 doesn't boot and the green led flashes twice each secound. red one is off.
Is there something I need to do before I update the initramfs like deleteing the old one or something like that? Or does "update-initramfs -c -k 4.14.111" automaticly overwrites the old one?

svpcom, 2019-07-31:
It should overwrite. You need to connect USB-UART to NEO2 serial console to debug

Tyrrx, 2019-07-31:
@svpcom 
Inside /boot/ there are two old files:
System.map-4.19.59-sunxi64 and config-4.19.59-sunxi64 but not for the new kernel.
Could that be the problem and where do I get them?

svpcom, 2019-07-31:
You need to mount boot partition into /boot (it depends on image, some images have separate boot partition and other single partition for boot and root)

Tyrrx, 2019-07-31:
@svpcom 
I am using the armbian Image.

svpcom, 2019-07-31:
What is on serial console during boot?

Tyrrx, 2019-07-31:
@svpcom 
I dont have a Uart Adapter right now but I will tell you as soon as possible.

Tyrrx, 2019-08-24:
@svpcom
I finally soldered a serial connector. This is the result i got:
https://pastebin.com/Vc1s8wuJ

svpcom, 2019-08-24:
```
Loading Environment from EXT4... ** File not found /boot/boot.env **
```
it seems that your u-boot environment in /boot directory is corrupted

closed 2021-02-06 by svpcom