svpcom/wfb-ng #40: Не приходят данные.

Atlantii, 2019-06-22:
Привет. 
Пытаюсь настроить передачу видео и телеметрии с одноплатного ПК на ноутбук с Ubuntu.
После установки по [этому](https://github.com/svpcom/wifibroadcast/wiki/Setup-HOWTO) гайду не получаю данных. 
`$ wfb-cli gs` показывает  Connection failed.
После прочтения [Troubleshooting](https://github.com/svpcom/wifibroadcast/wiki/Troubleshooting) попробовал использовать скрипты и обнаружил интересную деталь. При передаче с ноутбука данные передаются, echo test приходит. В обратную сторону не работает.
Думал что у меня что-то не так с портами, везде поменял 5600 на 5606, не помогло.

Вывод `sudo scripts/rx_standalone.sh wlx30144a0024e0` 
```
Setting wlx30144a0024e0 to channel 9
wlx30144a0024e0 has DLT_IEEE802_11_RADIO Encap
8037130	PKT	0:0:0:0:0:0
8038132	PKT	0:0:0:0:0:0
8039132	PKT	0:0:0:0:0:0
8040134	PKT	0:0:0:0:0:0
8041135	PKT	0:0:0:0:0:0
8042136	PKT	0:0:0:0:0:0
8043138	PKT	0:0:0:0:0:0
8044139	PKT	0:0:0:0:0:0
```
Пакеты от `echo test | nc -u 127.0.0.1 5606` не доходят.
`$ tcpdump` на стороне tx в момент отправки `$ echo test` выдаёт такое:
```
11:30:17.345399 [bit 15] 13:22:33:44:55:01 Unknown SSAP 0x84 > 13:22:33:44:55:01 802.1B I Supervisory, Reject, rcv seq 73, Flags [Final], length 73
11:30:17.345494 [bit 15] 13:22:33:44:55:01 > 13:22:33:44:55:01 Null Information, send seq 0, rcv seq 0, Flags [Command], length 32
	0x0000:  0100 0000 0000 0000 002c df6f b238 3c07  .........,.o.8<.
	0x0010:  eab8 4a18 901c d77a 5eea fe86 9dcb 43e5  ..J....z^.....C.

```
С обеих сторон в /etc/NetworkManager/NetworkManager.conf
tx:
```
[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false

[keyfile]
unmanaged-devices=interface-name:wlx30144a002943
```
rx:
```
[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false

[device]
wifi.scan-rand-mac-address=no

[keyfile]
unmanaged-devices=interface-name:wlx30144a0024e0

```
Вывод strace tx:
```
rock64@rock64:~$ ps uaxw | grep tx
root      3566  0.0  0.0   8148  3320 pts/1    S+   11:29   0:00 sudo scripts/tx_standalone.sh wlx30144a002943
root      3567  0.0  0.0   5924  2800 pts/1    S+   11:29   0:00 /bin/bash scripts/tx_standalone.sh wlx30144a002943
root      3622  0.0  0.0   7288  1452 pts/1    S+   11:29   0:00 ./wfb_tx -p 1 -u 5606 -K drone.key wlx30144a002943
rock64    4314  0.0  0.0   5524   564 pts/3    S+   11:49   0:00 grep tx
rock64@rock64:~$ sudo strace -ff -p 3622

strace: Process 3622 attached
ppoll([{fd=3, events=POLLIN}], 1, NULL, NULL, 0) = 1 ([{fd=3, revents=POLLIN}])
recvfrom(3, "test\n", 1446, 0, NULL, NULL) = 5
sendto(4, "\0\0\r\0\0\200\10\0\10\0007\0\1\10\1\0\0\377\377\377\377\377\377\23\"3DU\1\23\"3"..., 110, 0, NULL, 0) = 110
sendto(4, "\0\0\r\0\0\200\10\0\10\0007\0\1\10\1\0\0\377\377\377\377\377\377\23\"3DU\1\23\"3"..., 69, 0, NULL, 0) = 69
recvfrom(3, 0x7ffda48ba8, 1446, 0, NULL, NULL) = -1 EAGAIN (Resource temporarily unavailable)
ppoll([{fd=3, events=POLLIN}], 1, NULL, NULL, 0

```
Что бы я не делал данные не доходят.
На tx стороне Rock64 Debian stretch
На rx ноутбук Ubuntu 18.4.2
С обеих сторон используется китайский адаптер на AR9271. 

Кстати, при использовании wifibroadcast от befinitiv передача видео работает, что наталкивает меня на мысль что с железом проблем у меня нету.

Буду очень рад любой помощи.

svpcom, 2019-06-22:
В sh скриптах и в конфигурации по умолчанию для python-демона выстален диапазон 5ГГц. У вас карта поддерживает толко 2.4.  Установка через sh-скрипты сейчас deprecated. Я рекомендую использовать python-демон (он легко собирается через make deb или make rpm). Для atheros карточек кроме диапазона 2.4ГГц можно попробовать отключить STBC и LDPC в файле конфигурации.

svpcom, 2019-06-22:
состояние демона можно проверить через:
```
systemctl status wifibroadcast@gs  или wifibroadcast@drone
journalctl -xu wifibroadcast@gs или wifibroadcast@drone
```

Atlantii, 2019-06-22:
Пересобрал и переустановил пакеты на всякий случай. Проверил все конфиги. 
STBC на всякий случай выставил в 0
gs:
```
tyoma@Lenovo:~/wifibroadcast/deb_dist$ systemctl daemon-reload
tyoma@Lenovo:~/wifibroadcast/deb_dist$ systemctl start wifibroadcast@gs
tyoma@Lenovo:~/wifibroadcast/deb_dist$ systemctl status wifibroadcast@gs
● wifibroadcast@gs.service - WFB profile gs
   Loaded: loaded (/lib/systemd/system/wifibroadcast@.service; disabled; vendor 
   Active: activating (auto-restart) (Result: exit-code) since Sun 2019-06-23 01
  Process: 4856 ExecStart=/usr/bin/python -m telemetry.server gs ${WFB_NICS} (co
 Main PID: 4856 (code=exited, status=1/FAILURE)

tyoma@Lenovo:~/wifibroadcast/deb_dist$ journalctl -xu wifibroadcast@gs
-- Logs begin at Thu 2019-06-20 21:51:53 EEST, end at Sun 2019-06-23 03:35:57 EE
июн 21 00:04:01 Lenovo systemd[1]: Started WFB profile gs.
-- Subject: Запуск юнита wifibroadcast@gs.service завершен
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Процесс запуска юнита wifibroadcast@gs.service был завершен.
-- 
-- Результат: RESULT.
июн 21 00:04:02 Lenovo python[27181]: 2019-06-21 00:04:02+0300 [-] Log opened.
июн 21 00:04:03 Lenovo python[27181]: 2019-06-21 00:04:03+0300 [-] # iw reg set 
июн 21 00:04:03 Lenovo python[27181]: 2019-06-21 00:04:03+0300 [-] Upon execvpe 
июн 21 00:04:03 Lenovo python[27181]:         :Traceback (most recent call last)
июн 21 00:04:03 Lenovo python[27181]:           File "/usr/lib/python2.7/dist-pa
июн 21 00:04:03 Lenovo python[27181]:             environment)
июн 21 00:04:03 Lenovo python[27181]:           File "/usr/lib/python2.7/dist-pa
июн 21 00:04:03 Lenovo python[27181]:             os.execvpe(executable, args, e
июн 21 00:04:03 Lenovo python[27181]:           File "/usr/lib/python2.7/os.py",
июн 21 00:04:03 Lenovo python[27181]:             _execvpe(file, args, env)
июн 21 00:04:03 Lenovo python[27181]:           File "/usr/lib/python2.7/os.py",
июн 21 00:04:03 Lenovo python[27181]:             func(fullname, *argrest)
июн 21 00:04:03 Lenovo python[27181]:         OSError: [Errno 2] No such file or
июн 21 00:04:03 Lenovo python[27181]: 2019-06-21 00:04:03+0300 [-] Stopping reac
lines 1-23

```
drone:
```
rock64@rock64:~/wifibroadcast/deb_dist$ sudo systemctl daemon-reload
rock64@rock64:~/wifibroadcast/deb_dist$ sudo systemctl start wifibroadcast@dronerock64@rock64:~/wifibroadcast/deb_dist$ sudo systemctl start wifibroadcast@dronerock64@rock64:~/wifibroadcast/deb_dist$ sudo systemctl status wifibroadcast@drone
● wifibroadcast@drone.service - WFB profile drone
   Loaded: loaded (/lib/systemd/system/wifibroadcast@.service; disabled; vendor 
preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2019-06-22 22
:10:15 UTC; 2s ago
  Process: 6319 ExecStart=/usr/bin/python -m telemetry.server drone ${WFB_NICS} 
(code=exited, status=1/FAILURE)
 Main PID: 6319 (code=exited, status=1/FAILURE)

Jun 22 22:10:15 rock64 systemd[1]: wifibroadcast@drone.service: Failed w
ith result 'exit-code'.
rock64@rock64:~/wifibroadcast/deb_dist$ journalctl -xu wifibroadcast@drone
-- Logs begin at Thu 2016-11-03 17:16:43 UTC, end at Sat 2019-06-22 22:10:31 UTC
. --
Jun 22 21:40:58 rock64 systemd[1]: Started WFB profile drone.
-- Subject: Unit wifibroadcast@drone.service has finished start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit wifibroadcast@drone.service has finished starting up.
-- 
-- The start-up result is done.
Jun 22 21:41:01 rock64 python[1817]: 2019-06-22 21:41:01+0000 [-] Log opened.
Jun 22 21:41:01 rock64 python[1817]: 2019-06-22 21:41:01+0000 [-] # iw reg set G
Y
Jun 22 21:41:02 rock64 python[1817]: 2019-06-22 21:41:02+0000 [-] wlan0: ERROR w
hile getting interface flags: No such device
Jun 22 21:41:02 rock64 python[1817]: 2019-06-22 21:41:02+0000 [-] Stopping react
or due to fatal error: RC 255: ifconfig wlan0 down
Jun 22 21:41:02 rock64 python[1817]: 2019-06-22 21:41:02+0000 [-] Main loop term
inated.
Jun 22 21:41:02 rock64 python[1817]: 2019-06-22 21:41:02+0000 [-] Exiting with c
ode 1
Jun 22 21:41:02 rock64 systemd[1]: wifibroadcast@drone.service: Main pro
cess exited, code=exited, status=1/FAILURE

```
И вот что ещё интересное нашел. Похоже адаптер не поддерживает настройку мощности
```
sudo python -m telemetry.server drone wlx30144a002943
2019-06-22 22:13:56+0000 [-] Log opened.
2019-06-22 22:13:56+0000 [-] # iw reg set GY
2019-06-22 22:13:57+0000 [-] # ifconfig wlx30144a002943 down
2019-06-22 22:13:57+0000 [-] # iw dev wlx30144a002943 set monitor otherbss
2019-06-22 22:13:57+0000 [-] # ifconfig wlx30144a002943 up
2019-06-22 22:13:57+0000 [-] # iw dev wlx30144a002943 set channel 9 HT20
2019-06-22 22:13:57+0000 [-] command failed: Operation not supported (-95)
2019-06-22 22:13:57+0000 [-] Stopping reactor due to fatal error: RC 161: iw dev wlx30144a002943 set txpower fixed 58
2019-06-22 22:13:57+0000 [-] Main loop terminated.
2019-06-22 22:13:57+0000 [-] Exiting with code 1

```

svpcom, 2019-06-23:
А что случается на GS? Там чего-то не хватает из утилит похоже.
Я добавлил workaround -- если в конфиге указать
```
[common]
wifi_txpower = None
```
то не будет вызываться установка мощьности

Atlantii, 2019-06-23:
Да, теперь всё работает. Телеметрия передаётся, туннель работает, видео идёт, но у меня не получается использовать 40mb bandwidth и 13 канал.

При попытке переключиться на 40мб происходит это:
```
sudo python -m telemetry.server drone wlx30144a002943
2019-06-23 10:18:12+0000 [-] Log opened.
2019-06-23 10:18:12+0000 [-] # iw reg set GY
2019-06-23 10:18:12+0000 [-] # ifconfig wlx30144a002943 down
2019-06-23 10:18:12+0000 [-] # iw dev wlx30144a002943 set monitor otherbss
2019-06-23 10:18:13+0000 [-] # ifconfig wlx30144a002943 up
2019-06-23 10:18:13+0000 [-] # iw dev wlx30144a002943 set channel 9 HT40+
2019-06-23 10:18:13+0000 [-] Listen for telem stream 1(RX), 2(TX) on 0.0.0.0:14550
2019-06-23 10:18:13+0000 [-] UDPProxyProtocol starting on 14550
2019-06-23 10:18:13+0000 [-] UDPProxyProtocol starting on 14700
2019-06-23 10:18:13+0000 [-] UDPProxyProtocol starting on 38402
2019-06-23 10:18:13+0000 [-] Telem RX: /usr/bin/wfb_rx -p 1 -u 14700 -K /etc/drone.key -k 1 -n 2 wlx30144a002943
2019-06-23 10:18:13+0000 [-] Telem TX: /usr/bin/wfb_tx -p 2 -u 14701 -K /etc/drone.key -B 20 -G long -S 0 -L 0 -M 1 -k 1 -n 2 wlx30144a002943
2019-06-23 10:18:13+0000 [-] Started telem rx
2019-06-23 10:18:13+0000 [-] Started telem tx
2019-06-23 10:18:13+0000 [-] Listen for video stream 3 on 0.0.0.0:5602
2019-06-23 10:18:13+0000 [-] Started video tx
2019-06-23 10:18:13+0000 [-] Video: /usr/bin/wfb_tx -p 3 -u 5602 -K /etc/drone.key -B 40 -G short -S 1 -L 0 -M 23 -k 8 -n 12 wlx30144a002943
2019-06-23 10:18:13+0000 [-] UDPProxyProtocol starting on 14900
2019-06-23 10:18:13+0000 [-] UDPProxyProtocol starting on 44025
2019-06-23 10:18:13+0000 [-] Tunnel RX: /usr/bin/wfb_rx -p 4 -u 14900 -K /etc/drone.key -k 1 -n 2 wlx30144a002943
2019-06-23 10:18:13+0000 [-] Tunnel TX: /usr/bin/wfb_tx -p 5 -u 14901 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 -k 1 -n 2 wlx30144a002943
2019-06-23 10:18:13+0000 [-] Started tunnel rx
2019-06-23 10:18:13+0000 [-] Started tunnel tx
2019-06-23 10:18:13+0000 [-] telem tx: Listen on 14701 for wlx30144a002943
2019-06-23 10:18:13+0000 [-] telem rx: wlx30144a002943 has DLT_IEEE802_11_RADIO Encap
2019-06-23 10:18:13+0000 [-] video tx: Listen on 5602 for wlx30144a002943
2019-06-23 10:18:13+0000 [-] tunnel tx: Listen on 14901 for wlx30144a002943
2019-06-23 10:18:13+0000 [-] tunnel rx: wlx30144a002943 has DLT_IEEE802_11_RADIO Encap
2019-06-23 10:18:13+0000 [-] tunnel rx: New session detected
2019-06-23 10:18:13+0000 [-] tunnel rx: 484 packets lost
2019-06-23 10:18:20+0000 [-] tunnel rx: Error: socket error!
2019-06-23 10:18:20+0000 [-] telem rx: Error: socket error!
2019-06-23 10:18:20+0000 [-] Stopped RX tunnel rx with code 1
2019-06-23 10:18:20+0000 [-] Stopping reactor due to fatal error: A process has ended with a probable error condition: process ended with exit code 1.
2019-06-23 10:18:20+0000 [-] (UDP Port 14900 Closed)
2019-06-23 10:18:20+0000 [-] (UDP Port 44025 Closed)
2019-06-23 10:18:20+0000 [UDPProxyProtocol (UDP)] (UDP Port 14550 Closed)
2019-06-23 10:18:20+0000 [UDPProxyProtocol (UDP)] (UDP Port 14700 Closed)
2019-06-23 10:18:20+0000 [UDPProxyProtocol (UDP)] (UDP Port 38402 Closed)
2019-06-23 10:18:20+0000 [-] Stopped RX telem rx with code 1
2019-06-23 10:18:20+0000 [-] Main loop terminated.
2019-06-23 10:18:20+0000 [-] Exiting with code 1

```
При попытке использования 13го канала
```
 sudo python -m telemetry.server drone wlx30144a002943
2019-06-23 10:30:27+0000 [-] Log opened.
2019-06-23 10:30:27+0000 [-] # iw reg set GY
2019-06-23 10:30:27+0000 [-] # ifconfig wlx30144a002943 down
2019-06-23 10:30:27+0000 [-] # iw dev wlx30144a002943 set monitor otherbss
2019-06-23 10:30:27+0000 [-] # ifconfig wlx30144a002943 up
2019-06-23 10:30:27+0000 [-] # iw dev wlx30144a002943 set channel 13 HT20
2019-06-23 10:30:27+0000 [-] Listen for telem stream 1(RX), 2(TX) on 0.0.0.0:14550
2019-06-23 10:30:27+0000 [-] UDPProxyProtocol starting on 14550
2019-06-23 10:30:27+0000 [-] UDPProxyProtocol starting on 14700
2019-06-23 10:30:27+0000 [-] UDPProxyProtocol starting on 41855
2019-06-23 10:30:27+0000 [-] Telem RX: /usr/bin/wfb_rx -p 1 -u 14700 -K /etc/drone.key -k 1 -n 2 wlx30144a002943
2019-06-23 10:30:27+0000 [-] Telem TX: /usr/bin/wfb_tx -p 2 -u 14701 -K /etc/drone.key -B 20 -G long -S 0 -L 0 -M 1 -k 1 -n 2 wlx30144a002943
2019-06-23 10:30:27+0000 [-] Started telem rx
2019-06-23 10:30:27+0000 [-] Started telem tx
2019-06-23 10:30:27+0000 [-] Listen for video stream 3 on 0.0.0.0:5602
2019-06-23 10:30:27+0000 [-] Started video tx
2019-06-23 10:30:27+0000 [-] Video: /usr/bin/wfb_tx -p 3 -u 5602 -K /etc/drone.key -B 20 -G short -S 1 -L 0 -M 31 -k 8 -n 12 wlx30144a002943
2019-06-23 10:30:28+0000 [-] UDPProxyProtocol starting on 14900
2019-06-23 10:30:28+0000 [-] UDPProxyProtocol starting on 53005
2019-06-23 10:30:28+0000 [-] Tunnel RX: /usr/bin/wfb_rx -p 4 -u 14900 -K /etc/drone.key -k 1 -n 2 wlx30144a002943
2019-06-23 10:30:28+0000 [-] Tunnel TX: /usr/bin/wfb_tx -p 5 -u 14901 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 -k 1 -n 2 wlx30144a002943
2019-06-23 10:30:28+0000 [-] Started tunnel rx
2019-06-23 10:30:28+0000 [-] Started tunnel tx
2019-06-23 10:30:28+0000 [-] telem tx: Listen on 14701 for wlx30144a002943
2019-06-23 10:30:28+0000 [-] video tx: Listen on 5602 for wlx30144a002943
2019-06-23 10:30:28+0000 [-] telem rx: wlx30144a002943 has DLT_IEEE802_11_RADIO Encap
2019-06-23 10:30:28+0000 [-] tunnel tx: Listen on 14901 for wlx30144a002943
[... cut ...]
2019-06-23 10:30:35+0000 [-] Stopping reactor due to fatal error: A process has ended with a probable error condition: process ended with exit code 1.
2019-06-23 10:30:35+0000 [-] (UDP Port 14550 Closed)
2019-06-23 10:30:35+0000 [-] (UDP Port 14700 Closed)
2019-06-23 10:30:35+0000 [-] (UDP Port 41855 Closed)
2019-06-23 10:30:35+0000 [-] Stopped RX tunnel rx with code 1
2019-06-23 10:30:35+0000 [UDPProxyProtocol (UDP)] (UDP Port 53005 Closed)
2019-06-23 10:30:35+0000 [UDPProxyProtocol (UDP)] (UDP Port 14900 Closed)
2019-06-23 10:30:35+0000 [-] Main loop terminated.
2019-06-23 10:30:35+0000 [-] Exiting with code 1

```

svpcom, 2019-06-23:
https://en.wikipedia.org/wiki/List_of_WLAN_channels
40МГц есть только на 3 и 11 канале.

svpcom, 2019-06-23:
Доступность 13-го канала в регионе GY можно проверить через `iw reg get` и `iw list`

Atlantii, 2019-06-23:
Не понимаю почему не получается заставить работать 13 канал
```
rock64@rock64:~$ sudo iw reg get
global
country GY: DFS-UNSET
	(2402 - 2482 @ 40), (N/A, 30), (N/A)
	(5735 - 5835 @ 80), (N/A, 30), (N/A)

```
```
rock64@rock64:~$ sudo iw list
Wiphy phy0
	max # scan SSIDs: 4
	max scan IEs length: 2257 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports RSN-IBSS.
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x1 RX 0x1
	Configured Antennas: TX 0x1 RX 0x1
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * P2P-client
		 * P2P-GO
		 * outside context of a BSS
	Band 1:
		Capabilities: 0x116e
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 8 usec (0x06)
		HT TX/RX MCS rate indexes supported: 0-7
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps (short preamble supported)
			* 5.5 Mbps (short preamble supported)
			* 11.0 Mbps (short preamble supported)
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (20.0 dBm) (no IR)
			* 2472 MHz [13] (20.0 dBm) (no IR)
			* 2484 MHz [14] (disabled)
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * set_bss
		 * authenticate
		 * associate
		 * deauthenticate
		 * disassociate
		 * join_ibss
		 * remain_on_channel
		 * set_tx_bitrate_mask
		 * frame
		 * frame_wait_cancel
		 * set_wiphy_netns
		 * set_channel
		 * set_wds_peer
		 * tdls_mgmt
		 * tdls_oper
		 * probe_client
		 * set_noack_map
		 * register_beacons
		 * start_p2p_device
		 * set_mcast_rate
		 * testmode
		 * channel_switch
		 * set_qos_map
		 * connect
		 * disconnect
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * mesh point: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0x40 0xb0 0xc0 0xd0
		 * managed: 0x40 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	software interface modes (can always be added):
		 * AP/VLAN
		 * monitor
	valid interface combinations:
		 * #{ managed, P2P-client } <= 2, #{ AP, P2P-GO } <= 2,
		   total <= 2, #channels <= 1
	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	Device supports TX status socket option.
	Device supports HT-IBSS.
	Device supports SAE with AUTHENTICATE command
	Device supports low priority scan.
	Device supports scan flush.
	Device supports AP scan.
	Device supports per-vif TX power setting
	Driver supports a userspace MPM
	Device supports ACK timeout estimation.
	Device supports configuring vdev MAC-addr on create.

```

closed 2019-09-16 by svpcom