svpcom/wfb-ng #41: Different information about latency of UDP over WFB

urpylka, 2019-10-24:
Please let me know what information is correct?

1. In the [px4 documentation](https://dev.px4.io/v1.9.0/en/qgc/video_streaming_wifi_broadcast.html#faq)

> The new version was rewritten to use UDP as data source and pack one source UDP packet into one radio packet. Radio packets now have variable size that depends on payload size. This significantly reduces a video latency.

2. Or in the [this repo's wiki](https://github.com/svpcom/wifibroadcast/wiki/Setup-HOWTO)

> For IPoverWB (IPv4 over Wifibroadcast tunnel) you need only tun/tap kernel driver (tun.ko). Tunnel will be configured and work out of box. Drone side will have 10.5.0.2/24 address and GS - 10.5.0.1/24. Please note that tunnel use less efficient coding rate (to minimize latency) than video and mavlink streams and use it only for low-bandwidth traffic (like ssh), not as general link for video and telemetry streams.

Maybe did you mean (in px4 documentation) that small UDP packages in summary takes less from common bandwidth and due to that video packages arrive faster?

___

And how it will be better to transmit large files (like 8MB JPEG images or 1GB video files) over WFB.

Thanks!

svpcom, 2019-10-24:
1. Yes. Original wifibroadcast  pack incoming data from the pipe and send a new wifi packet only if it is full. This affects to latency because x264 frame size is not equal to wifi packet size.

2. Video stream use 8/12 fec rate (only 50% overhead). This mean that for recover one or more lost packet you need to receive 12 packets. Video has constant packet rate so delay will be small. But for example TCP stream for ssh connection will not have constant rate, so you will got freezes in case of packet lost. Due to this IPoverWFB use 1/2 fec rate (100% overhead). This allow to recover one packet lost without awaiting other packets.

For you case it depends:  How frequently you will get lost packets? If there is noisy environment with burst noise - you can use 8/12 or (for example) 80/120, but you will need to send additional traffic (or modify wfb sources to emit empty packets) to close fec block.

urpylka, 2019-10-24:
Thanks for quick response.

1. Does all differenting consist in the FEC coefficient? (I want to ask you. If I will be used UDP w 8/12 FEC it will be same that if I will be used Video Stream w 8/12 FEC)? My basic message in the last question consist in why we are don't using UDP for RTP or MAVLink traffic? Because we can change FEC coefficient.

2. And I can't understood if I send data a lot (for example 100 JPEG files by 4MB each in the one queue), why I get a gap in queue packages? I guess it will just one queue. Is it correct?

closed 2021-02-06 by svpcom