svpcom/wfb-ng #43: Comparison between projects

Summer88, 2019-11-25:
Dear Sirs, 
I'm looking around for a way to transmit both video and RC Mavlink data over 2.4 Ghz, found many projects about this but cannot get a starting point:

From this document https://dronebridge.gitbook.io/docs/comparison they put comparison on:

svpcoms WifiBroadcast (basic TX-RX)
DroneBridge (system design with library)
OpenHD (follow-up on EZ-WifiBroadcast)
EZ-Wifibroadcast (not under development)

But looking at code the most updated seems that svpcoms WifiBroadcast and is able to manage both video and Mavlink. 
Can please someone confirm  that will run on a pizero  please?

svpcom, 2019-11-25:
My WFB supports following data streams:
1. Video stream (RTP) - unidirectional
2. Mavlink (with packet aggregation and RSSI data injection) - bidirectional
3. IP encapsulation (using tun driver) - bidirectional

Summer88, 2019-11-25:
Thank you for reply, I've just  downloaded Raspbian.
Will try to setup your project  on pi zero.
BR

closed 2021-02-06 by svpcom