svpcom/wfb-ng #44: Collect video from different hosts.

eltsov-da, 2019-11-29:
Dear Sirs. In main features you mentioned "Distributed operation. It can gather data from cards on different hosts" 
We working on video translation for orienteering project. And we are looking for solution which will make possible gathering one video stream from more then one RX host.
We need it because some times forest is not radio transparent. Our idea is to create wired network of controller on some area where operator can move. So we need some re-transmitter from WiFi to wired network and then some kind 
 RX program which will get data  from wired interface and give us picture (or RTP ...) 
Is "Distributed operation" is feature we need? And if it is how we can configure it.

Best Regards. 
Denis
P.S. I already try classic wifibroadcasting and re-transmitter using tcpdump ->netcat ->netcat -> tcpreplay 
I can't say it doesn't work at all but I'v got lot of artifacts and some time stream fail. Seems that tcpreplay take to much processor time.

svpcom, 2019-11-29:
Hi,
Yes. It is what you needed. You can have several hosts with wifi cards (scattered on the forest) which will capture packets and send them via ethernet to central host and apply FEC there. This will work like cell network:
```
$ ./wfb_rx 
Local receiver: ./wfb_rx [-K rx_key] [-k RS_K] [-n RS_N] [-c client_addr] [-u client_port] [-p radio_port] [-l log_interval] interface1 [interface2] ...
Remote (forwarder): ./wfb_rx -f [-c client_addr] [-u client_port] [-p radio_port] interface1 [interface2] ...
Remote (aggregator): ./wfb_rx -a server_port [-K rx_key] [-k RS_K] [-n RS_N] [-c client_addr] [-u client_port] [-l log_interval]
```

On the forwarder nodes you need to setup wfb_rx with "-a" flag. On aggregator node with "-a" flag.
Use `scripts/rx_standalone.sh` and `scripts/rx_standalone.sh` for reference (how to initialize HW without python daemon). See picture in https://github.com/svpcom/wifibroadcast/wiki/Enhanced-setup for sample setup

Main disadvantage of distributed mode - it is unidirected (this is because it is not supported in python daemon)

eltsov-da, 2019-12-01:
Thank you for your answer and for work.
For a test I try 
Rspberry Cam - Raspberry 3 as TX
Raspberry 3 as RX and my notebook as gstreamer host and display. (may be it's problem?)
I can got picture but I can't call it video :) 
I use wlan cards on atheros 9172 chip 2.4 GHz. 
This configuration work fine on "classic" Wifibroadcast solution.
May be you have some idea?
Best regards Denis.

svpcom, 2019-12-02:
What is rx and tx setup scripts and wfb_rx/rx debug output? I'll recommend for fist try use simple config (use python daemons) to be sure that always configured correctly.

eltsov-da, 2020-03-02:
Hi. Have not still test tx side with my atheros cards. But any way with Mediatek 7662 it work perfect. (Net of forwarding stations connected throw ethernet)  
Now I try to use mesh network of forwarding stations to send data to  aggregate station.
I try  to use batman-adv based mesh. (https://medium.com/swlh/setting-up-an-ad-hoc-mesh-network-with-raspberry-pi-3b-using-batman-adv-1c08ee565165)  Of course I use separate cards and WiFi channel in mesh and broadcasting. 
I succeed in starting mesh network (I can use ssh etc) but if I try send wifibroadcasting data throw it, I'v got't "unable to decrypt packe ..." on most lines of output on aggregate station. Even if mesh have only two nodes and lie on the table. 
I try to use Raspberry Pi as aggregate station or route traffic to my laptop and aggregate on it.  
May be you have some good idea for something like it.

P.S. I understand that this solution may be dead end when I try to scale it. But it's looks very attractive to use wifi mesh instead of wire ethernet.

svpcom, 2020-03-02:
Check maximum MTU which mesh network support. It can truncate UDP packets from WFB. Use tcpdump and/or wireshark to inspect traffic. Also check max mesh bandwidth (use iperf3). If it use ad-hoc mode you will be limited by 2mbit/s bandwidth.

eltsov-da, 2020-03-04:
It's seems that  bandwidth isn't enough. The same is for using RPi as access point.

eltsov-da, 2020-04-19:
After some investigations I 
1. Drop idea of using WiFi as channel between stations. Slow 
2. I change raspberry os image and atheros cards work fine on this image.

But now I try to fund best antennas for our, forest, project. But I can't find any simple way to see signal characteristics. Is it possible to see RSSI or something else in log output ?

svpcom, 2020-04-19:
add to wifibroadcast.cfg:
```
[common]
debug = True
```

svpcom, 2020-04-19:
Could you describe your project? I can add it as use case to wiki.

eltsov-da, 2020-04-19:
but I use configuration with out python. Because of forwarder and aggregator nodes.

eltsov-da, 2020-04-19:
About our project:
In some sports like orienteering it's impossible to be sure where sportsmen will run, and so it's hard to show sportsmen in action. So we use special operator, we call cameramen, who run part of the course with sportsmen and take online video (https://youtu.be/kCEWn-lfGEI?t=1279)
Also, often there is no good cellular network coverage in the forest so we can't use it. 
Instead we build our own network based on WiFibroadcast.
Test go's look's something like this:  https://www.youtube.com/watch?v=PpuEirXuFGQ
(of course in real transmissions we use stadycam for stabilization) 
Here is something like project blog (in Russina) https://vk.com/ogpscenter?w=wall-185828882_21

svpcom, 2020-04-21:
@eltsov-da 

wfb_rx выдает на stdout такой лог:
```
timestamp ANT  номер_антенны:всего_пакетов:мин_rssi:средний_rssi:макс_rssi
...
timestamp PKT  всего_пакетов:пакетов_с_ошибкой:без_ошибки:восстановленных:пропущеных:плохих
```

статистика обновляется раз в секунду. Как отлаживать - можно смотреть на RSSI и на FEC (количество восстановленных пакетов). Если оно начинает расти, значит сигнал ослаб или канал зашумлен (если rssi хороший).  Short-GI или 40MHz канал не советую - сильно падает помехозащищенность.

Антенны лучше выбирать с круговой поляризацией - в при наличии переотраженных сигналов это лучший вариант. Я использую такие:  https://www.maartenbaert.be/quadcopters/antennas/pagoda-antenna/ и https://www.maartenbaert.be/quadcopters/antennas/triple-feed-patch-antenna/.

У меня базовая станция состоит из 6 направленных антенн (каждая отвечает за 60 градусов азимута) и 2 всенаправленных (две, потому, что у каждой диаграмма не шар, а тор). В качестве карточек - 4 x rtl8812au (по две антенны на карточку).

А как приемные станции связаны между собой? Неужели в лесу прокладываете оптику?

eltsov-da, 2020-04-22:
Огромное спасибо. 
В данном проекте приемные станции связываем просто витой парой. Т.к. на практике в лесу больше 60 метров уверенного приема пока получить не удалось. В том числе и по этому изучаю вопрос с антеннами.
Но в целом, оптику в лесу тянем постоянно. На практике это совсем не так страшно как кажется. FTTH кабель просто развешиваем по деревьям. Он заранее нарезан по 300-700 метров и законцован.

eltsov-da, 2020-04-22:
А если не секрет, что на передающей стороне используете?

svpcom, 2020-04-22:
rtl8812au с двумя антеннами (если включить STBC, то можно их одновременно задействовать) Начинал с alfa awus036ach, но у нее есть странные проблемы. Сейчас использую https://item.taobao.com/item.htm?spm=a21wu.10013406.0.0.292a1481TQt4JT&id=597040872396 (можно купить через посредников). 

alfa умеет выключаться **в воздухе**. То есть карточка выглядит как рабочая, то перестает передавать пакеты. Даже после powercycle. Через некоторое время (пару часов) все восстанавливается. На земле такого добиться не получается. У меня были подозрения на DFS (в 20 км от меня есть аэропорт), но в драйвере он выключен и карточка с taobao (на том же чипе ведет себя нормально).  Единственный минус карточки с taobao - она только 5GHz  (2.4 не поддерживается) и очень сильно греется (радиатор обязателен).

По поводу антенн. Я использую 5.8ГГц с круговой поляризацией. Но это в воздухе в пределах прямой видимости. В лесу наверное лучше 2.4 (но тут надо эксперимент провести, так как 2.4 хорошо водой поглощается). Некоторые делали конвертор частоты, который сдвигал диапазон на 1.2GHz или на 433MHz (получался линк на сотни километров), но это не совсем законно. Подробности можно поискать на форумах rcdesign.ru, но, к сожалению, авиамоделисты очень не любят делиться своими наработками (хвастаться - любят). Вот такое вот токсичное community :-(

svpcom, 2020-04-22:
Если есть желание написать статью про настройку распределенного приема, то я ее выложу в wiki - может кому-нибудь пригодится

eltsov-da, 2020-04-22:
Спасибо, за информацию (мне кажется, что вот это тоже самое только дороже https://aliexpress.ru/item/4000542389424.html?spm=a2g0o.productlist.0.0.4a261212SVZMhY&algo_pvid=728d4202-0786-4934-8b7a-1857d6196d1a&algo_expid=728d4202-0786-4934-8b7a-1857d6196d1a-3&btsid=0b8b036a15875554217983058ef078&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) 
По поводу статьи - займусь.

svpcom, 2020-04-22:
Да, похоже его стали продавать и на aliexpress

svpcom, 2021-01-12:
@eltsov-da Удалсь запустить в production? Я недавно выложил новый релиз - в нем я исправил старую проблему с большой потерей пакетов при плохой связи (неполный FEC блок раньше выкидывался целиком, если после него шел полный) и немного уменьшил latency при приеме пакетов.

eltsov-da, 2021-01-12:
Ну с этой пандемией у нас большинство стартов отменили. Единственный раз когда попробовали на реальном старте описан вот здесь. https://vk.com/ogpscenter?w=wall-185828882_100
Новую версию посмотрю. Спасибо.

svpcom, 2021-01-12:
Диапазон 2.4 или 5.8? Антенны обычные (с линейной поляризацией) или с круговой? Просто я заметил, что видео иногда рассыпается. Расстояния вроде не сильно большие (50м). С круговой поляризацией у меня стабильно линк идет на 900м (правда в воздухе).

eltsov-da, 2021-01-12:
2.4, с линейной. Но там возникла проблемма с передающей системой. И в результате карточка и питалась напрямую от raspberry, и самое главное лежала в рюкзаке. 
Ну и приемные антенны не направленные.

svpcom, 2021-01-12:
2.4 очень узок (всего 3 независимых канала) и в населенных пунктах замусорен (wifi, сигнализации и тд) - если есть возможность от него отказаться (то есть если есть запас по мощьности и/или чувствительности антенны), то лучше это сделать.

С линейными антеннами 2 проблемы:
- антенны приемника и передатчика должны находиться в одной плоскости (в вашем случае это скорее всего будет не так)
- переотраженные сигналы от препятствий накладываются на основной сигнал

Как вариант можно попробовать FPV антенны на 5.8ГГц (типа клевер или pagoda) и взять карточку с STBC (на основе rtl8812au) так как даже у клевера диаграмма направленносте не шар, а тор и 2 антенны развенутые под 90 градусов будут давать сигнал во все стороны. Если карточка с усилителем (alfa или noname с taobao), то ее нужно активно охлаждать (радиатор и кулер). Иначе она начнет терять пакеты и усилитель быстро деградирует. Канал нужно выбирать как можно ближе к резонансной частоте антенны (в случае 5.8 это канал 165)

С LKV373A я эксперементировал пару лет назад - для меня у нее очень большая задержка. Плюс она большая и тяжелая. В вашем случае лучше взять что-то типа https://www.kurokesu.com/shop/cameras/CAMERA_C1_PRO. Оно дает на выходе h264 поток (поддерживается gstreamer'ом, но у продавца нужно спросить исходник программы по настройке режимов) и можно вкрутить любую оптику.

closed 2021-02-06 by svpcom