svpcom/wfb-ng #47: large packet loss

florentmaye, 2019-12-06:
Hi,

Your project seems really amazing! Great work! :)

I tried to use it but I'm facing some issues. When I tried to stream video, Video stream are totally corrupted (ex: almost all grey  or Y data almost not existing, depending on the method use).

However, basic tests seems to work properly, I can send messages without issues between the drone and the GS. But when I tried to send large data it seems it is mostly lost, which could explain my issue. For instance I did a for loop to count till 1000000 on TX side, that I netcat to WFB, and I redirected the RX output to a file to count how many were recieved: I had almost 90% of the messages lost. I did similar tests with video files and had also big loss.

Do you have any idea on what could cause this issue?

I'm using raspberry pi 4 with TPLink w722n wifi adapter.

Thank you for all your work!

svpcom, 2019-12-06:
Due to absence of ACKs you should not exceed channel bandwidth. For MCS1 it is ~7mbit/s with large (1400 bytes packets). For small packets it will be less. Run `wfb-cli` on RX size and see packets statistics

florentmaye, 2019-12-06:
Thank you for your answer, I understand your point. Still I'm surprised not
to be able to run a simple video stream, like those provided as example.
Do you have any clue on how to make it work?

Thank you for your help!

svpcom, 2019-12-07:
What chipset is in TPLink w722n? Do you use it on the both ends?

florentmaye, 2019-12-19:
I think the bandwith in 2.4Ghz is simply not large enough for my expectation: reducing the bitrate made the video transmission work properly.
It is the Atheros AR9271 (I wasn't able to use the RealTeck version) and I use it on both sides.

Thank you!

closed 2019-12-19 by florentmaye