svpcom/wfb-ng #48: OSError: [Errno 8] Exec format error

lukaszwielgosz, 2019-12-17:
Hello. I have some issues trying to run on Nanopi NEO2, on linux mint everything works perfectly, but when starting on Nanopi NEO2 I'm getting following errors:

`Dec 17 17:42:36 NanoPi-NEO2 systemd[1]: Started WFB profile drone.
Dec 17 17:42:37 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:37+0000 [-] Log opened.
Dec 17 17:42:37 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:37+0000 [-] # iw reg set BO
Dec 17 17:42:37 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:37+0000 [-] # ifconfig wlan0 down
Dec 17 17:42:37 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:37+0000 [-] # iw dev wlan0 set monitor otherbss
Dec 17 17:42:37 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:37+0000 [-] # ifconfig wlan0 up
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] # iw dev wlan0 set channel 161 HT20
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] Listen for telem stream 1(RX), 2(TX) on 0.0.0.0:14560
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] UDPProxyProtocol starting on 14560
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] UDPProxyProtocol starting on 14700
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] UDPProxyProtocol starting on 46656
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] Telem RX: /usr/bin/wfb_rx -p 1 -u 14700 -K /etc/drone.key -k 1 -n 2 wlan0
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] Telem TX: /usr/bin/wfb_tx -p 2 -u 14701 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 -k 1 -n 2 wlan0
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] Started telem rx
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] Started telem tx
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] Listen for video stream 3 on 0.0.0.0:5602
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] Started video tx
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] Video: /usr/bin/wfb_tx -p 3 -u 5602 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 -k 8 -n 12 wlan0
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] UDPProxyProtocol starting on 14900
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] UDPProxyProtocol starting on 35328
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] Tunnel RX: /usr/bin/wfb_rx -p 4 -u 14900 -K /etc/drone.key -k 1 -n 2 wlan0
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] Tunnel TX: /usr/bin/wfb_tx -p 5 -u 14901 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 -k 1 -n 2 wlan0
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] Started tunnel rx
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] Started tunnel tx
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem rx: Upon execvpe /usr/bin/wfb_rx ['/usr/bin/wfb_rx', '-p', '1', '-u', '14700', '-K', '/etc/drone.key', '-k', '1', '-n', '2', 'wlan0'] in environment id 6982432
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem rx: :Traceback (most recent call last):
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem rx:   File "/usr/lib/python2.7/dist-packages/twisted/internet/process.py", line 430, in _fork
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem rx:     environment)
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem rx:   File "/usr/lib/python2.7/dist-packages/twisted/internet/process.py", line 508, in _execChild
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem rx:     os.execvpe(executable, args, environment)
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem rx:   File "/usr/lib/python2.7/os.py", line 355, in execvpe
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem rx:     _execvpe(file, args, env)
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem rx:   File "/usr/lib/python2.7/os.py", line 370, in _execvpe
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem rx:     func(file, *argrest)
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem rx: OSError: [Errno 8] Exec format error
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem tx: Upon execvpe /usr/bin/wfb_tx ['/usr/bin/wfb_tx', '-p', '2', '-u', '14701', '-K', '/etc/drone.key', '-B', '20', '-G', 'long', '-S', '1', '-L', '0', '-M', '1', '-k', '1', '-n', '2', 'wlan0'] in environment id 6982432
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem tx: :Traceback (most recent call last):
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem tx:   File "/usr/lib/python2.7/dist-packages/twisted/internet/process.py", line 430, in _fork
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem tx:     environment)
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem tx:   File "/usr/lib/python2.7/dist-packages/twisted/internet/process.py", line 508, in _execChild
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem tx:     os.execvpe(executable, args, environment)
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem tx:   File "/usr/lib/python2.7/os.py", line 355, in execvpe
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem tx:     _execvpe(file, args, env)
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem tx:   File "/usr/lib/python2.7/os.py", line 370, in _execvpe
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem tx:     func(file, *argrest)
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] telem tx: OSError: [Errno 8] Exec format error
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] video tx: Upon execvpe /usr/bin/wfb_tx ['/usr/bin/wfb_tx', '-p', '3', '-u', '5602', '-K', '/etc/drone.key', '-B', '20', '-G', 'long', '-S', '1', '-L', '0', '-M', '1', '-k', '8', '-n', '12', 'wlan0'] in environment id 6982432
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] video tx: :Traceback (most recent call last):
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] video tx:   File "/usr/lib/python2.7/dist-packages/twisted/internet/process.py", line 430, in _fork
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] video tx:     environment)
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] video tx:   File "/usr/lib/python2.7/dist-packages/twisted/internet/process.py", line 508, in _execChild
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] video tx:     os.execvpe(executable, args, environment)
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] video tx:   File "/usr/lib/python2.7/os.py", line 355, in execvpe
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] video tx:     _execvpe(file, args, env)
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] video tx:   File "/usr/lib/python2.7/os.py", line 370, in _execvpe
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] video tx:     func(file, *argrest)
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] video tx: OSError: [Errno 8] Exec format error
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] tunnel rx: Upon execvpe /usr/bin/wfb_rx ['/usr/bin/wfb_rx', '-p', '4', '-u', '14900', '-K', '/etc/drone.key', '-k', '1', '-n', '2', 'wlan0'] in environment id 6982432
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] tunnel rx: :Traceback (most recent call last):
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] tunnel rx:   File "/usr/lib/python2.7/dist-packages/twisted/internet/process.py", line 430, in _fork
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] tunnel rx:     environment)
Dec 17 17:42:38 NanoPi-NEO2 python[12908]: 2019-12-17 17:42:38+0000 [-] tunnel rx:   File "/usr/lib/python2.7/dist-packages/twisted/internet/process.py", line 508, in _execChild
[... cut ...]
Dec 17 17:42:38 NanoPi-NEO2 systemd[1]: wifibroadcast@drone.service: Main process exited, code=exited, status=1/FAILURE
Dec 17 17:42:38 NanoPi-NEO2 systemd[1]: wifibroadcast@drone.service: Unit entered failed state.
Dec 17 17:42:38 NanoPi-NEO2 systemd[1]: wifibroadcast@drone.service: Failed with result 'exit-code'.
Dec 17 17:42:43 NanoPi-NEO2 systemd[1]: wifibroadcast@drone.service: Service hold-off time over, scheduling restart.
Dec 17 17:42:43 NanoPi-NEO2 systemd[1]: Stopped WFB profile drone.
Dec 17 17:42:43 NanoPi-NEO2 systemd[1]: Started WFB profile drone.`

svpcom, 2019-12-17:
It seems that you running x86 binaries on arm. You should build WFB on NEO2 or use cross-compiler

lukaszwielgosz, 2019-12-17:
Yes, that was the problem. I have built *.deb packages on NEO2 and it solved the problem. Thank You!

closed 2019-12-17 by svpcom