svpcom/wfb-ng #50: svpcoms Wifibroadcast with Navio2 setup

Mysnomer0, 2020-01-05:
Hi svpcoms,

I wrote a detailed guide on how to setup your repo with the Navio 2 (See attached) Would you be interested in adding this to the wiki?

Thanks,

Milan

[svpcoms Wifibroadcast with Navio2 setup.docx](https://github.com/svpcom/wifibroadcast/files/4022953/svpcoms.Wifibroadcast.with.Navio2.setup.docx)

archont, 2020-01-06:
Amazing work Milan - I wanted to do precisely the same thing but you got
there first :D

 I'll be rolling a similar setup with slightly different hardware, I'll see
how I can contribute. I was thinking of more of either an image builder or
GCS integration route - to make it easy for others to replicate this setup
from GUI only.

Mysnomer0, 2020-01-06:
:) Glad to hear I'm not alone.

Sounds great. :) Now that I know it's possible to setup svpcoms Wifibroadcast with Navio 2, an image builder sounds like a good next step.

On a side note, I've been noticing that the 2.4ghz doesn't seem to have enough bandwidth to handle HD video that isn't too delayed and the bidirectional Mavlink. I had to turn the video way down to get acceptable delay for flying. I'd love to hear how you manage to tune the setup!

Mysnomer0, 2020-01-15:
@svpcom Thoughts on adding my guide to the wiki?

svpcom, 2020-01-15:
Yes, I'll add it today

svpcom, 2020-01-15:
done: https://github.com/svpcom/wifibroadcast/wiki/Navio2-Setup

Mysnomer0, 2020-01-15:
Wonderful, thanks! I'll let you know of revisions as I continue working on the project.

closed 2020-01-15 by Mysnomer0