svpcom/wfb-ng #51: More details on IPoverWB (IPv4 over Wifibroadcast tunnel)?

Mysnomer0, 2020-02-05:
Hi @svpcom ,

I'm confused on how to setup IPoverWB mentioned at the end of the Setup HOWTO wiki page. After doing some research, I'm still confused on how to setup and install a tun/tap kernel driver for Ubuntu. I understand what a tun/tap kernel driver does in theory, but how to install and use it in this application is still unclear. Do you have some references you could point me to? Or, perhaps, a list of commands to run? If I'm able to set it up, I'll try to contribute to the wiki a more detailed guide on setting it up.

Thanks,

Milan

Summer88, 2020-02-07:
Hi, 
I'm interested to an example would be great. Or at least the list of steps to make this work.
1 - set  ip address on the machines:
`sudo ifconfig eth0 "sender" netmask 255.255.255.0 up`
`sudo ifconfig eth0 "receiver" netmask 255.255.255.0 up`
2 - where you specify sender and receiver addresses in config?
Thanks, 
BR

Mysnomer0, 2020-02-07:
Hi BR,

I'm sorry, I'm confused by your comment. Are you also wanting the answer to my question, or are you trying to answer it yourself?

Thanks,

Milan

svpcom, 2020-02-07:
For IPoverWB (IPv4 over Wifibroadcast tunnel) you need only tun/tap kernel driver (tun.ko). Tunnel will be configured and work out of box. Drone side will have 10.5.0.2/24 address and GS - 10.5.0.1/24.
So you need:
1. Install .deb package on the both machines
2. Edit `/etc/default/wifibroadcast` and set `WFB_NICS="wlanXXX"`
3. Run `sudo systemctl restart wifibroadcast@gs` on the first and `sudo systemctl restart wifibroadcast@drone` on the second
4. Use `ifconfig` or `ip link ls` to check tunnel interface

svpcom, 2020-02-07:
Builtin config file with all default values: /usr/lib/python2.7/dist-packages/telemetry/conf/master.cfg
Create /etc/wifibroadcast.cfg if you need to override some of them

appetito, 2020-03-07:
Hello!

Mysnomer0, 2020-03-07:
@svpcom Thanks a ton for your input, I'm sorry I didn't get back to you. I'll try out your suggestions as soon as I can.

closed 2021-02-06 by svpcom