svpcom/wfb-ng #53: Help for writing a detailed installation and setup guide for RPI

Yes21, 2020-03-07:
As many of you I find it a nightmare to install and configure WFB, because the wiki is very incomplete and is missing detailed informations.
Especially for a noobs like me :)

So I've begin to write a very detailed guide and want to share it with you.
But I'm still missing informations.

Please don't hesitate to correct it, and to add what is missing ...

Many thanks for advance.

Yes21, 2020-03-07:
- Base image was created with Open.Hd Image Builder (https://github.com/OpenHD/Open.HD_Image_Builder) skiping the 4 and 5 stages to avoid installation of Open.HD. The resulting image has the needed kernel patches and the patched aircrack-ng rtl8812au driver (https://github.com/aircrack-ng/rtl8812au.git 5.2.20 branch) installed.
This base image is working with all raspberry Pis except the RPI4 (support is coming soon)

- burn this image on a SD card and put it into a rapberry Pi ethernet connected with your PC.

- then connect to it via ssh

ssh pi@192.168.0.xx
password = raspberry

- Next steps:

sudo -s

[Install libsodium]

git clone https://github.com/jedisct1/libsodium --branch stable
cd libsodium
./configure
make && make check
sudo make install
cd ..
rm -rf libsodium

[Install mavlink-router]

git clone https://github.com/intel/mavlink-router
cd mavlink-router/
git submodule update --init --recursive
./autogen.sh && ./configure CFLAGS='-g -O2' \
        --sysconfdir=/etc --localstatedir=/var --libdir=/usr/lib64 \
    --prefix=/usr
make
make install
cd ..
rm -rf mavlink-router

[Install gstreamer]

wget https://github.com/OpenHD/gst-raspberrypi/releases/download/1.10.4-stretch/GStreamer-1.10.4-stretch.tar.gz
tar xvf GStreamer-1.10.4-stretch.tar.gz
dpkg -i GStreamer-1.10.4-stretch/*.deb
rm GStreamer-1.10.4-stretch.tar.gz
rm -rf GStreamer-1.10.4-stretch

[Install some dependencies]

apt-get install python3-virtualenv virtualenv
apt-get install python-twisted libpcap-dev libsodium-dev python-pyroute2 python-future python-configparser python-all

[Install wifibroadcast]

cd ~
git clone https://github.com/svpcom/wifibroadcast.git
cd wifibroadcast
make deb
cd deb_dist
dpkg -i wifibroadcast*.deb

[Generate the keys for encryption]

wfb_keygen
cp *.key /etc

[Create the wifibroadcast conf file]

cd /etc
> wifibroadcast.cfg

nano wifibroadcast

copy these lines into it :

----------------------------------------------------------------------------------------------

[common]
wifi_channel = 161     # 161 -- radio channel @5805 MHz, range: 5795–5815 MHz, width 20MHz
                       # 1 -- radio channel @2412 Mhz, 
                       # see https://en.wikipedia.org/wiki/List_of_WLAN_channels for reference
wifi_region = 'BO'     # Your country for CRDA (use BO or GY if you want max tx power) 

[gs_mavlink]
peer = 'connect://127.0.0.1:14550'  # outgoing connection
# peer = 'listen://0.0.0.0:14550'   # incoming connection

bandwidth = 20     # bandwidth 20 or 40 MHz
short_gi = False   # use short GI or not
stbc = 1           # stbc streams: 1, 2, 3 or 0 if unused
mcs_index = 1      # mcs index

[gs_video]
peer = 'connect://127.0.0.1:5600'  # outgoing connection for
                                   # video sink (QGroundControl on GS)
                                   
bandwidth = 20     # bandwidth 20 or 40 MHz
short_gi = False   # use short GI or not
stbc = 1           # stbc streams: 1, 2, 3 or 0 if unused
mcs_index = 1      # mcs index

[drone_mavlink]
peer = 'listen://0.0.0.0:14550'   # incoming connection
# peer = 'connect://127.0.0.1:14550'  # outgoing connection

bandwidth = 20     # bandwidth 20 or 40 MHz
short_gi = False   # use short GI or not
stbc = 1           # stbc streams: 1, 2, 3 or 0 if unused
mcs_index = 1      # mcs index

[drone_video]
peer = 'listen://0.0.0.0:5602'  # listen for video stream (gstreamer on drone)

bandwidth = 20     # bandwidth 20 or 40 MHz
short_gi = False   # use short GI or not
stbc = 1           # stbc streams: 1, 2, 3 or 0 if unused
mcs_index = 1      # mcs index

----------------------------------------------------------------------------------------------

[Create a conf file for mavlink-router]

mkdir /etc/mavlink-router
> /etc/mavlink-router/main.conf

Edit this file and copy into it :

nano /etc/mavlink-router/main.conf

----------------------------------------------
[UdpEndpoint wifibroadcast]
Mode = Normal
Address = 127.0.0.1
Port = 14550
----------------------------------------------

[Set the right interface in /etc/default/wifibroadcast]

Edit /etc/default/wifibroadcast and repace wlan0 with proper wifi interface name :

nano /etc/default/wifibroadcast

(wlan0 should work if you have only one wifi board connected to the Pi, or you can use the MAC address of the boards)

[Setup RTP video streaming on the drone to udp://127.0.0.1:5602]

I don't know what to do here ... !!

[setup a pipeline for gstreamer]

I don't how and where to setup the pipeline (I have the pipeline ... !!

[create the wifi access point for connecting QGC ?]

I don't know were to configure it : board, band, SSID, pass ?

[put the start sequences in .profile]

nano /home/pi/.profile

and add to it :

--------------------------------------------
# uncomment for gs
# systemctl daemon-reload, systemctl start wifibroadcast@gs

# uncomment for drone
# systemctl start wifibroadcast@drone
--------------------------------------------

Mysnomer0, 2020-03-07:
Hi Yes21,

First off, I'm glad to see someone working with and using the wifibroadcast project! :) Second, check out the guide I contributed to the wiki, a detailed installation and setup guide for RPI running a specific setup:

https://github.com/svpcom/wifibroadcast/wiki/Navio2-Setup

Hope this helps on the parts you're confused! If you have any questions or enhancements to my guide, let me know and we can ask svpcom to add it to the wiki.

Thanks,

Milan

Yes21, 2020-03-07:
Hi Milan,

Many thanks for your answer ! I didn't see your guide in the wiki. Sorry ...

I'm sure I will find in it some useful informations for my guide.
I will return to you later if I need more precisions.

Yves

Mysnomer0, 2020-03-08:
Hi Yves,

Are you able to access the link in my previous comment? Or, if you look in the wiki, hopefully you'll see "Navio2-Setup". That's it!

Thanks,

Milan

Yes21, 2020-03-08:
Hi Milan,
Many thanks for your help.
I've read your guide and think I understood where to put the pipeline.
I've edited my guide, so you could perhaps verify if that' correct.

It's now missing me how to create an access point and to redirect video and telemetry flows for QGC.

closed 2021-02-06 by svpcom