svpcom/wfb-ng #55: Issue making WFB working on RPI

Yes21, 2020-03-10:
I've build 2 images using this procedure : 
https://github.com/svpcom/wifibroadcast/issues/53

My setup :

    drone :
    Pi2B + Pi cam + AWUS036ACH (rtl8812au)

    gs :
    Pi3B + 7 " HDMI screen + AWUS1900 (rtl8814au) or ASUS AC56 (rtl8812au)

Both Pis are ethernet connected with my desktop, and I can ssh to them.

The 2 Pis are booting perfectly, and the 2 wifi boards are up and in monitor mode.

But using "wfb-cli gs" on the gs Pi, I can see that all 3 connections failed (refused by the other side)

Here are the output of "sudo python -m telemetry.server gs wlan00" :

on drone :

Yves Chaufour, [10.03.20 16:03]
Here are the output of "sudo python -m telemetry.server gs wlan00" :

on drone :

```
2020-03-10 14:56:17+0000 [-] Log opened.
2020-03-10 14:56:17+0000 [-] # iw reg set BO
2020-03-10 14:56:17+0000 [-] # ifconfig wlan0 down
2020-03-10 14:56:17+0000 [-] # iw dev wlan0 set monitor otherbss
2020-03-10 14:56:17+0000 [-] # ifconfig wlan0 up
2020-03-10 14:56:17+0000 [-] # iw dev wlan0 set channel 161 HT20
2020-03-10 14:56:17+0000 [-] Connect telem stream 2(RX), 1(TX) to 127.0.0.1:14550
2020-03-10 14:56:17+0000 [-] UDPProxyProtocol starting on 58782
2020-03-10 14:56:17+0000 [-] UDPProxyProtocol starting on 14600
2020-03-10 14:56:17+0000 [-] UDPProxyProtocol starting on 50385
2020-03-10 14:56:17+0000 [-] Telem RX: /usr/bin/wfb_rx -p 2 -u 14600 -K /etc/gs.key -k 1 -n 2 wlan0
2020-03-10 14:56:17+0000 [-] Telem TX: /usr/bin/wfb_tx -p 1 -u 14601 -K /etc/gs.key -B 20 -G long -S 1 -L 0 -M 1 -k 1 -n 2 wlan0
2020-03-10 14:56:17+0000 [-] AntennaFactory starting on 8001
2020-03-10 14:56:17+0000 [-] Started telem rx
2020-03-10 14:56:17+0000 [-] Started telem tx
2020-03-10 14:56:17+0000 [-] Send video stream 3 to 127.0.0.1:5600
2020-03-10 14:56:17+0000 [-] AntennaFactory starting on 8002
2020-03-10 14:56:17+0000 [-] Started video rx
2020-03-10 14:56:17+0000 [-] Video: /usr/bin/wfb_rx -p 3 -c 127.0.0.1 -u 5600 -K /etc/gs.key -k 8 -n 12 wlan0
2020-03-10 14:56:17+0000 [-] Stopping reactor due to fatal error: [Errno 2] No such file or directory: '/proc/net/psched'
2020-03-10 14:56:17+0000 [UDPProxyProtocol (UDP)] (UDP Port 50385 Closed)
2020-03-10 14:56:17+0000 [main.AntennaFactory] (TCP Port 8002 Closed)
2020-03-10 14:56:17+0000 [UDPProxyProtocol (UDP)] (UDP Port 14600 Closed)
2020-03-10 14:56:17+0000 [UDPProxyProtocol (UDP)] (UDP Port 58782 Closed)
2020-03-10 14:56:17+0000 [main.AntennaFactory] (TCP Port 8001 Closed)
2020-03-10 14:56:17+0000 [-] Stopped RX telem rx with code 1
2020-03-10 14:56:17+0000 [-] Stopped RX video rx with code 1
2020-03-10 14:56:18+0000 [-] Unhandled error in Deferred:
2020-03-10 14:56:18+0000 [-] Unhandled Error
        Traceback (most recent call last):
          File "/usr/lib/python2.7/dist-packages/twisted/internet/defer.py", line 457, in callback
            self._startRunCallbacks(result)
          File "/usr/lib/python2.7/dist-packages/twisted/internet/defer.py", line 565, in _startRunCallbacks
            self._runCallbacks()
          File "/usr/lib/python2.7/dist-packages/twisted/internet/defer.py", line 651, in _runCallbacks
            current.result = callback(current.result, *args, **kw)
          File "/usr/lib/python2.7/dist-packages/telemetry/server.py", line 319, in _init_services
            defer.maybeDeferred(init_tunnel, profile, wlans)])\
        --- <exception caught here> ---
          File "/usr/lib/python2.7/dist-packages/twisted/internet/defer.py", line 150, in maybeDeferred
            result = f(*args, **kw)
          File "/usr/lib/python2.7/dist-packages/telemetry/server.py", line 435, in init_tunnel
            tun_ep = TUNTAPTransport(reactor, p_in, cfg.ifname, cfg.ifaddr, mtu=settings.common.radio_mtu)
          File "/usr/lib/python2.7/dist-packages/telemetry/tuntap.py", line 61, in init
            with closing(IPRoute()) as ip:
          File "/usr/lib/python2.7/dist-packages/pyroute2/__init__.py", line 89, in new
            cls.register(cls.registry['class'])
          File "/usr/lib/python2.7/dist-packages/pyroute2/__init__.py", line 69, in getitem
            [name], 0)
          File "/usr/lib/python2.7/dist-packages/pyroute2/iproute.py", line 109, in <module>
            from pyroute2.netlink.rtnl.tcmsg import get_codel_parameters
          File "/usr/lib/python2.7/dist-packages/pyroute2/netlink/rtnl/tcmsg.py", line 31, in <module>
            _psched = open('/proc/net/psched', 'r')
        exceptions.IOError: [Errno 2] No such file or directory: '/proc/net/psched'
2020-03-10 14:56:18+0000 [-] Unhandled error in Deferred:
2020-03-10 14:56:18+0000 [-] Unhandled Error
        Traceback (most recent call last):
        Failure: twisted.internet.error.ProcessTerminated: A process has ended with a probable error condition: process ended with exit code 1.

2020-03-10 14:56:18+0000 [-] Stopped TX telem tx with code 1
2020-03-10 14:56:18+0000 [-] Unhandled error in Deferred:
2020-03-10 14:56:18+0000 [-] Unhandled Error
        Traceback (most recent call last):
        Failure: twisted.internet.error.ProcessTerminated: A process has ended with a probable error condition: process ended with exit code 1.

```
on gs :

```
2020-03-10 14:52:49+0000 [-] Log opened.                                                                              │
2020-03-10 14:52:49+0000 [-] # iw reg set BO                                                                          │
2020-03-10 14:52:49+0000 [-] # ifconfig wlan0 down                                                                    │
2020-03-10 14:52:49+0000 [-] # iw dev wlan0 set monitor otherbss                                                      │
2020-03-10 14:52:49+0000 [-] # ifconfig wlan0 up                                                                      │
2020-03-10 14:52:49+0000 [-] # iw dev wlan0 set channel 161 HT20──────────────────────────────────────────────────────┘
2020-03-10 14:52:49+0000 [-] Connect telem stream 2(RX), 1(TX) to 127.0.0.1:14550─────────────────────────────────────┐
2020-03-10 14:52:49+0000 [-] UDPProxyProtocol starting on 38488Connection refused.                                    │
2020-03-10 14:52:49+0000 [-] UDPProxyProtocol starting on 14600                                                       │
2020-03-10 14:52:49+0000 [-] UDPProxyProtocol starting on 52844                                                       │
2020-03-10 14:52:49+0000 [-] Telem RX: /usr/bin/wfb_rx -p 2 -u 14600 -K /etc/gs.key -k 1 -n 2 wlan0                   │
2020-03-10 14:52:49+0000 [-] Telem TX: /usr/bin/wfb_tx -p 1 -u 14601 -K /etc/gs.key -B 20 -G long -S 2 -L 0 -M 1 -k 1 -n 2 wlan0                                                                                                             │
2020-03-10 14:52:49+0000 [-] AntennaFactory starting on 8001                                                          │
2020-03-10 14:52:49+0000 [-] Started telem rx                                                                         │
2020-03-10 14:52:49+0000 [-] Started telem tx                                                                         │
2020-03-10 14:52:49+0000 [-] Send video stream 3 to 127.0.0.1:5600                                                    │
2020-03-10 14:52:49+0000 [-] AntennaFactory starting on 8002                                                          │
2020-03-10 14:52:49+0000 [-] Started video rx─────────────────────────────────────────────────────────────────────────┘
2020-03-10 14:52:49+0000 [-] Video: /usr/bin/wfb_rx -p 3 -c 127.0.0.1 -u 5600 -K /etc/gs.key -k 8 -n 12 wlan0
2020-03-10 14:52:50+0000 [-] Stopping reactor due to fatal error: [Errno 2] No such file or directory: '/proc/net/psched'
2020-03-10 14:52:50+0000 [UDPProxyProtocol (UDP)] (UDP Port 52844 Closed)
2020-03-10 14:52:50+0000 [main.AntennaFactory] (TCP Port 8001 Closed)
2020-03-10 14:52:50+0000 [main.AntennaFactory] (TCP Port 8002 Closed)
2020-03-10 14:52:50+0000 [UDPProxyProtocol (UDP)] (UDP Port 14600 Closed)
2020-03-10 14:52:50+0000 [UDPProxyProtocol (UDP)] (UDP Port 38488 Closed)
2020-03-10 14:52:50+0000 [-] Main loop terminated.
2020-03-10 14:52:50+0000 [-] Exiting with code 1
```

Yes21, 2020-03-10:
```
File "/usr/lib/python2.7/dist-packages/pyroute2/netlink/rtnl/tcmsg.py", line 31, in <module>
            _psched = open('/proc/net/psched', 'r')
        exceptions.IOError: [Errno 2] No such file or directory: '/proc/net/psched'
```

It seems that I'm missing "/proc/net/psched".
Don't now what it means ...

svpcom, 2020-03-10:
https://github.com/svinota/pyroute2/issues/245
this means that your custom kernel doesn't include CONFIG_NET_SCHED setting. 
There are two ways:
1. Set CONFIG_NET_SCHED=y and rebuild kernel
2. Use updated version of  python-pyroute2  (Try to run: pip install --upgrade pyroute2)

Yes21, 2020-03-10:
Ok Thanks.
I think I'm able to do this ...

Yes21, 2020-03-10:
I'm now building a new base image with kernels compiled with CONFIG_NET_SCHED=y

Yes21, 2020-03-11:
Not enough ...

Now with : 
CONFIG_NET_SCHED=y
CONFIG_NET_SCH_FIFO=y
CONFIG_PROC_CHILDREN=y

Yes21, 2020-03-12:
I've now a "/proc/net/psched" file and the related errors have disappeared.
But it's not working, and I can't understand why.

So I give up.
It's really too difficult for me ...

svpcom, 2020-03-12:
What is program output?

closed 2021-02-06 by svpcom