svpcom/wfb-ng #56: Using cards with RTL8812BU chipset

massimilianonball, 2020-03-19:
I use svpcom/wifibroadcast in a 2.4 GHz configurations from months until now without a problem with a pi 3b+ on the drone and another pi 3b+ (or alternatively with a laptop running ubuntu 18.04) as groundstation.

As in the object, i have problem with a couple of 5.8GHz china's wifi adapters that i have recently purchased; the problem is that the description of the products reports RTL8812au as chipset, while i have found that they are the 8812b version..

I have used this drivers below and the wifi adapters are recognized from the system but they won't work with svpcom/wifibroadcast:
https://github.com/cilynx/rtl88x2bu

I paste below some data on my systems (the pi and the laptop as groundstations), on the wifi cards and on the errors.

Is there a way to have those cards working with svpcom/wifibroadcast?

Many thanks!!

`lsusb:
Bus 002 Device 002: ID 0bda:b812 Realtek Semiconductor Corp. `

Pi groundstation running buster and regularly updated :
`? wifibroadcast@gs.service - WFB profile gs
   Loaded: loaded (/lib/systemd/system/wifibroadcast@.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2020-03-18 15:59:26 CET; 4s ago
  Process: 1150 ExecStart=/usr/bin/python -m telemetry.server gs ${WFB_NICS} (code=exited, status=1/FAILURE)
 Main PID: 1150 (code=exited, status=1/FAILURE)

Mar 18 15:59:26 GroundPI systemd[1]: wifibroadcast@gs.service: Failed with result 'exit-code'.

-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit wifibroadcast@gs.service has finished shutting down.
Mar 18 16:00:22 GroundPI systemd[1]: Started WFB profile gs.
-- Subject: Unit wifibroadcast@gs.service has finished start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit wifibroadcast@gs.service has finished starting up.
-- 
-- The start-up result is done.
Mar 18 16:00:24 GroundPI python[1296]: 2020-03-18 16:00:24+0100 [-] Log opened.
Mar 18 16:00:24 GroundPI python[1296]: 2020-03-18 16:00:24+0100 [-] # iw reg set BO
Mar 18 16:00:24 GroundPI python[1296]: 2020-03-18 16:00:24+0100 [-] # ifconfig wlan1 down
Mar 18 16:00:24 GroundPI python[1296]: 2020-03-18 16:00:24+0100 [-] command failed: Operation not supported (-95)
Mar 18 16:00:24 GroundPI python[1296]: 2020-03-18 16:00:24+0100 [-] Stopping reactor due to fatal error: RC 161: iw dev wlan1 set monitor otherbss
Mar 18 16:00:24 GroundPI python[1296]: 2020-03-18 16:00:24+0100 [-] Main loop terminated.
Mar 18 16:00:24 GroundPI python[1296]: 2020-03-18 16:00:24+0100 [-] Exiting with code 1
Mar 18 16:00:24 GroundPI systemd[1]: wifibroadcast@gs.service: Main process exited, code=exited, status=1/FAILURE
Mar 18 16:00:24 GroundPI systemd[1]: wifibroadcast@gs.service: Unit entered failed state.
Mar 18 16:00:24 GroundPI systemd[1]: wifibroadcast@gs.service: Failed with result 'exit-code'.
Mar 18 16:00:29 GroundPI systemd[1]: wifibroadcast@gs.service: Service hold-off time over, scheduling restart.
Mar 18 16:00:29 GroundPI systemd[1]: Stopped WFB profile gs.
-- Subject: Unit wifibroadcast@gs.service has finished shutting down
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit wifibroadcast@gs.service has finished shutting down.`

..and ubuntu laptop as groundstation:
`cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04.4 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.4 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

uname -a
Linux cacilinux 5.3.0-42-generic #34~18.04.1-Ubuntu SMP Fri Feb 28 13:42:26 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

massimiliano@cacilinux:~$ sudo systemctl status wifibroadcast@gs
? wifibroadcast@gs.service - WFB profile gs
   Loaded: loaded (/lib/systemd/system/wifibroadcast@.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2020-03-18 16:17:56 CET; 3s ago
  Process: 1340 ExecStart=/usr/bin/python -m telemetry.server gs ${WFB_NICS} (code=exited, status=1/FAILURE)
 Main PID: 1340 (code=exited, status=1/FAILURE)

 -- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unit wifibroadcast@gs.service has finished shutting down.
mar 18 16:18:07 cacilinux systemd[1]: Started WFB profile gs.
-- Subject: Unit wifibroadcast@gs.service has finished start-up
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unit wifibroadcast@gs.service has finished starting up.
-- 
-- The start-up result is RESULT.
mar 18 16:18:08 cacilinux python[1371]: 2020-03-18 16:18:08+0100 [-] Log opened.
mar 18 16:18:08 cacilinux python[1371]: 2020-03-18 16:18:08+0100 [-] # iw reg set BO
mar 18 16:18:08 cacilinux python[1371]: 2020-03-18 16:18:08+0100 [-] # ifconfig wlx000f00bd1ba9 down
mar 18 16:18:08 cacilinux python[1371]: 2020-03-18 16:18:08+0100 [-] command failed: Operation not supported (-95)
mar 18 16:18:08 cacilinux python[1371]: 2020-03-18 16:18:08+0100 [-] Stopping reactor due to fatal error: RC 161: iw dev wlx000f00bd1ba9 set monitor otherbss
mar 18 16:18:08 cacilinux python[1371]: 2020-03-18 16:18:08+0100 [-] Main loop terminated.
mar 18 16:18:08 cacilinux python[1371]: 2020-03-18 16:18:08+0100 [-] Exiting with code 1
mar 18 16:18:08 cacilinux systemd[1]: wifibroadcast@gs.service: Main process exited, code=exited, status=1/FAILURE
mar 18 16:18:08 cacilinux systemd[1]: wifibroadcast@gs.service: Failed with result 'exit-code'.
mar 18 16:18:12 cacilinux systemd[1]: Stopped WFB profile gs.
-- Subject: Unit wifibroadcast@gs.service has finished shutting down
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unit wifibroadcast@gs.service has finished shutting down.`

svpcom, 2020-03-19:
Your driver doesn't support packet injection:
```
 iw dev wlan1 set monitor otherbss
```
The easiest solution is to try to refund cards (especially if they are not 8812au).

massimilianonball, 2020-03-19:
What a pity ..i was hoping for a workaround.. 

Ok i'll try with the refound, manu thanks!!

closed 2021-02-06 by svpcom