svpcom/wfb-ng #59: Higher bit rate video TX not working with 2.4GHz

tmandal1988, 2020-04-25:
Hi,
I followed the guide at https://github.com/svpcom/wifibroadcast/wiki/Navio2-Setup to install wifibroadcast on Navio + Raspberry Pi 3. I am using 2.4GHz Alfa AWUS036NHA Wifi cards (with Atheros AR9271 chipset) as both TX and RX modules. My /etc/wifibroadcast.cfg on drone side looks like following
`[common]
wifi_channel = 1       # 161 -- radio channel @5825 MHz, range: 5815–5835 MHz, width 20MHz
                       # 1 -- radio channel @2412 Mhz, 
                       # see https://en.wikipedia.org/wiki/List_of_WLAN_channels for reference
wifi_region = 'BO'     # Your country for CRDA (use BO or GY if you want max tx power)

[drone_mavlink]
peer = 'listen://0.0.0.0:14550'   # incoming connection
#peer = 'connect://127.0.0.1:14550'  # outgoing connection

[drone_video]
peer = 'listen://0.0.0.0:5602'  # listen for video stream (gstreamer on drone)
bandwidth = 40     # bandwidth 20 or 40 MHz
short_gi = True   # use short GI or not
stbc = 3          # stbc streams: 1, 2, 3 or 0 if unused
ldpc = 0
mcs_index = 23      # mcs index`

The most I could transmit without greying out and excessive lag was 400,00 bit rate and 60 fps. Has anyone else experienced such limitations with 2.4GHz or is there a way to increase the TX throughput for 2.4GHz system or I am at the limit?

Mysnomer0, 2020-04-25:
Hi,

I'm glad you found the guide helpful, I wrote it back in January. Yes, unfortunately, 400,00 bit rate at 60 fps sounds about right. In fact, I had to drop it considerably more to bring the video latency down enough to fly with. Check this issue out, a number of us have had that same problem:

https://github.com/svpcom/wifibroadcast/issues/46

Thanks,

Milan

tmandal1988, 2020-04-25:
Hey Milan,
Thanks for creating the super helpful guide (it worked without any issues).Do you think using a 5.8GHz system will be helpful?

Mysnomer0, 2020-04-25:
Haha, thanks for the great feedback, you're one of the first to try it out. :) That's a great question. I've been wanting to try a 5.8ghz system, it's in my backlog of drone projects. I think the increased bandwidth would provide excellent video with low latency, but the link would not be anywhere near as resilient to weather and long range as 2.4ghz. Still, I think swapping out the cards for a Alfa AWUS036ACH or something similar would be cool. Let us know what comes of your work!

closed 2020-07-26 by svpcom