svpcom/wfb-ng #61: openwrt package

booo, 2020-05-06:
Hi,

first of all thanks for your work! I wrote my own packet injection code and your radiotap header enabled me to inject with HT rates.

I created a openwrt package for your project:

https://github.com/booo/openwrt-wifibroadcast

I still have to test the code. In the original wifibroadcast code there seams to be a problem with the pcap filters and non standard header format. Receiving packets does not work on openwrt targets.

With wifibroadcast for openwrt you can run wifibroadcast on a lot of popular wifi routers such as the atheros based hardware from ubnt.

Best Regards
Philipp

svpcom, 2020-05-12:
Thanks!
Will you submit it to openwrt upstream?

booo, 2020-05-12:
Once everything is working I will definitely consider it.

closed 2020-07-26 by svpcom