svpcom/wfb-ng #62: make deb issue

g5554, 2020-07-25:
I receive this error when trying to 'make deb' on Parrot OS: 

"dh: error: unable to load addon python3: Can't locate Debian/Debhelper/Sequence/python3.pm"

does anyone know what the cause is and how to rectify it? Below is the relevant output. Thanks:

make[1]: Entering directory '/home/g/wifibroadcast/deb_dist/wifibroadcast-20.7.24.69434'
dh clean --with python3 --buildsystem=python_distutils
dh: warning: Compatibility levels before 10 are deprecated (level 7 in use)
dh: error: unable to load addon python3: Can't locate Debian/Debhelper/Sequence/python3.pm in @INC (you may need to install the Debian::Debhelper::Sequence::python3 module) (@INC contains: /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.30.0 /usr/local/share/perl/5.30.0 /usr/lib/x86_64-linux-gnu/perl5/5.30 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.30 /usr/share/perl/5.30 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base) at (eval 3) line 1.
BEGIN failed--compilation aborted at (eval 3) line 1.

make[1]: *** [debian/rules:7: clean] Error 255
make[1]: Leaving directory '/home/g/wifibroadcast/deb_dist/wifibroadcast-20.7.24.69434'
dpkg-buildpackage: error: fakeroot debian/rules clean subprocess returned exit status 2
Traceback (most recent call last):
  File "./setup.py", line 62, in <module>
    setup(
  File "/home/g/wifibroadcast/env/lib/python3.8/site-packages/setuptools/__init__.py", line 165, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.8/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/g/wifibroadcast/env/lib/python3.8/site-packages/stdeb/command/bdist_deb.py", line 33, in run
    self.run_command('sdist_dsc')
  File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/g/wifibroadcast/env/lib/python3.8/site-packages/stdeb/command/sdist_dsc.py", line 135, in run
    build_dsc(debinfo,
  File "/home/g/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 1468, in build_dsc
    dpkg_buildpackage(*args, cwd=fullpath_repackaged_dirname)
  File "/home/g/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 534, in dpkg_buildpackage
    process_command(args, cwd=cwd)
  File "/home/g/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 201, in process_command
    check_call(args, cwd=cwd)
  File "/home/g/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 46, in check_call
    raise CalledProcessError(retcode)
stdeb.util.CalledProcessError: 2
make: *** [Makefile:52: deb] Error 1

svpcom, 2020-07-26:
What linux distro do you use? I use Ubuntu 18.04. Check what python version is in virtualenv:
```
$ ./env/bin/python --version
Python 2.7.17
```

g5554, 2020-07-26:
@svpcom I'm using Parrot OS (Debain based but with a focus on security). Seems that it is python 3.8.3, should it be python 2? I had some trouble getting some of the packages with apt, so used pip - could this cause some issues? Thanks for your help btw

svpcom, 2020-07-29:
Yes, you should use python2 (replace `virtualenv env` to `virtualenv env --python=python2`) and do `make env`

g5554, 2020-07-29:
Thanks, have tried that but it returns the same error when building. Maybe python-twisted failed to install/ is installed in a different place?

iamfaith, 2021-01-30:
> Thanks, have tried that but it returns the same error when building. Maybe python-twisted failed to install/ is installed in a different place? (I was following the Setup from scratch page, but apt-get python-twisted failed)

same here, how do you solve this issue?

closed 2021-02-06 by svpcom