svpcom/wfb-ng #64: fcsfail not set

julled, 2020-10-16:
Hi vasily,

thanks for your nice library, it helps me a lot in getting video down from our drone to my groundstation.

Today we did a test to check how far we can get using  TX / R11e-2HPnD @30dbm and RX / Ubiquiti AM-2G15-120 + 2x Tl-wn722n. We only cam about 1km at a free field with the RX 4m over ground.

We were wondering why at some point (> 1km)   no packets anymore coming through to the RX PC / WFB. Also tcpdump showed no packets anymore. 
Could this be, because the "fcsfail" is not set by your script by default, and slightly broken packets are not let through? Other wfb implementations set this in there toolchain. Is there a reason for you to not set it? 

Sadly we didnt test it today to just set it and see whats happing, but maybe you could give a small bit of info already for your reasons to not set it.

Thank you so much!

svpcom, 2020-10-16:
Hi, what freq band (2.4 or 5.8) and antennas do you use? Check your setup with Friis Equation (https://en.wikipedia.org/wiki/Friis_transmission_equation)

Fcsfail will not help. h264 stream is not like analog video transmission. Also WFB use encryption so corrupted packets will not decrypted. We use forward error correction (based on reed-solomon codes) to deal with bad and/or missed packets.
I'll suggest following:
1. Check you setup (friis equation). Use directed antenna(s) on GS. Circular polarized antennas preferred. 
2. Check that used wifi channel is not used (especially for 2.4 band)
3. Enable LDPC codes (additional FEC supported by rtl8812au)
4. Use lower mcs_index (0 or 1). WiFi card has better sensitivity on smaller speeds
5. Don't use 40MHz bandwidth (use 20 instead).

julled, 2020-11-23:
Thanks for your reply. i began to understand that it doesnt make sense to use broken pakets (as it cannot be decrypted anymore). It may make sense for the original WFB, but there its also not clear whether actually broken pakets are used if they cant be repaired.

1. why do you prefer circular polarized antennas? Because better isolation against noise from a particular polarisation? 
2. we can confirm that if the actual channel is used a lot , the interface is not getting a slot in the shared medium and therefore the injectionrate is low.
4. yes lower mcs seems to be much better suited for long range as i found out in the datasheet (better sensitivity)
5. did you try to use even lower bandwidths, like 5MHz? As there the wifi interfaces seem to have even higher RX sensitivity.

svpcom, 2020-11-24:
1. No interference with reflected signals. Works with any position of copter vs ground station.
2. Yes. CSMA/CA still active in WFB
5. Official supported wifi bitrates are 20 and 40 mhz. my hardware doesn't support 5 and 10 mhz bandwidth.

julled, 2020-12-13:
We have just realized, that the radiotap parsing for  in the kernel doesnt parse LDPC and STBC info:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/net/mac80211/tx.c#n2110
Is it possible that this might be also the case for your setup or did you explicitly set this info otherwise via 8812au driver?

svpcom, 2020-12-13:
8812au driver has its own parser (independent from mac80211). See hal/rtl8812a/usb/rtl8812au_xmit.c in https://github.com/svpcom/rtl8812au

closed 2021-02-06 by svpcom