svpcom/wfb-ng #67: Dev Question - is wpacket_hdr_t redundant (packet_size)

Consti10, 2020-12-04:
Hello,
What's the reasoning behind writing the packet size in each wb packet ?

I'd guess it is redundant, since the packet size can be calculated on rx by substracting header from packet ?

This way we get 2 more bytes into each packet.

Consti10, 2020-12-04:
I think it is an relict of how you wrote the fec encoder / decoder part.
But there it can be stored somewhere else, and must not be included in the actual packet going out.

Consti10, 2020-12-04:
I think I understand now why the packet size is written with each fec packet:
If a data packet is lost, and this data packet is reconstructed by FEC, the information where the actual data of this packet is and where the '0' padding begins cannot be recovered.
So this information needs to be reconstructed with the FEC step, and therefore needs to stand in the packet

svpcom, 2020-12-04:
yes

closed 2020-12-04 by svpcom

Consti10, 2020-12-04:
Pretty smart :)