svpcom/wfb-ng #70: Dev / Paper Question

Consti10, 2020-12-09:
Hello,
In your paper you use the "Moep library" for testing.
However, your tx / rx uses pcap for packet injection.
I am wondering what's the reason behind this decision.
Because on my Ubuntu PC pcap seems to create roughly 6ms additional latency and I am wondering if moep would be better in this case.

svpcom, 2020-12-09:
This is not my paper. I use pcap due to simplicity. You can use raw sockets it you need to reduce latency to a few ms. Pull requests are welcome :-)

Consti10, 2020-12-10:
Hi,
Thanks for all your quick responses.
I have spent the last couple of days modifying your wifibroadcast library with the following in mind:
1) modularize the encryption, fec and pcap parts as much as possible, since the shier amount of lines of code inside your rx/tx implementation is a bit overhelming
2) use c++ - style as much as possible even though some compromises have been made since some of the libraries used are still c-style
3) add more documentation as I am trying to understand what your intention(s) were when writing the code.
The goal from the beginning was to provide OpenHD with a clean svpcom_wifibroadcast tx/rx implementation that can become more specialized moving forward.

However, one compromise was made:
Your 'advanced setup' with multiple boards for tx and rx seemed a bit too complicated for OpenHD and also obsolete with the recent improvements on single board computers (like the rpi 4)
Therefore, I removed your custom 'forwarding of raw packets via udp' functionality, even though it could obviosly be added again if the need arises.

Here is the code, If you have any feedback / see some obvious bugs, please let me know:
https://github.com/Consti10/wifibroadcast

closed 2020-12-10 by Consti10

svpcom, 2020-12-15:
@Consti10 I'll try to explain why I implement it in minimal C++ (but mostly in plain C)
1. No memory fragmentation. Memory allocated only on startup.
2. All operations are in-place to reduce data moving.  recvmsg/sendmsg used for collect packets from multiple memory buffers.
3. Code is simple for reading and maximum portable (can be compiled with old gcc)
4. Advanced setup used by some users to have DIY "cellular network" in the forest for video translation for orienteering competitions.

Consti10, 2020-12-15:
The discussion "C is better at memory management" - well, I don't agree, but thats a different topic :)
But I understand your reasoning behind using minimal c++.

There is one memory allocation at run time we both have
https://github.com/svpcom/wifibroadcast/blob/master/src/rx.cpp#L471

I never bothered to check if crypto_aead_chacha20poly1305_decrypt can safely take the same memory pointer for input and output or if it needs a different one.
But in my experince allocating on the stack doesn't matter for the level of real-time we are going for as long as you don't leak anything.