svpcom/wfb-ng #71: Optimization - use actual size used for fec encoding on RX instead of MAX_PACKET_SIZE

Consti10, 2020-12-10:
Hello,
In your rx implementation you use MAX_PACKET_SIZE when reconstructing primary FEC fragments with secondary fragments:
https://github.com/svpcom/wifibroadcast/blob/master/src/rx.cpp#L616

However, I am wondering if it wouldn't be better to use the same size that was used on the tx.
Since each FEC secondary packet has the size of max_packet_size (as calculated on tx for this block) the information should be available in the apply_fec step.

I'd use omething like this (hope this example makes my intentions clearer:
https://github.com/Consti10/wifibroadcast/blob/master/src/FEC.hpp#L196

Consti10, 2020-12-10:
I am not sure how fec even works when using different values for the last "sz" param in tx and rx but aparently the magic math behind doesn't care (probably due to the 'zeroing' trick)

svpcom, 2020-12-15:
FEC needs that all packets are with equal size. When tx we known all packet sizes and send first K packets with their original sizes (to reduce bandwidth), but when calculating FEC, we pad them to max(size_of_first_k_packets). So only (n-k) (i.e. only FEC packets) will be large size. But when rx we don't known max_packet_size (because we can miss one of first K packets) so we can use MAC_FEC_PAYLOAD for zero padding and fec decoding.  Yes, you can optimize it and use size of FEC packets. I don't remember why I use MAC_FEC_PAYLOAD instead of it.

svpcom, 2020-12-15:
See https://github.com/tahoe-lafs/zfec for FEC docs

Consti10, 2020-12-15:
> Yes, you can optimize it and use size of FEC packets. I don't remember why I use MAC_FEC_PAYLOAD instead of it.

For some reason applyFec() can be called even though no fec packets are available.
https://github.com/Consti10/wifibroadcast/blob/master/src/FEC.hpp#L218
I didn't bother to look for the reason in rx ring and rather used a simple workaround
https://github.com/Consti10/wifibroadcast/blob/master/src/FEC.hpp#L222

closed 2021-02-06 by svpcom