svpcom/wfb-ng #72: How does the RX_RING work

Consti10, 2020-12-10:
Hello,
I don't fully understand yet how your Rx ring works.
Can you tell me what these 2 do:
1) https://github.com/svpcom/wifibroadcast/blob/master/src/rx.hpp#L129
get_block_ring_idx()
2) https://github.com/svpcom/wifibroadcast/blob/master/src/rx.hpp#L130
rx_ring_push()

Btw, I've timestamped the time points when packets enter and when they leave the rx ring.
With high packet loss a primary fragment can stay up to 15ms in the rx ring.
Could that be improved ?

svpcom, 2020-12-15:
Ring buffer (rx ring) is used because rx packets can arrive out of order (especially if multiple cards used). So we need to accumulate them in some place and yield full (or partial if we have sufficient number of parts for FEC) packet.
If we get fragments in-order (without drop) we don't need to wait for FEC (n-k) packets and can yield fragments without delay. If we have one or more dropouts then we must wait for all other fragments to recover missed parts from FEC.

Consti10, 2020-12-15:
If too many fragments are lost such that a block cannot be fully recovered -
At which point does the RX ring forward the next block with valid data in this case ?

svpcom, 2020-12-15:
If there is next block with enough fragments to decode

closed 2021-02-06 by svpcom