svpcom/wfb-ng #73: "Link lost" with 8814au

Abu-Abdullah, 2020-12-11:
Hi,

first thank you for this library. I bought 2 asus USB-AC68 to try it in my drone for camera streaming. i followed most fo the setup steps here but eventually i got Link lost

GS
```
pi@raspberrypi:~ $ journalctl -xu wifibroadcast@gs
-- Logs begin at Fri 2020-12-11 06:06:08 GMT, end at Fri 2020-12-11 09:27:54 GMT. --
Dec 11 06:06:21 raspberrypi systemd[1]: Started WFB profile gs.
-- Subject: A start job for unit wifibroadcast@gs.service has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit wifibroadcast@gs.service has finished successfully.
--
-- The job identifier is 446.
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] Log opened.
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] # iw reg set BO
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] # ifconfig wlan1 down
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] # iw dev wlan1 set monitor otherbss
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] # ifconfig wlan1 up
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] # iw dev wlan1 set channel 161 HT20
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] Connect telem stream 2(RX), 1(TX) to 127.0.0.1:14550
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] UDPProxyProtocol starting on 54071
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] UDPProxyProtocol starting on 14600
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] UDPProxyProtocol starting on 34423
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] Telem RX: /usr/bin/wfb_rx -p 2 -u 14600 -K /etc/gs.key -k 1 -n 2 wlan1
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] Telem TX: /usr/bin/wfb_tx -p 1 -u 14601 -K /etc/gs.key -B 20 -G long -S 1 -L 0 -M 1 -k
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] AntennaFactory starting on 8001
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] Started telem rx
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] Started telem tx
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] Send video stream 3 to 127.0.0.1:5600
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] AntennaFactory starting on 8002
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] Started video rx
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] Video: /usr/bin/wfb_rx -p 3 -c 127.0.0.1 -u 5600 -K /etc/gs.key -k 8 -n 12 wlan1
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] UDPProxyProtocol starting on 14800
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] UDPProxyProtocol starting on 50816
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] Tunnel RX: /usr/bin/wfb_rx -p 5 -u 14800 -K /etc/gs.key -k 1 -n 2 wlan1
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] Tunnel TX: /usr/bin/wfb_tx -p 4 -u 14801 -K /etc/gs.key -B 20 -G long -S 1 -L 0 -M 1 -k
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] AntennaFactory starting on 8003
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] Started tunnel rx
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] Started tunnel tx
Dec 11 06:06:24 raspberrypi python[621]: 2020-12-11 06:06:24+0000 [-] telem tx: Listen on 14601 for wlan1
Dec 11 06:06:25 raspberrypi python[621]: 2020-12-11 06:06:25+0000 [-] tunnel tx: Listen on 14801 for wlan1
Dec 11 06:06:25 raspberrypi python[621]: 2020-12-11 06:06:25+0000 [-] telem rx: wlan1 has DLT_IEEE802_11_RADIO Encap
Dec 11 06:06:25 raspberrypi python[621]: 2020-12-11 06:06:25+0000 [-] tunnel rx: wlan1 has DLT_IEEE802_11_RADIO Encap
Dec 11 06:06:25 raspberrypi python[621]: 2020-12-11 06:06:25+0000 [-] video rx: wlan1 has DLT_IEEE802_11_RADIO Encap
```

Drone
```
pi@raspberrypi:~ $ journalctl -xu wifibroadcast@drone
-- Logs begin at Fri 2020-12-11 06:06:04 GMT, end at Fri 2020-12-11 09:27:54 GMT. --
Dec 11 06:06:18 raspberrypi systemd[1]: Started WFB profile drone.
-- Subject: A start job for unit wifibroadcast@drone.service has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit wifibroadcast@drone.service has finished successfully.
--
-- The job identifier is 443.
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] Log opened.
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] # iw reg set BO
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] # ifconfig wlan1 down
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] # iw dev wlan1 set monitor otherbss
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] # ifconfig wlan1 up
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] # iw dev wlan1 set channel 161 HT20
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] Listen for telem stream 1(RX), 2(TX) on 0.0.0.0:14550
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] UDPProxyProtocol starting on 14550
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] UDPProxyProtocol starting on 14700
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] UDPProxyProtocol starting on 36170
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] Telem RX: /usr/bin/wfb_rx -p 1 -u 14700 -K /etc/drone.key -k 1 -n 2 wlan1
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] Telem TX: /usr/bin/wfb_tx -p 2 -u 14701 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] Started telem rx
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] Started telem tx
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] Listen for video stream 3 on 0.0.0.0:5602
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] Started video tx
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] Video: /usr/bin/wfb_tx -p 3 -u 5602 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 -k 8
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] UDPProxyProtocol starting on 14900
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] UDPProxyProtocol starting on 54925
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] Tunnel RX: /usr/bin/wfb_rx -p 4 -u 14900 -K /etc/drone.key -k 1 -n 2 wlan1
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] Tunnel TX: /usr/bin/wfb_tx -p 5 -u 14901 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] Started tunnel rx
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] Started tunnel tx
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] telem tx: Listen on 14701 for wlan1
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] video tx: Listen on 5602 for wlan1
Dec 11 06:06:21 raspberrypi python[619]: 2020-12-11 06:06:21+0000 [-] tunnel tx: Listen on 14901 for wlan1
Dec 11 06:06:22 raspberrypi python[619]: 2020-12-11 06:06:22+0000 [-] telem rx: wlan1 has DLT_IEEE802_11_RADIO Encap
Dec 11 06:06:22 raspberrypi python[619]: 2020-12-11 06:06:22+0000 [-] tunnel rx: wlan1 has DLT_IEEE802_11_RADIO Encap
```

i compiled the driver from the src:
https://github.com/aircrack-ng/rtl8814au

steps in raspberry lite:
```
$ sudo apt-get update
$ sudo apt-get upgrade
$ sudo apt-get install git raspberrypi-kernel-headers build-essential dkms
$ git clone https://github.com/aircrack-ng/rtl8814au.git
$ cd rtl8814au

# modify Makefile and update the line:
# dkms install -m 8814au -v $(DRIVER_VERSION)
# to
# dkms install --force -m 8814au -v $(DRIVER_VERSION)

$ ARCH=arm make
$ sudo make install

$ sudo ARCH=arm make dkms_install
```

any help is appreciated

svpcom, 2020-12-11:
Hi.  I had a bad experience with  rtl8814au. I use only rtl8812au with https://github.com/svpcom/rtl8812au drivers.

Abu-Abdullah, 2020-12-11:
thank you for the quick response. i wish i knew before buying those modules.

i will try to look further into this and go with another purchase if i didn't reach anywhere.

closed 2020-12-11 by Abu-Abdullah