svpcom/wfb-ng #74: Decrease latency by ~8ms on pcap version 1.5.0 or higher

Consti10, 2020-12-14:
Hello,
Just wanted to let you know that I got rid of 8ms latency in the tx-rx pipeline just by enabling immediate mode in libpcap.
They silently changed immediate mode to default=off starting with version 1.5.0
Its a pretty simple fix:
https://github.com/Consti10/wifibroadcast/blob/master/src/rx.cpp#L83

svpcom, 2020-12-15:
What is 'immediate mode'? In referenced line pcap has been set to non-blocking mode. This is needed if rx are from several cards.

Consti10, 2020-12-15:
Whoops wrong line:
https://github.com/Consti10/wifibroadcast/blob/master/src/rx.cpp#L81

Consti10, 2020-12-15:
I tracked it down by using the timestamp assigned with each pcap packet if you enable
https://github.com/Consti10/wifibroadcast/blob/master/src/rx.cpp#L57

Consti10, 2020-12-15:
And I also wrote a test utility that tests end-to-end latency if you have 2 cards connected to the same pc:
https://github.com/Consti10/OpenHDTesting

Consti10, 2020-12-15:
Should be as low as 1ms on average now

svpcom, 2020-12-15:
I also have such test: telemetry/latency_test.py

Consti10, 2020-12-15:
Btw, we are using my fork of your code ( that is nothing else than your ideas but in c++ and modularized) in the next major OpenHD release. If you want to switch systems / contribute to OpenHD people like you are needed !

Consti10, 2020-12-15:
If I would have known that the latency created comes from this one-liner I probably could've saved a lot of time.
But 2 weeks ago I was left with wb_tx / wb_rx : Has more latency than it should have.
To find out what's wrong I had to understand your code and the easiest way to to me was to move through it bit by bit and since I was already doing that I converted it to c++.
Also all the uint8_t* for memory chuncks that can be anything were overwhelming, so I used c++ const reference as much as possible.

closed 2021-02-06 by svpcom