svpcom/wfb-ng #75: There are examples avariable?

LuisHerasme, 2021-01-02:
Hi, I'm new to this, and I would like to know if there are examples of how to send data.
In addition to sending video, I would like to send a JSON object with some data that I read from the pins, I am not using this in a drone, but something like a remote control car and I have some sensors and actuators and I want to send data in a JSON format from the Ground PI to the AirPI. And I want to know if there is a python example on how to do this.

Mysnomer0, 2021-01-02:
I'd be curious how to do this too, for custom telemetry. I'm quite certain it's possible since we're able to stream bytes, but I'm not sure how.

svpcom, 2021-01-02:
If your telemetry protocol is resistant to messages lost  (for example mavlink) you can use UDP transport (see config for mavlink stream in HOWTO). If you telemetry needs reliable stream of bytes (like TCP socket) you need to use IPoverWB tunnel (see HOWTO). For mavlink telemetry examples see: https://github.com/mavlink-router/mavlink-router/tree/master/examples

svpcom, 2021-01-02:
You can pack json into UDP packet.

closed 2021-02-06 by svpcom