svpcom/wfb-ng #76: Problems with BPF

manumar91, 2021-01-25:
Hi, first thanks for this tool! It works quite well for me with normal hardware rtl8812au and rtl8812bu on arm and x86.
I wanted to give a try to a spare ubiquiti litebeam m5 that I have. No packets where going through so I:
1-disabled BPF: Things where coming but broken
2-updated libradiotap: Worked, tested with an rtsp stream

So what I cannot get to work is just the filtering, can it be that it is a big endian system? or the wifi card? I know nothing in this area.
With tcpdump I did tcpdump -vvv -s 96 -ni wlan0 which is supposed to show the header and I see a difference, the broken system headers seems choped:

PC:
cpdump: listening on wlan2, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 96 bytes
01:58:12.213522 [bit 15] 13:22:33:44:55:02 Unknown SSAP 0x46 > 13:22:33:44:55:02 802.1B I Information, send seq 97, rcv seq 10, Flags [Command], length 73
	0x0000:  0246 c214 3318 4307 ed40 0b8c 0760 a97f  .F..3.C..@...`..
	0x0010:  63a7 219a 6100 a3f0 d70b 1b41 89b6 1400  c.!.a......A....
	0x0020:  d941 393e cdab 9cbb 3d32 922b 8144 4ae1  .A9>....=2.+.DJ.
	0x0030:  4e7f 3515 8c86 1605 51cc c1              N.5.....Q..
01:58:12.213589 [bit 15] 13:22:33:44:55:02 > 13:22:33:44:55:02 Null Information, send seq 0, rcv seq 0, Flags [Command], length 31
	0x0000:  0100 0000 0000 0004 006a 2697 8433 ac84  .........j&..3..
	0x0010:  aedb 1790 ab93 afcb 574b c31d b575 02    ........WK...u.
01:58:12.213609 [bit 15] 13:22:33:44:55:02 > 13:22:33:44:55:02 Null Information, send seq 0, rcv seq 0, Flags [Command], length 31
	0x0000:  0100 0000 0000 0004 01cf 3900 d9f7 8ea6  ..........9.....
	0x0010:  d14e 1f4f 66d5 6db4 4163 7403 ea8b bf    .N.Of.m.Act....

Litebeam:
root@OpenWrt:~# tcpdump -vvv -s 96 -ni wlan0
tcpdump: listening on wlan0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 96 bytes
00:58:12.213245 9135012633us tsft 5825 MHz 11n -72dBm signal 13.0 Mb/s MCS 1 20 MHz long GI RX-STBC0 -72dBm signal antenna 0 13:22:33:44:55:02 Unknown SSAP 0x46 > 13:22:33:44:55:02 802.1B I Information, send seq 97, rcv seq 10, Flags [Command], length 73
	0x0000:  0246 c214 3318 4307 ed40 0b8c 0760 a97f  .F..3.C..@...`..
	0x0010:  63a7 219a 6100 a3f0 d7                   c.!.a....
00:58:12.213297 9135012795us tsft 5825 MHz 11n -74dBm signal 13.0 Mb/s MCS 1 20 MHz long GI RX-STBC0 -74dBm signal antenna 0 13:22:33:44:55:02 > 13:22:33:44:55:02 Null Information, send seq 0, rcv seq 0, Flags [Command], length 31
	0x0000:  0100 0000 0000 0004 006a 2697 8433 ac84  .........j&..3..
	0x0010:  aedb 1790 ab93 afcb 57                   ........W
00:58:12.213319 9135012931us tsft 5825 MHz 11n -27dBm signal 13.0 Mb/s MCS 1 20 MHz long GI RX-STBC0 -27dBm signal antenna 0 13:22:33:44:55:02 > 13:22:33:44:55:02 Null Information, send seq 0, rcv seq 0, Flags [Command], length 31
	0x0000:  0100 0000 0000 0004 01cf 3900 d9f7 8ea6  ..........9.....
	0x0010:  d14e 1f4f 66d5 6db4 41                   .N.Of.m.A

Do you have an idea? Thanks

svpcom, 2021-01-25:
BPF is used for filtering in libpcap. How you disable it?
Are you sure that AR9342 (which is inside) supports packet injection?
Some ath9x cards need custom firmware https://wiki.debian.org/ath9k_htc/open_firmware for that.
Also atheros need to create monitor device `wlanconfig ath1 create wlandev wifi0 wlanmode monitor`
What linux kernel version is inside this device?

manumar91, 2021-01-25:
Thanks for your quick answer.
I just commented the lines setting the filter.
I wasnt very sure about the injection so it was the first thing I tested, just echoing something with netcat, maybe I should test with something more demanding, altough mavlink which would be the only thing going out from gs is not much.
Hmmmm the new interface..That I havent done it, I just did how you have it on the code.
And Im using OpenWRT 19.07.6 with 4.14. Libpcap 1.9.1.
Btw, posting the tcpdump output was stupid. Seems capturing and looking in wireshark is easier and what I see is that the data is the same and the headers in atheros are fill with more stuff, but wireshark is able to get the ethernet src and dest in both captures so I dont know what is wrong. Maybe the filter since it seems to be applied on an address directly?

svpcom, 2021-01-25:
Ubnt may have custom wifi firmware which doesn't support injection. First of all you need to be sure that injection works. Try with https://www.aircrack-ng.org/doku.php?id=interactive_packet_replay. If it doesn't works they you can try to use open firmware mentioned above.

manumar91, 2021-01-25:
But injection is for sending no? I tested with wfb_tx on the ubnt and wfb_rx on a PC and it worked. I also just used aireplay -9 and reported injection is working!!
Something interesting I just saw: when setting monitor through iw dev wlan0 set monitor otherbss I had most of the frequencies disabled, however if doing iw phy phy0 interface add mon0 type monitor then none gets disabled.
Still same results with the filtering :(

manumar91, 2021-01-25:
```
root@OpenWrt:~# tcpdump -vve -i wlan0
tcpdump: listening on wlan0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
14:59:29.265008 181861246us tsft 5825 MHz 11n -41dBm signal 13.0 Mb/s MCS 1 20 MHz long GI RX-STBC0 -41dBm signal antenna 0 0us BSSID:Broadcast SA:13:22:33:44:55:02 (oui Unknown) DA:13:22:33:44:55:02 (oui Unknown) LLC, dsap 802.1B I (0x02) Individual, ssap Unknown (0x68) Response, ctrl 0x047d: Supervisory, ?, rcv seq 2, Flags [Response], length 73
```
So tcpdump shows the correct mac but if saying to filter by that one, nothing comes :\
`root@OpenWrt:~# tcpdump -vve -i wlan0 ether src 132233445502`

For me that works on rpi, laptop...

svpcom, 2021-01-25:
If this device is high-endian, try to fix:
``` .c
switch (link_encap)
    {
    case DLT_PRISM_HEADER:
        fprintf(stderr, "%s has DLT_PRISM_HEADER Encap\n", wlan);
        program = string_format("radio[0x4a:4]==0x13223344 && radio[0x4e:2] == 0x55%.2x", radio_port);
        break;

    case DLT_IEEE802_11_RADIO:
        fprintf(stderr, "%s has DLT_IEEE802_11_RADIO Encap\n", wlan);
        program = string_format("ether[0x0a:4]==0x13223344 && ether[0x0e:2] == 0x55%.2x", radio_port);
        break;

    default:
        throw runtime_error(string_format("unknown encapsulation on %s", wlan));
    }
```
in src/rx.cpp

svpcom, 2021-01-25:
first of all try to supply "radio[0x4a:4]==0x13223344" to tcpdump as filter. Maybe you need to convert constants

manumar91, 2021-01-25:
"ether[0x0a:4]==0x00000000" seems to give the packets......
How can it be? As I showed, without filter it shows the MAC
Do you know how to print the mac info, or in general all the header data libcap is playing with? The last 2 bytes are not zero but neither 02
I wrote the filter also as which works on pc but still no luck:
`string_format("ether src 13:22:33:44:55:%.2x", radio_port)`

svpcom, 2021-01-25:
Save packets to file via `tcpdump -w filename` and open this dump in wireshark

manumar91, 2021-01-25:
Yes I did that, and wireshark parses the packages perfect.
I think something is wrong with the parsing in libpcap. I have eventually got it to work shifting the filter 16 bytes... :-)
Let see how it works the whole thing, Ill let u know.

Thanks Vasily!

evirgenfurkan, 2021-01-29:
Hi, I am using openwrt and I have same problem I think.  I receive "Error parsing radiotap header!" message from RX side. How can I fix it with byte shifting? 
btw: tcpdump works well.

svpcom, 2021-01-29:
@evirgenfurkan 
If you know offset (i.e. if packet has some garbage before ieee80211_radiotap header), you can skip it like this:
``` .patch
diff --git a/src/rx.cpp b/src/rx.cpp
index a7a030c..bcbe691 100644
--- a/src/rx.cpp
+++ b/src/rx.cpp
@@ -125,6 +125,13 @@ void Receiver::loop_iter(void)
         int8_t rssi[RX_ANT_MAX];
         uint8_t flags = 0;
         struct ieee80211_radiotap_iterator iterator;
+
+        if(pktlen > SOME_OFFSET)
+        {
+            pktlen -= SOME_OFFSET;
+            pkt += SOME_OFFSET;
+        }
+
         int ret = ieee80211_radiotap_iterator_init(&iterator, (ieee80211_radiotap_header*)pkt, pktlen, NULL);
 
         // Fill all antenna slots with 0xff (unused)
```

svpcom, 2021-01-29:
Also try fix from ba5ede141cdef23c5444f1cca9d1b4eda661f92c

evirgenfurkan, 2021-01-29:
I tried this code but not change anything (manumar91 said 16byte shift)
        if(pktlen > 16)
        {
            pktlen -= 16;
            pkt += 16;
        }

and I also tried fix from ba5ede1 and not change anything, I get same "Error parsing radiotap header!" message. 
"int ret = ieee80211_radiotap_iterator_init" function return always -22.

svpcom, 2021-01-29:
@evirgenfurkan try fa24784ec8a97a169ad1ac9670c13c79409a6c21
it was broken on high-endian machines

svpcom, 2021-01-31:
@evirgenfurkan Does latest patch solve the problem?

svpcom, 2021-02-02:
@evirgenfurkan @manumar91 Please try latest release and report if issue fixed

evirgenfurkan, 2021-02-03:
@svpcom issue fixed for me, thank you, this is really awesome.

closed 2021-02-03 by svpcom