svpcom/wfb-ng #77: fail to decode video using avdec_h264

iamfaith, 2021-01-31:
Hello, thanks for making this amazing repo.

But I try to install this on my virtual machine as the RX side. Everything looks good until I try to decode the video stream.

System info:
[faith@localhost ~]$ uname -a
Linux localhost.localdomain 5.8.15-301.fc33.x86_64 #1 SMP Thu Oct 15 16:58:06 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Problem:

no element avdec_h264

[faith@localhost ~]$ gst-inspect-1.0 |grep 264
openh264:  openh264dec: OpenH264 video decoder
openh264:  openh264enc: OpenH264 video encoder
videoparsersbad:  h264parse: H.264 parser
uvch264:  uvch264mjpgdemux: UVC H264 MJPG Demuxer
uvch264:  uvch264src: UVC H264 Source
uvch264:  uvch264deviceprovider (GstDeviceProviderFactory)
typefindfunctions: video/x-h264: h264, x264, 264
rtp:  rtph264depay: RTP H264 depayloader
rtp:  rtph264pay: RTP H264 payloader

Question: May I use openh264dec instead of avdec_h264? I have searched by any methods, and I fail to install avdec_h264 plugin.

Any suggestions or comments help me a lot. Thanks in advance.

svpcom, 2021-01-31:
Yes, you can use any h264 decoder. avdec_h264 belongs to gstreamer1.0-libav (ubuntu 18.04)

iamfaith, 2021-01-31:
> Yes, you can use any h264 decoder. avdec_h264 belongs to gstreamer1.0-libav (ubuntu 18.04)

Thanks for your kind reply. I was wondering is stream redirecting possible?
For example, I use RPI for the ground side to receive the video stream then it redirects the video stream to my iPad or Macbook for decoding. I just notice that there is an iOS App called QGroundControl which can decode the video stream at the same time.

If it's possible, maybe I can use Pi zero as the ground receiver and more powerful devices for decoding.

thanks a lot for your hard work and for making such an amazing project.

svpcom, 2021-01-31:
Yes, just set in /etc/wifibroadcast.cfg:
```
[gs_video]
peer = 'connect://1.2.3.4:5600'
```
where 1.2.3.4 is IP of you device with qgroundcontrol running

iamfaith, 2021-02-01:
Thanks for your response, since I don't use pixhawk flight controller and I don't need mavlink telemetry. Is there any way to launch wifibroadcast with only video transmitting?

iamfaith, 2021-02-01:
> Thanks for your response, since I don't use pixhawk flight controller and I don't need mavlink telemetry. Is there any way to launch wifibroadcast with only video transmitting?
> 

I am using betaflight. I am not sure that the mavlink telemetry in the repo works in betaflight or not

svpcom, 2021-02-01:
Just don't use telemetry ports. It doesn't wast any bandwidth if not used

iamfaith, 2021-02-01:
> Just don't use telemetry ports. It doesn't wast any bandwidth if not used

So in this case, how do I launch the service? Just use the rx_standalone.sh and tx_standalone.sh?

I was wondering if I use telemetry ports, the video will show telemetry information or I have to use cli to fetch telemetry information?

svpcom, 2021-02-01:
https://github.com/svpcom/wifibroadcast/wiki/Setup-HOWTO
`systemctl start wifibroadcast@gs` on the GS
`systemctl start wifibroadcast@drone` on the drone.
video is separated from telemetry (they are different udp streams). Use wfb-cli to see transmission statistics.

iamfaith, 2021-02-01:
> https://github.com/svpcom/wifibroadcast/wiki/Setup-HOWTO
> `systemctl start wifibroadcast@gs` on the GS
> `systemctl start wifibroadcast@drone` on the drone.
> video is separated from telemetry (they are different udp streams). Use wfb-cli to see transmission statistics.

Thanks, I just watch find some flying video which is using wifibroadcasting. I was wondering how they can show the flying information(such as speed and Rssi) at the real-time video. How does wifibroadcasting combine these two different udp streams? thanks very much

svpcom, 2021-02-01:
You can use OSD app. For example:  https://github.com/svpcom/wifibroadcast_osd
WIfibroadcast doesn't combine any streams. It is only a packet transport. This do qgroundcontrol or separate OSD application.

iamfaith, 2021-02-01:
Thank you very much. So does it means the wifibroadcast can provide both the video stream and telemetry stream for the downside application, such as the OSD application?

svpcom, 2021-02-01:
Yes. Also it provides IPv4 tunnel

iamfaith, 2021-02-01:
Thanks, it's a really smart and wise design. 

Have a nice day. You solve all my questions. I will close this issue.

closed 2021-02-01 by iamfaith

martins63, 2021-02-04:
Is it possible to redirect  video stream from GS  to 2 or more  IPs?

svpcom, 2021-02-05:
@martins63 Yes, just stream to multicast address 224.0.0.1

martins63, 2021-02-05:
OK, thank you. I will try.