svpcom/wfb-ng #78: WFB not establishing link

justas-, 2021-02-03:
I am unable to establish a link between the GS and the drone. I'm using vanilla setup with rtl8812au cards with the  patched driver. My syslog after starting GS (drone side is identical apart from the adapter name) is at the bottom. I get TUN adapter but packets do not go through to the other side. I've tried to run Wireshark on both ends and I see a lot of traffic with MAC 132233445505. Not sure if this is WFB traffic or not. wfb-cli reports link as lost.

> Feb  3 21:29:48 raspberrypi systemd[1]: Started WFB profile gs.
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] Log opened.
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] # iw reg set BO
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] # ifconfig wlx48d890cdb0d5 down
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] # iw dev wlx48d890cdb0d5 set monitor otherbss
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] # ifconfig wlx48d890cdb0d5 up
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] # iw dev wlx48d890cdb0d5 set channel 1 HT20
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] Connect telem stream 2(RX), 1(TX) to 127.0.0.1:14550
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] UDPProxyProtocol starting on 56795
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] UDPProxyProtocol starting on 14600
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] UDPProxyProtocol starting on 50925
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] Telem RX: /usr/bin/wfb_rx -p 2 -u 14600 -K /etc/gs.key -k 1 -n 2 wlx48d890cdb0d5
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] Telem TX: /usr/bin/wfb_tx -p 1 -u 14601 -K /etc/gs.key -B 20 -G long -S 1 -L 0 -M 1 -k 1 -n 2 wlx48d890cdb0d5
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] AntennaFactory starting on 8001
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] Started telem rx
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] Started telem tx
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] Send video stream 3 to 127.0.0.1:5600
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] AntennaFactory starting on 8002
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] Started video rx
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] Video: /usr/bin/wfb_rx -p 3 -c 127.0.0.1 -u 5600 -K /etc/gs.key -k 8 -n 12 wlx48d890cdb0d5
> Feb  3 21:29:49 raspberrypi systemd-udevd[1384]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] UDPProxyProtocol starting on 14800
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] UDPProxyProtocol starting on 55966
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] Tunnel RX: /usr/bin/wfb_rx -p 5 -u 14800 -K /etc/gs.key -k 1 -n 2 wlx48d890cdb0d5
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] Tunnel TX: /usr/bin/wfb_tx -p 4 -u 14801 -K /etc/gs.key -B 20 -G long -S 1 -L 0 -M 1 -k 1 -n 2 wlx48d890cdb0d5
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] AntennaFactory starting on 8003
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] Started tunnel rx
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] Started tunnel tx
> Feb  3 21:29:49 raspberrypi kernel: [ 5157.518839] device wlx48d890cdb0d5 entered promiscuous mode
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] telem tx: Listen on 14601 for wlx48d890cdb0d5
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] tunnel tx: Listen on 14801 for wlx48d890cdb0d5
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] telem rx: wlx48d890cdb0d5 has DLT_IEEE802_11_RADIO Encap
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] video rx: wlx48d890cdb0d5 has DLT_IEEE802_11_RADIO Encap
> Feb  3 21:29:49 raspberrypi python[1386]: 2021-02-03 21:29:49+0000 [-] tunnel rx: wlx48d890cdb0d5 has DLT_IEEE802_11_RADIO Encap
> Feb  3 21:30:07 raspberrypi python[1386]: 2021-02-03 21:30:07+0000 [-] tunnel rx: New session detected
> Feb  3 21:30:07 raspberrypi python[1386]: 2021-02-03 21:30:07+0000 [-] tunnel rx: 17 packets lost
> Feb  3 21:31:22 raspberrypi python[1386]: 2021-02-03 21:31:22+0000 [-] tunnel rx: 82 packets lost
> Feb  3 21:31:45 raspberrypi python[1386]: 2021-02-03 21:31:45+0000 [-] tunnel rx: 25 packets lost

svpcom, 2021-02-03:
According to log session was established, but unstable.
```
Feb 3 21:30:07 raspberrypi python[1386]: 2021-02-03 21:30:07+0000 [-] tunnel rx: New session detected
Feb 3 21:30:07 raspberrypi python[1386]: 2021-02-03 21:30:07+0000 [-] tunnel rx: 17 packets lost
Feb 3 21:31:22 raspberrypi python[1386]: 2021-02-03 21:31:22+0000 [-] tunnel rx: 82 packets lost
```

1. What is in `/etc/wifibroadcast.cfg` on both nodes?
2. Do you use wifi channel 1? 2.4GHz band is very polluted and rtl8812au has issues with it. I'll recommend to use 5.8GHz band.
3. Do you disable these cards in NetworkManager (as described in HOWTO)?
4. Please attach wifibroadcast logs from both nodes
5. What is output of `ifconfig -a` on both nodes?

justas-, 2021-02-04:
For these tests I have WiFi cards appx. 10 cm away from each other. Each card has it's own power supply.

1. The minimal based on setup howto:

GS:

> [common]
> wifi_channel = 161
> wifi_region = 'BO'
> debug = False
> [gs_mavlink]
> peer = 'connect://127.0.0.1:14550'
> [gs_video]
> peer = 'connect://127.0.0.1:5600'

UAV:

> [common]
> wifi_channel = 161
> wifi_region = 'BO'
> [drone_mavlink]
> peer = 'listen://0.0.0.0:14550'
> [drone_video]
> peer = 'listen://0.0.0.0:5602'

2. I tried channel 1 and 161. Exactly the same behavior

3.  Both GS and Drone are headless, without NetworkManager.

$ sudo systemctl stop NetworkManager
Failed to stop NetworkManager.service: Unit NetworkManager.service not loaded.

4.

> ● wifibroadcast@drone.service - WFB profile drone
>    Loaded: loaded (/lib/systemd/system/wifibroadcast@.service; disabled; vendor preset: enabled)
>    Active: active (running) since Thu 2021-02-04 15:40:14 GMT; 5min ago
>  Main PID: 699 (python)
>     Tasks: 6 (limit: 729)
>    CGroup: /system.slice/system-wifibroadcast.slice/wifibroadcast@drone.service
>            ├─699 /usr/bin/python -m telemetry.server drone wlan1
>            ├─733 /usr/bin/wfb_rx -p 1 -u 14700 -K /etc/drone.key -k 1 -n 2 wlan1
>            ├─738 /usr/bin/wfb_tx -p 2 -u 14701 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 -k 1 -n 2 wlan1
>            ├─743 /usr/bin/wfb_tx -p 3 -u 5602 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 -k 8 -n 12 wlan1
>            ├─772 /usr/bin/wfb_rx -p 4 -u 14900 -K /etc/drone.key -k 1 -n 2 wlan1
>            └─773 /usr/bin/wfb_tx -p 5 -u 14901 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 -k 1 -n 2 wlan1
> 
> Feb 04 15:40:26 raspberrypi-w0 python[699]: 2021-02-04 15:40:26+0000 [-] telem rx: wlan1 has DLT_IEEE802_11_RADIO Encap
> Feb 04 15:40:26 raspberrypi-w0 python[699]: 2021-02-04 15:40:26+0000 [-] video tx: Listen on 5602 for wlan1
> Feb 04 15:40:26 raspberrypi-w0 python[699]: 2021-02-04 15:40:26+0000 [-] tunnel rx: wlan1 has DLT_IEEE802_11_RADIO Encap
> Feb 04 15:40:26 raspberrypi-w0 python[699]: 2021-02-04 15:40:26+0000 [-] tunnel tx: Listen on 14901 for wlan1
> Feb 04 15:40:38 raspberrypi-w0 python[699]: 2021-02-04 15:40:38+0000 [-] tunnel rx: New session detected
> Feb 04 15:40:38 raspberrypi-w0 python[699]: 2021-02-04 15:40:38+0000 [-] tunnel rx: 8 packets lost
> Feb 04 15:42:47 raspberrypi-w0 python[699]: 2021-02-04 15:42:47+0000 [-] tunnel rx: 149 packets lost
> Feb 04 15:44:25 raspberrypi-w0 python[699]: 2021-02-04 15:44:25+0000 [-] tunnel rx: 108 packets lost
> Feb 04 15:45:22 raspberrypi-w0 python[699]: 2021-02-04 15:45:22+0000 [-] tunnel rx: 62 packets lost
> Feb 04 15:45:28 raspberrypi-w0 python[699]: 2021-02-04 15:45:28+0000 [-] tunnel rx: 6 packets lost

> Feb 04 15:40:23 raspberrypi-w0 python[699]: 2021-02-04 15:40:23+0000 [-] Log opened.
> Feb 04 15:40:23 raspberrypi-w0 python[699]: 2021-02-04 15:40:23+0000 [-] # iw reg set BO
> Feb 04 15:40:24 raspberrypi-w0 python[699]: 2021-02-04 15:40:24+0000 [-] # ifconfig wlan1 down
> Feb 04 15:40:24 raspberrypi-w0 python[699]: 2021-02-04 15:40:24+0000 [-] # iw dev wlan1 set monitor otherbss
> Feb 04 15:40:24 raspberrypi-w0 python[699]: 2021-02-04 15:40:24+0000 [-] # ifconfig wlan1 up
> Feb 04 15:40:24 raspberrypi-w0 python[699]: 2021-02-04 15:40:24+0000 [-] # iw dev wlan1 set channel 161 HT20
> Feb 04 15:40:24 raspberrypi-w0 python[699]: 2021-02-04 15:40:24+0000 [-] Listen for telem stream 1(RX), 2(TX) on 0.0.0.0:14550
> Feb 04 15:40:24 raspberrypi-w0 python[699]: 2021-02-04 15:40:24+0000 [-] UDPProxyProtocol starting on 14550
> Feb 04 15:40:24 raspberrypi-w0 python[699]: 2021-02-04 15:40:24+0000 [-] UDPProxyProtocol starting on 14700
> Feb 04 15:40:24 raspberrypi-w0 python[699]: 2021-02-04 15:40:24+0000 [-] UDPProxyProtocol starting on 46181
> Feb 04 15:40:24 raspberrypi-w0 python[699]: 2021-02-04 15:40:24+0000 [-] Telem RX: /usr/bin/wfb_rx -p 1 -u 14700 -K /etc/drone.key -k 1 -n 2 wlan1
> Feb 04 15:40:24 raspberrypi-w0 python[699]: 2021-02-04 15:40:24+0000 [-] Telem TX: /usr/bin/wfb_tx -p 2 -u 14701 -K /etc/drone.key -B 20 -G long -S 1
[... 134 lines cut ...]
        TX packets 3  bytes 252 (252.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 803  bytes 34408 (33.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 803  bytes 34408 (33.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.57  netmask 255.255.255.0  broadcast 192.168.1.255
        ether b8:27:eb:73:09:fb  txqueuelen 1000  (Ethernet)
        RX packets 1066  bytes 81187 (79.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 599  bytes 106921 (104.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan1: flags=4099<UP,BROADCAST,MULTICAST>  mtu 2312
        unspec 48-D8-90-CD-B0-C8-00-00-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 3665  bytes 119703 (116.8 KiB)
        RX errors 0  dropped 71  overruns 0  frame 0
        TX packets 1508  bytes 92636 (90.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

svpcom, 2021-02-04:
Why drone has wlan0 and wlan1 (used wlan1 for wfb) and gs has wlan0 and wlx48d890cdb0d5 (used wlx48d890cdb0d5 for wfb) if they have the same rtl8812au cards with the same patched driver? It seems that the problem is here

justas-, 2021-02-04:
Both GS and UAV have a built-in RPi wi-fi card that I use for console access. Both also have a patched driver and USB connected rtl8812au. I can try and disable built-in cards and use serial console for verification.

svpcom, 2021-02-04:
Check why rtl8812au has different interface names on GS and UAV. Use `ethtool -i wlanX` to check what driver is used

closed 2021-02-06 by svpcom