svpcom/wfb-ng #79: Can not receive any data from TX

iamfaith, 2021-02-04:
Hello, thanks for making such an amazing project.

But I face some problems while using this.

Hardware: 

gs as a Linux virtual machine with TPLink TL-WN722N V1.

It can be set into monitor mode like the following picture:
[image]

start with: 
[image]

drone as raspberry pi zero with Linux Kernal patch, which can be set into monitor mode also.

[image]

start with: ./wfb_tx -p 1 -u 5600 -K drone.key wlan0

Problem:

I print some logs at the file of rx.cpp at line 688 like this:

[image]

But the rc value is always 0.

How could I solve this problem? I can verify that the GStreamer works well because I can forward the video stream from Pi zero to my PC via standard wifi.

iamfaith, 2021-02-04:
BTW, when I enable DEBUG_TX, my PC can receive the UDP packets from Pi Zero. But fail for PcapTransmitter.

iamfaith, 2021-02-04:
the command I used:
TX: ./wfb_tx -p 3 -u 5600 -K drone.key -S 1 -L 1 -B 20 wlan0
RX: ./wfb.rx -c 10.0.2.15 -p3 -u 5600 -K gs.key $rx

svpcom, 2021-02-04:
1. Is it ath9k? If yes, you need to upload custom firmware https://github.com/OpenHD/Open.HD/tree/master/firmware/AR9271
2. DEBUG_TX used for FEC testing only - it doesn't inject packets to cards, but send it as UDP packets to 127.0.0.1:5601
3. what is `$rx`? Do you run these commands on the same host or on the different?

svpcom, 2021-02-04:
I don't recommend ath9k - use rt28xx or rtl8812au instead

iamfaith, 2021-02-04:
For the drone side, I use the re4son patch on my raspberry pi zero which can enable monitor mode. So I just use internal wifi card for raspberry pi zero.

For the gs side, I use this wifi adapter.
[image]

the $rx is just the variable of my wifi interface.

svpcom, 2021-02-04:
Without firmware patch ath9k will unable to switch radio channel. See https://github.com/svpcom/wifibroadcast/issues/46#issuecomment-770083818
I'll suggest to use patched ath9k on drone (as TX) and internal raspberry pi card as RX. But I highly recommend to use rtl8812au and/or rt28xx instead of them

iamfaith, 2021-02-04:
I did not apply any patch on ath9k, but it can switch the channel?
[image]

Because pi zero is very small then I would like to put it on the drone. And use my laptop as RX on the ground. I would like to use the internal wifi card of pi Zero rather than external card, which can reduce the weight.

svpcom, 2021-02-04:
Are you sure, that internal zero card can do packet injection?

svpcom, 2021-02-04:
monitor mode != packet injection in general case

iamfaith, 2021-02-04:
> monitor mode != packet injection in general case

Thanks for your clarification. But do I check it supports injection or not? I was thinking monitor mode == packet injection originally.

iamfaith, 2021-02-04:
[image]

But actually, I add some logs here. It will never throw out the runtime error. I think it should support packet injection.

I need to find some way to install the patch on my RX wifi card.

iamfaith, 2021-02-04:
> Without firmware patch ath9k will unable to switch radio channel. See [#46 (comment)](https://github.com/svpcom/wifibroadcast/issues/46#issuecomment-770083818)
> I'll suggest to use patched ath9k on drone (as TX) and internal raspberry pi card as RX. But I highly recommend to use rtl8812au and/or rt28xx instead of them

If I use rtl8812au for the ground side, do I need to patch rtl8812au? Or it just supports the monitor mode.

svpcom, 2021-02-04:
1. monitor_mode == allow to capture packets. Not all cards support that (but much more than support injection)
2. If card doesn't support injection it can silently ignore it and doesn't return any errors. Only a few wifi cards supports injection (ath9k with patches, rt28xx out of box, rtl8812au with patches, some builtin broadcom cards in RPI with custom firmware and patches)
3. In case of rtl8812au patch is only needed for injection. it already supports monitor mode

svpcom, 2021-02-04:
ath9k without patches inject packets but ignore wifi channel and modulation type (MCS) settings. So RX card will listen one channel and TX transmit on other.

iamfaith, 2021-02-04:
Well explained. I have tried on one channel. But it still doesn't work. I think you are right. My Pi zero internal wifi card doesn't support packet injection. 

Maybe I need to buy an RTL8812AU wifi adapter for my Pi Zero. 

I just notice another wonderful project from you: https://github.com/svpcom/rtl8812au

Can I use `sudo ./dkms-install.sh` to install this patch on any Linux OS?

Thanks

svpcom, 2021-02-04:
yes. For new kernel (5.10+) try v5.6.4.2 branch

iamfaith, 2021-02-04:
I just used this patch on my Pi Zero. 

[image]

It shows support injection mode on 8812AU. I will try it

iamfaith, 2021-02-04:
> 1. Is it ath9k? If yes, you need to upload custom firmware https://github.com/OpenHD/Open.HD/tree/master/firmware/AR9271
> 2. DEBUG_TX used for FEC testing only - it doesn't inject packets to cards, [...]

For DEBUG_TX, just send data content as UDP packets. While disabling DEBUG_TX, it will inject the data content into the 802.11 packets. Is this correct?

svpcom, 2021-02-04:
> For DEBUG_TX, just send data content as UDP packets. While disabling DEBUG_TX, it will inject the data content into the 802.11 packets. Is this correct?
yes

iamfaith, 2021-02-05:
> I don't recommend ath9k - use rt28xx or rtl8812au instead

Dear svpvom,

I just find the driver of my wifi adapter is like this. Is this what you refer to as rt28xx driver? If it is, how do I patch it in order to use it in raspberry pi zero as injection mode? thanks a lot

[image]

iamfaith, 2021-02-05:
BTW, I reinstall the patch again, it works for aireplay-ng command. Does this work in injection mode?

[image]

Then I try on wifibroadcast. But on the gs side, it always shows not receiving any data packet. But in GStreamer, it occurs error, showing that it can not decrypt the stream, not a rtp stream

svpcom, 2021-02-05:
yes. You don't need to patch - it works out of box.
Only possible patch is to disable CRDA to maximize output power.

> > I don't recommend ath9k - use rt28xx or rtl8812au instead
> 
> Dear svpvom,
> 
> I just find the driver of my wifi adapter is like this. Is this what you refer to as rt28xx driver? If it is, how do I patch it in order to use it in raspberry pi zero as injection mode? thanks a lot
> 
> [image]

svpcom, 2021-02-05:
Injection only working if you able to receive injected packets from other host. To to catch them via tcpdump from other machine

> BTW, I reinstall the patch again, it works for aireplay-ng command. Does this work in injection mode?
>
> [image]
>
> Then I try on wifibroadcast. But on the gs side, it always shows not receiving any [...]

iamfaith, 2021-02-05:
> Injection only working if you able to receive injected packets from other host. To to catch them via tcpdump from other machine

Sorry, I am not familiar with catching packets. Which parameters of the tcpdump command do I use to catch this packet?

svpcom, 2021-02-05:
Do you try `wlan0mon` device for wifibroadcast injection test?

```
ifconfig $WLAN down
iw dev $WLAN set monitor otherbss
ifconfig $WLAN up
iw dev $WLAN set channel $CHANNEL
tcpdump -nn -v -i $WLAN
```

iamfaith, 2021-02-05:
> Do you try `wlan0mon` device for wifibroadcast injection test?
> 
> ```
> ifconfig $WLAN down
> iw dev $WLAN set monitor otherbss
> ifconfig $WLAN up
> iw dev $WLAN set channel $CHANNEL
> tcpdump -nn -v -i $WLAN
> ```

Yes, I try it as injection mode. Only several times I have received packets, each time only one packet! 
[image]

iamfaith, 2021-02-05:
> tcpdump -nn -v -i

I try this command, but there are too many packets. How to filter my pakcets?

[image]

svpcom, 2021-02-05:
Dump them to file via `-w filename` and open it in wireshark

iamfaith, 2021-02-05:
Yes, I follow your advice.

My mac address of wlan0mon is this:
[image]

But in Wireshark, I sort by source. I don't find it.
[image]

How to solve this? Is anything I am missing?

svpcom, 2021-02-05:
Do you use rt28xx/rtl8812au from both sides? If no, I can't help, because I haven't experience with other cards

iamfaith, 2021-02-05:
I try to reverse. Use rt28xx for aireplay-ng, and Pi zero for tcpdump.

I got the result like this. are there injected packet sent from aireplay-ng?

[image]
[image]
[image]

Just notice, I test aireplay-ng on Pi zero 's wlan0mon interface. I can not find these Clear-to-send and Acknowledgement packets.
Is this meaning that this rt28xx card can work in injection mode?

svpcom, 2021-02-05:
Yes. rt28xx card work in injection/monitoring modes out of box.
I don't know what packets aireplay-ng injects into wifi. Need to read its manual

iamfaith, 2021-02-05:
> Yes. rt28xx card work in injection/monitoring modes out of box.
> I don't know what packets aireplay-ng injects into wifi. Need to read its manual

Thanks for your reply. is it the same for rtl8812au with injection/monitoring modes out of box

svpcom, 2021-02-05:
No. For rtl8812au you should use https://github.com/svpcom/rtl8812au driver. Stock driver doesn't support injection.

iamfaith, 2021-02-05:
> No. For rtl8812au you should use https://github.com/svpcom/rtl8812au driver. Stock driver doesn't support injection.

OOOOh, thanks. The stock driver of rtl8812au only supports monitor mode. right? If used this ground side, it doesn't need this driver. But for the drone side, need to use https://github.com/svpcom/rtl8812au driver, is this correct?

svpcom, 2021-02-05:
If you want to use only one-side communication (like video only), you can use it. But ip tunnel will not work

iamfaith, 2021-02-05:
Yes, what do you mean by ip tunnel? Is it for mavlink telemetry

svpcom, 2021-02-05:
https://github.com/svpcom/wifibroadcast/wiki/Setup-HOWTO
Out of box WFB provide 3 channels:
1. Video downstream
2. Bidirectional mavlink
3. IP tunnel

In case of unidirectional communication will work only (1) and (2) in downstream mode. (3) will not work at all

iamfaith, 2021-02-05:
Thanks, I see this! Is it based on UDP? It can be used to send commands to Drone.

[image]

BTW, is it possible to disable encryption on the video tunnel which may reduce latency? Or can it be as a switch in the configuration file?

iamfaith, 2021-02-05:
Dear svpcom,
 
Do you suggest using RTL 3070 or RTL 5370 wifi cards?

svpcom, 2021-02-05:
Both should work, but I used only RT5572 and RT3572

svpcom, 2021-02-05:
Encryption doesn't affect to latency. it is always on. If you don't need it you can try to use https://github.com/OpenHD/Open.HD. They have premade images

iamfaith, 2021-02-05:
> Both should work, but I used only RT5572 and RT3572

Thanks. Do they support injection mode out of the box? How can I check that before I buy it?

svpcom, 2021-02-05:
RT5572 and RT3572 supported by rt28xx without any patches. But these chipsets are out-of-stock now.
They works for me, but I'm unable to guarantee that they will work in all cases. Packets injection and monitor modes is not officially supported in any chipset.  For new setups I'll recommend to use **rtl8812au**  (not "bu" or other models)

> > Both should work, but I used only RT5572 and RT3572
> 
> Thanks. Do they support injection mode out of the box? How can I check that before I buy it?

iamfaith, 2021-02-05:
Thanks. Why do you recommend rtl8812au? It's a little bit expensive compare with rt28xx in my country. But in my country, I can still buy rt28xx chipsets.

[image]

I just find that TX power is only 20 dBm. Is there any way to maximize it?

You help me a lot.

Realllly appreciate a lot

iamfaith, 2021-02-05:
In contrast, my raspberry pi zero has 31 dBm in TX power which higher than my  rt28xx wifi card.

[image]

svpcom, 2021-02-05:
This is a fake TX power. 31dBm is ~ 1W of output. Only card with heatsink can operate. Some of rtl8812au can output such power, for example https://www.aliexpress.com/item/4001243968808.html, but heatsink (or active cooling) is required

closed 2021-02-06 by svpcom