svpcom/wfb-ng #112: Packets lost while running in virtual machine with NAT network[BUG]

iamfaith, 2021-02-08:
**Describe the bug**
Packets lost while running in the virtual machine with NAT network. Both Rx and Tx are using rt28xx wifi cards.

**To Reproduce**
Steps to reproduce the behavior:

1. run RX in virtual box with NAT network with rt28xx wifi cards.
2. run TX in raspberry pi zero with rt28xx wifi cards.

the output result is like this.

[image]

[image]

iamfaith, 2021-02-08:
I found the problem. It 's because i am using the monitor interface. Now I close it

iamfaith, 2021-02-08:
pi(gstreamer) --> pi(wfb_tx:rt:28xxx) -------------(radio)---------------> virtual box ubuntu(wfb_rx: rt28xxx)--(NAT)---> mac os(gstreamer)
This gets a really bad video.

If pi(gstreamer)  --(wifi)----> mac os(gstreamer), pretty good video

svpcom, 2021-02-08:
What is your gstreamer pipeline?

svpcom, 2021-02-08:
What frequency channel do you use?

iamfaith, 2021-02-08:
> What frequency channel do you use?

channel 6 with 20hz

On pi zero:  gst-launch-1.0  filesrc location=a.mp4 ! decodebin ! omxh264enc ! rtph264pay ! udpsink host=127.0.0.1 port=5600

On mac:  
 gst-launch-1.0 udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' \
               ! rtph264depay ! avdec_h264 ! clockoverlay valignment=bottom ! autovideosink fps-update-interval=1000 sync=false

iamfaith, 2021-02-08:
Dear svpcom, 

I found the problem is that in this channel, there are 16 APs! It seems there are interferences. Since I live in city, each channel seems really busy. is there any way to overcome this?

svpcom, 2021-02-08:
Yes, use 5.8Ghz band

svpcom, 2021-02-08:
2.4 GHz is not usable in cities

iamfaith, 2021-02-08:
> 2.4 GHz is not usable in cities

But there are many 5G signals from wifi as well. Why will 5G Hz be usable?

svpcom, 2021-02-08:
In 2.4 band only 3 independent channels. In 5.8 - much more https://en.wikipedia.org/wiki/List_of_WLAN_channels

svpcom, 2021-02-08:
Also most of consumer radio devices use 2.4 band (bluetooth, auto alarms, microwave ovens etc)

closed 2021-02-10 by iamfaith