svpcom/wfb-ng #118: [BUG] Link lost!

Abu-Abdullah, 2021-03-03:
I got link lost after trying the installation procedures. the only difference is that i installed rtl8812au from https://github.com/aircrack-ng/rtl8812au.git since there are few patches for 5.10 kernel in Raspberry.

the logs shows everything fine:
```
GS
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] Log opened.
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] # iw reg set BO
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] # ifconfig wlan1 down
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] # iw dev wlan1 set monitor otherbss
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] # ifconfig wlan1 up
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] # iw dev wlan1 set channel 161 HT20
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] Connect telem stream 2(RX), 1(TX) to 127.0.0.1:14550
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] UDPProxyProtocol starting on 42331
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] UDPProxyProtocol starting on 14600
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] UDPProxyProtocol starting on 60935
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] Telem RX: /usr/bin/wfb_rx -p 2 -u 14600 -K /etc/gs.key -k 1 -n 2 wlan1
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] Telem TX: /usr/bin/wfb_tx -p 1 -u 14601 -K /etc/gs.key -B 20 -G long -S 1 -L 0 -M 1 -k
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] AntennaFactory starting on 8001
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] Started telem rx
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] Started telem tx
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] Send video stream 3 to 127.0.0.1:5600
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] AntennaFactory starting on 8002
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] Started video rx
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] Video: /usr/bin/wfb_rx -p 3 -c 127.0.0.1 -u 5600 -K /etc/gs.key -k 8 -n 12 wlan1
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] UDPProxyProtocol starting on 14800
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] UDPProxyProtocol starting on 50003
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] Tunnel RX: /usr/bin/wfb_rx -p 5 -u 14800 -K /etc/gs.key -k 1 -n 2 wlan1
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] Tunnel TX: /usr/bin/wfb_tx -p 4 -u 14801 -K /etc/gs.key -B 20 -G long -S 1 -L 0 -M 1 -
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] AntennaFactory starting on 8003
Mar 03 07:22:51 raspberrypi python[1258]: 2021-03-03 07:22:51+0000 [-] Started tunnel rx
Mar 03 07:22:52 raspberrypi python[1258]: 2021-03-03 07:22:52+0000 [-] Started tunnel tx
Mar 03 07:22:52 raspberrypi python[1258]: 2021-03-03 07:22:52+0000 [-] telem tx: Listen on 14601 for wlan1
Mar 03 07:22:52 raspberrypi python[1258]: 2021-03-03 07:22:52+0000 [-] tunnel tx: Listen on 14801 for wlan1
Mar 03 07:22:52 raspberrypi python[1258]: 2021-03-03 07:22:52+0000 [-] telem rx: wlan1 has DLT_IEEE802_11_RADIO Encap
Mar 03 07:22:52 raspberrypi python[1258]: 2021-03-03 07:22:52+0000 [-] video rx: wlan1 has DLT_IEEE802_11_RADIO Encap
Mar 03 07:22:52 raspberrypi python[1258]: 2021-03-03 07:22:52+0000 [-] tunnel rx: wlan1 has DLT_IEEE802_11_RADIO Encap
```

```
drone
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] Log opened.
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] # iw reg set BO
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] # ifconfig wlan1 down
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] # iw dev wlan1 set monitor otherbss
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] # ifconfig wlan1 up
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] # iw dev wlan1 set channel 161 HT20
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] Listen for telem stream 1(RX), 2(TX) on 0.0.0.0:14550
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] UDPProxyProtocol starting on 14550
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] UDPProxyProtocol starting on 14700
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] UDPProxyProtocol starting on 49134
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] Telem RX: /usr/bin/wfb_rx -p 1 -u 14700 -K /etc/drone.key -k 1 -n 2 wlan1
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] Telem TX: /usr/bin/wfb_tx -p 2 -u 14701 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] Started telem rx
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] Started telem tx
Mar 03 07:22:57 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] Listen for video stream 3 on 0.0.0.0:5602
Mar 03 07:22:58 raspberrypi python[715]: 2021-03-03 07:22:57+0000 [-] Started video tx
Mar 03 07:22:58 raspberrypi python[715]: 2021-03-03 07:22:58+0000 [-] Video: /usr/bin/wfb_tx -p 3 -u 5602 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1 -k 8
Mar 03 07:22:58 raspberrypi python[715]: 2021-03-03 07:22:58+0000 [-] UDPProxyProtocol starting on 14900
Mar 03 07:22:58 raspberrypi python[715]: 2021-03-03 07:22:58+0000 [-] UDPProxyProtocol starting on 60501
Mar 03 07:22:58 raspberrypi python[715]: 2021-03-03 07:22:58+0000 [-] Tunnel RX: /usr/bin/wfb_rx -p 4 -u 14900 -K /etc/drone.key -k 1 -n 2 wlan1
Mar 03 07:22:58 raspberrypi python[715]: 2021-03-03 07:22:58+0000 [-] Tunnel TX: /usr/bin/wfb_tx -p 5 -u 14901 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1
Mar 03 07:22:58 raspberrypi python[715]: 2021-03-03 07:22:58+0000 [-] Started tunnel rx
Mar 03 07:22:58 raspberrypi python[715]: 2021-03-03 07:22:58+0000 [-] Started tunnel tx
Mar 03 07:22:58 raspberrypi python[715]: 2021-03-03 07:22:58+0000 [-] telem tx: Listen on 14701 for wlan1
Mar 03 07:22:58 raspberrypi python[715]: 2021-03-03 07:22:58+0000 [-] video tx: Listen on 5602 for wlan1
Mar 03 07:22:58 raspberrypi python[715]: 2021-03-03 07:22:58+0000 [-] tunnel tx: Listen on 14901 for wlan1
Mar 03 07:22:58 raspberrypi python[715]: 2021-03-03 07:22:58+0000 [-] tunnel rx: wlan1 has DLT_IEEE802_11_RADIO Encap
Mar 03 07:22:58 raspberrypi python[715]: 2021-03-03 07:22:58+0000 [-] telem rx: wlan1 has DLT_IEEE802_11_RADIO Encap
```

configuration on both of them
```
wifi_channel = 161     # 161 -- radio channel @5805 MHz, range: 5795–5815 MHz, width 20MHz
                       # 1 -- radio channel @2412 Mhz, 
                       # see https://en.wikipedia.org/wiki/List_of_WLAN_channels for reference
wifi_region = 'BO'     # Your country for CRDA (use BO or GY if you want max tx power) 

[gs_mavlink]
peer = 'connect://127.0.0.1:14550'  # outgoing connection
# peer = 'listen://0.0.0.0:14550'   # incoming connection

bandwidth = 20     # bandwidth 20 or 40 MHz
short_gi = False   # use short GI or not
stbc = 1           # stbc streams: 1, 2, 3 or 0 if unused
mcs_index = 1      # mcs index

[gs_video]
peer = 'connect://127.0.0.1:5600'  # outgoing connection for
                                   # video sink (QGroundControl on GS)
                                   
bandwidth = 20     # bandwidth 20 or 40 MHz
short_gi = False   # use short GI or not
stbc = 1           # stbc streams: 1, 2, 3 or 0 if unused
mcs_index = 1      # mcs index

[drone_mavlink]
peer = 'listen://0.0.0.0:14550'   # incoming connection
# peer = 'connect://127.0.0.1:14550'  # outgoing connection

bandwidth = 20     # bandwidth 20 or 40 MHz
short_gi = False   # use short GI or not
stbc = 1           # stbc streams: 1, 2, 3 or 0 if unused
mcs_index = 1      # mcs index

[drone_video]
peer = 'listen://0.0.0.0:5602'  # listen for video stream (gstreamer on drone)

bandwidth = 20     # bandwidth 20 or 40 MHz
short_gi = False   # use short GI or not
stbc = 1           # stbc streams: 1, 2, 3 or 0 if unused
mcs_index = 1      # mcs index
```

and mavlink is the normal
```
[UdpEndpoint wifibroadcast]
Mode = Normal
Address = 127.0.0.1
Port = 14550
```

svpcom, 2021-03-03:
Use https://github.com/svpcom/rtl8812au  v5.6.4.2 branch. Other drivers may doesn't support packet injection.

Abu-Abdullah, 2021-03-03:
i got the same error after removing the old driver and compile the new one. do i need to go back to older Pi version

svpcom, 2021-03-03:
Use `ethtool -i wlanX` and `modinfo _kernel_module_name_from_ethtool_` to be sure that right driver used. RPI can have builtin driver for rtl8812au

Abu-Abdullah, 2021-03-03:
it seems the correct driver is uploaded
```
pi@raspberrypi:~ $ ethtool -i wlan1
driver: rtl88XXau
version: 5.10.11-v7+
firmware-version:
expansion-rom-version:
bus-info: 1-1.4:1.0
supports-statistics: no
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

```
pi@raspberrypi:~ $ modinfo 88XXau
filename:       /lib/modules/5.10.11-v7+/updates/88XXau.ko
version:        v5.6.4.2_35491.20191025
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     3C44D63DCE13BE5157A716C
```

svpcom, 2021-03-03:
No. It seems that loaded default driver because version is not empty:
```
$ ethtool -i wlan1
driver: rtl88XXau
version: 
firmware-version: 
expansion-rom-version: 
bus-info: 3-1:1.0
supports-statistics: no
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

$ ethtool -i wlan0
driver: ath9k
version: 5.4.0-66-generic
firmware-version: N/A
expansion-rom-version: 
bus-info: 0000:02:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

Try `find /lib/modules | grep -i 88xx`. If found, remove it and rebuild initramfs

Abu-Abdullah, 2021-03-03:
thank you, will do it and come back

Abu-Abdullah, 2021-03-03:
i tried many things like flash the SD again, the 88XXau.ko seems coming from compiling the src code not from any builtin driver. if i removed 88XXau from /lib/modules, the wlan1 will not come up

i tried also to compile it manually instead of using the dkms_install, the issue is still the same, the only difference is in the path: 
/lib/modules/5.10.11-v7+/kernel/drivers/net/wireless/

sorry that im not familiar on dealing with drivers here. if there is any detailed steps i can follow

thank you again

Abu-Abdullah, 2021-03-04:
i just flashed new OS and didnt upgrade to kernel 5.10 (kernel 5.4) and it seems the output of $ethtool -i wlan1 is the expected one now. yet still the issue remain.

```
pi@raspberrypi:~ $ ethtool -i wlan1
driver: rtl88XXau
version:
firmware-version:
expansion-rom-version:
bus-info: 1-1.4:1.0
supports-statistics: no
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

following is the logs

GS
```
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] Log opened.
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] # iw reg set BO
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] # ifconfig wlan1 down
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] # iw dev wlan1 set monitor otherbss
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] # ifconfig wlan1 up
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] # iw dev wlan1 set channel 161 HT20
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] Connect telem stream 2(RX), 1(TX) to 127.0.0.1:14550
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] UDPProxyProtocol starting on 53929
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] UDPProxyProtocol starting on 14600
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] UDPProxyProtocol starting on 56398
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] Telem RX: /usr/bin/wfb_rx -p 2 -u 14600 -K /etc/gs.key -k 1 -n 2 wlan1
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] Telem TX: /usr/bin/wfb_tx -p 1 -u 14601 -K /etc/gs.key -B 20 -G long -S 1 -L 0 -M 1 -k
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] AntennaFactory starting on 8001
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] Started telem rx
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] Started telem tx
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] Send video stream 3 to 127.0.0.1:5600
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] AntennaFactory starting on 8002
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] Started video rx
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] Video: /usr/bin/wfb_rx -p 3 -c 127.0.0.1 -u 5600 -K /etc/gs.key -k 8 -n 12 wlan1
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] UDPProxyProtocol starting on 14800
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] UDPProxyProtocol starting on 49179
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] Tunnel RX: /usr/bin/wfb_rx -p 5 -u 14800 -K /etc/gs.key -k 1 -n 2 wlan1
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] Tunnel TX: /usr/bin/wfb_tx -p 4 -u 14801 -K /etc/gs.key -B 20 -G long -S 1 -L 0 -M 1 -
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] AntennaFactory starting on 8003
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] Started tunnel rx
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] Started tunnel tx
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] telem tx: Listen on 14601 for wlan1
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] tunnel tx: Listen on 14801 for wlan1
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] video rx: wlan1 has DLT_IEEE802_11_RADIO Encap
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] telem rx: wlan1 has DLT_IEEE802_11_RADIO Encap
Mar 04 17:40:33 raspberrypi python[2530]: 2021-03-04 17:40:33+0000 [-] tunnel rx: wlan1 has DLT_IEEE802_11_RADIO Encap
```

drone
```
Mar 04 17:40:54 raspberrypi python[1164]: 2021-03-04 17:40:54+0000 [-] Log opened.
Mar 04 17:40:54 raspberrypi python[1164]: 2021-03-04 17:40:54+0000 [-] # iw reg set BO
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] # ifconfig wlan1 down
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] # iw dev wlan1 set monitor otherbss
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] # ifconfig wlan1 up
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] # iw dev wlan1 set channel 161 HT20
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] Listen for telem stream 1(RX), 2(TX) on 0.0.0.0:14550
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] UDPProxyProtocol starting on 14550
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] UDPProxyProtocol starting on 14700
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] UDPProxyProtocol starting on 43520
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] Telem RX: /usr/bin/wfb_rx -p 1 -u 14700 -K /etc/drone.key -k 1 -n 2 wlan1
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] Telem TX: /usr/bin/wfb_tx -p 2 -u 14701 -K /etc/drone.key -B 20 -G long -S 1 -L 0 -M 1
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] Started telem rx
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] Started telem tx
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] Listen for video stream 3 on 0.0.0.0:5602
[... cut ...]
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] tunnel tx: Listen on 14901 for wlan1
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] telem rx: wlan1 has DLT_IEEE802_11_RADIO Encap
Mar 04 17:40:55 raspberrypi python[1164]: 2021-03-04 17:40:55+0000 [-] tunnel rx: wlan1 has DLT_IEEE802_11_RADIO Encap
Mar 04 17:41:18 raspberrypi python[1164]: 2021-03-04 17:41:18+0000 [-] tunnel rx: unable to decrypt packet #0x3300
Mar 04 17:41:18 raspberrypi python[1164]: 2021-03-04 17:41:18+0000 [-] tunnel rx: unable to decrypt packet #0x3301
```

svpcom, 2021-03-04:
But now drone got some packages from GS:
```
Mar 04 17:41:18 raspberrypi python[1164]: 2021-03-04 17:41:18+0000 [-] tunnel rx: unable to decrypt packet #0x3300
Mar 04 17:41:18 raspberrypi python[1164]: 2021-03-04 17:41:18+0000 [-] tunnel rx: unable to decrypt packet #0x3301
```
but it unable to decrypt them. Check that drone.key corresponds to gs.key
Also check that NetworkManager (or other tools like wpa_supplicant) are disabled for `wlan1`. Do `ps uaxwwww | grep wlan1` it should show nothing

Abu-Abdullah, 2021-03-05:
thank you very much @svpcom 
the issue was with wpa_supplicant, i just disable it by having 'denyinterfaces wlan1' in /etc/dhcpcd.conf

closed 2021-03-05 by Abu-Abdullah