svpcom/wfb-ng #120: make deb fail

iamfaith, 2021-03-12:
uname -a
Linux faith 5.4.0-53-generic #59~18.04.1-Ubuntu SMP Wed Oct 21 12:14:56 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

make deb fail:

make[2]: Entering directory '/home/faith/wifi_vtx/deb_dist/wifibroadcast-21.3.12.15779'
python setup.py install --force --root=debian/wifibroadcast --no-compile -O0 --install-layout=deb --prefix=/usr
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --install-layout not recognized
debian/rules:17: recipe for target 'override_dh_auto_install' failed
make[2]: *** [override_dh_auto_install] Error 1
make[2]: Leaving directory '/home/faith/wifi_vtx/deb_dist/wifibroadcast-21.3.12.15779'
debian/rules:7: recipe for target 'binary' failed
make[1]: *** [binary] Error 2
make[1]: Leaving directory '/home/faith/wifi_vtx/deb_dist/wifibroadcast-21.3.12.15779'
dpkg-buildpackage: error: fakeroot debian/rules binary subprocess returned exit status 2
Traceback (most recent call last):
  File "./setup.py", line 82, in <module>
    license="GPLv3",
  File "/home/faith/wifi_vtx/env/local/lib/python2.7/site-packages/setuptools/__init__.py", line 162, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python2.7/distutils/core.py", line 151, in setup
    dist.run_commands()
  File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/home/faith/wifi_vtx/env/local/lib/python2.7/site-packages/stdeb/command/bdist_deb.py", line 67, in run
    util.process_command(syscmd, cwd=target_dirs[0])
  File "/home/faith/wifi_vtx/env/local/lib/python2.7/site-packages/stdeb/util.py", line 227, in process_command
    check_call(args, cwd=cwd)
  File "/home/faith/wifi_vtx/env/local/lib/python2.7/site-packages/stdeb/util.py", line 59, in check_call
    raise CalledProcessError(retcode)
stdeb.util.CalledProcessError: 2
Makefile:52: recipe for target 'deb' failed
make: *** [deb] Error 1

iamfaith, 2021-03-12:
FYI: --install-layout is a debian modification to Pythons "distutils" module. That option is maintained by and only shipped with Debian(-derivates), it is not part of the official Python release. So anyone using other python releases cannot use --install-layout.

Does this mean it cannot use in ubuntu?

iamfaith, 2021-03-12:
My python is installed by anaconda.

iamfaith, 2021-03-12:
Solve by modifying utils.py

closed 2021-03-12 by iamfaith

hewittzgh, 2022-01-14:
Hi, I also encountered the same problem as you, "error: option --install-layout not recognized". Can you tell me the detailed solution.Thanks!

hewittzgh, 2022-01-14:
@iamfaith Hi, I also encountered the same problem as you, "error: option --install-layout not recognized". Can you tell me the detailed solution.Thanks!

richard-scott, 2022-08-12:
@hewittzgh edit line 1661 of env/lib/python3.8/site-packages/stdeb/util.py and remove the '--install-layout=deb' text.
Or run this to remove it for you.

```
$ sed -i -r 's/\s--install-layout=deb//' env/lib/python3.8/site-packages/stdeb/util.py
```

ppkliu, 2022-10-07:
I downgrade pip and setuptools like the following comment, it can solve this issue
#https://github.com/svpcom/wifibroadcast/issues/246

`Try to replace --upgrade pip==20.2.3 setuptools==44.1.1 with --upgrade pip setuptools in the Makefile, the run make clean, make env and make deb`