svpcom/wfb-ng #121: Update python to 3

Abu-Abdullah, 2021-03-12:
Hi Vasily,

is there a chance to update python to version 3, Ubuntu 20.04 stopped supporting those libs:
```
E: Package 'python-twisted' has no installation candidate
E: Package 'python-pyroute2' has no installation candidate
E: Package 'python-future' has no installation candidate
```

I tried to install the version 3 of these libs and the installation goes on but it will not work:
```
The following packages have unmet dependencies:
 wifibroadcast : Depends: python-twisted but it is not installable
                 Depends: python-pyroute2 but it is not installable
                 Depends: python-future but it is not installable
```

lucasgrelaud, 2021-03-12:
These dependencies are specified here for the debian-based disto : https://github.com/svpcom/wifibroadcast/blob/master/stdeb.cfg
Updating this file with the right dependencies should do the trick and let you build the package.

svpcom, 2021-03-12:
```
# make env PYTHON=python3
# make deb
# dpkg -I deb_dist/wifibroadcast*.deb
 new Debian package, version 2.0.
 size 145704 bytes: control archive=1704 bytes.
      63 bytes,     2 lines      conffiles            
     368 bytes,    10 lines      control              
    2368 bytes,    28 lines      md5sums              
     165 bytes,     9 lines   *  postinst             #!/bin/sh
     401 bytes,    12 lines   *  prerm                #!/bin/sh
 Package: wifibroadcast
 Version: 21.3.12.65053-1
 Architecture: all
 Maintainer: Vasily Evseenko <svpcom@p2ptech.org>
 Installed-Size: 1345
 Depends: python3:any (>= 3.3.2-2~), python3-twisted, libpcap-dev, libsodium-dev, python3-pyroute2, python3-future
 Section: python
 Priority: optional
 Homepage: http://github.com/svpcom/wifibroadcast
 Description: Wifibroadcast server
```

Abu-Abdullah, 2021-03-13:
thank you, i got this error with `make deb`
```
dpkg-buildpackage: warning: (Use -d flag to override.)
Traceback (most recent call last):
  File "./setup.py", line 62, in <module>
    setup(
  File "/home/oracle/wifibroadcast/env/lib/python3.8/site-packages/setuptools/__init__.py", line 162, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.8/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/oracle/wifibroadcast/env/lib/python3.8/site-packages/stdeb/command/bdist_deb.py", line 34, in run
    self.run_command('sdist_dsc')
  File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/oracle/wifibroadcast/env/lib/python3.8/site-packages/stdeb/command/sdist_dsc.py", line 137, in run
    build_dsc(debinfo,
  File "/home/oracle/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 1569, in build_dsc
    dpkg_buildpackage(*args, cwd=fullpath_repackaged_dirname)
  File "/home/oracle/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 585, in dpkg_buildpackage
    process_command(args, cwd=cwd)
  File "/home/oracle/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 226, in process_command
    check_call(args, cwd=cwd)
  File "/home/oracle/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 59, in check_call
    raise CalledProcessError(retcode)
stdeb.util.CalledProcessError: 3
make: *** [Makefile:53: deb] Error 1

```

svpcom, 2021-03-13:
What is full log? It seems that is `dpkg-buildpackage` error. What linux distributive do you use? I've test all on Ubuntu 18.04.

Abu-Abdullah, 2021-03-13:
it is ubuntu 20.04, let me redo it again to get the full logs

Abu-Abdullah, 2021-03-13:
```
oracle@oracle-X58A-UD7:~/wifibroadcast$ make env PYTHON=python3
virtualenv env --python=python3
created virtual environment CPython3.8.5.final.0-64 in 1223ms
  creator CPython3Posix(dest=/home/oracle/wifibroadcast/env, clear=False, global=False)
  seeder FromAppData(download=False, certifi=latest, html5lib=latest, urllib3=latest, pep517=latest, pkg_resources=latest, CacheControl=latest, pytoml=latest, pyparsing=latest, progress=latest, distro=latest, six=latest, ipaddr=latest, packaging=latest, retrying=latest, colorama=latest, pip=latest, appdirs=latest, chardet=latest, requests=latest, contextlib2=latest, idna=latest, distlib=latest, msgpack=latest, webencodings=latest, lockfile=latest, wheel=latest, setuptools=latest, via=copy, app_data_dir=/home/oracle/.local/share/virtualenv/seed-app-data/v1.0.1.debian)
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
./env/bin/pip install --upgrade pip==20.2.3 setuptools==44.1.1 stdeb
Collecting pip==20.2.3
  Using cached pip-20.2.3-py2.py3-none-any.whl (1.5 MB)
Collecting setuptools==44.1.1
  Using cached setuptools-44.1.1-py2.py3-none-any.whl (583 kB)
Processing /home/oracle/.cache/pip/wheels/a1/ae/c4/84a87b212b90c02abd4f1fc91ec69e9742b123c38d307e9106/stdeb-0.10.0-py3-none-any.whl
Installing collected packages: pip, setuptools, stdeb
  Attempting uninstall: pip
    Found existing installation: pip 20.0.2
    Uninstalling pip-20.0.2:
      Successfully uninstalled pip-20.0.2
  Attempting uninstall: setuptools
    Found existing installation: setuptools 44.0.0
    Uninstalling setuptools-44.0.0:
      Successfully uninstalled setuptools-44.0.0
Successfully installed pip-20.2.3 setuptools-44.1.1 stdeb-0.10.0
oracle@oracle-X58A-UD7:~/wifibroadcast$ make deb
g++  -Wall -O2 -DWFB_VERSION='"21.3.13.28609-03c6e80f"' -std=gnu++11 -c -o src/rx.o src/rx.cpp
cc  -Wall -O2 -DWFB_VERSION='"21.3.13.28609-03c6e80f"' -std=gnu99 -c -o src/radiotap.o src/radiotap.c
src/radiotap.c: In function ‘ieee80211_radiotap_iterator_init’:
src/radiotap.c:116:27: warning: taking address of packed member of ‘struct ieee80211_radiotap_header’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  116 |  iterator->_next_bitmap = &radiotap_header->it_present;
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc  -Wall -O2 -DWFB_VERSION='"21.3.13.28609-03c6e80f"' -std=gnu99 -c -o src/fec.o src/fec.c
g++  -Wall -O2 -DWFB_VERSION='"21.3.13.28609-03c6e80f"' -std=gnu++11 -c -o src/wifibroadcast.o src/wifibroadcast.cpp
g++ -o wfb_rx src/rx.o src/radiotap.o src/fec.o src/wifibroadcast.o  -lrt -lpcap -lsodium
g++  -Wall -O2 -DWFB_VERSION='"21.3.13.28609-03c6e80f"' -std=gnu++11 -c -o src/tx.o src/tx.cpp
g++ -o wfb_tx src/tx.o src/fec.o src/wifibroadcast.o  -lrt -lpcap -lsodium
cc  -Wall -O2 -DWFB_VERSION='"21.3.13.28609-03c6e80f"' -std=gnu99 -c -o src/keygen.o src/keygen.c
cc -o wfb_keygen src/keygen.o  -lrt -lpcap -lsodium
rm -rf deb_dist
./env/bin/python ./setup.py --command-packages=stdeb.command bdist_deb
running bdist_deb
running sdist_dsc
running egg_info
creating wifibroadcast.egg-info
writing wifibroadcast.egg-info/PKG-INFO
writing dependency_links to wifibroadcast.egg-info/dependency_links.txt
writing entry points to wifibroadcast.egg-info/entry_points.txt
writing top-level names to wifibroadcast.egg-info/top_level.txt
writing manifest file 'wifibroadcast.egg-info/SOURCES.txt'
reading manifest file 'wifibroadcast.egg-info/SOURCES.txt'
writing manifest file 'wifibroadcast.egg-info/SOURCES.txt'
running sdist
running check
creating wifibroadcast-21.3.13.28616
creating wifibroadcast-21.3.13.28616/scripts
creating wifibroadcast-21.3.13.28616/scripts/default
creating wifibroadcast-21.3.13.28616/telemetry
creating wifibroadcast-21.3.13.28616/telemetry/conf
creating wifibroadcast-21.3.13.28616/wifibroadcast.egg-info
copying files to wifibroadcast-21.3.13.28616...
copying README.md -> wifibroadcast-21.3.13.28616
copying setup.py -> wifibroadcast-21.3.13.28616
copying wfb_keygen -> wifibroadcast-21.3.13.28616
copying wfb_rx -> wifibroadcast-21.3.13.28616
copying wfb_tx -> wifibroadcast-21.3.13.28616
copying scripts/98-wifibroadcast.conf -> wifibroadcast-21.3.13.28616/scripts
copying scripts/storm32.service -> wifibroadcast-21.3.13.28616/scripts
copying scripts/wifibroadcast.service -> wifibroadcast-21.3.13.28616/scripts
copying scripts/wifibroadcast@.service -> wifibroadcast-21.3.13.28616/scripts
copying scripts/default/wifibroadcast -> wifibroadcast-21.3.13.28616/scripts/default
copying telemetry/__init__.py -> wifibroadcast-21.3.13.28616/telemetry
[... 38 lines cut ...]
  File "/usr/lib/python3.8/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/oracle/wifibroadcast/env/lib/python3.8/site-packages/stdeb/command/bdist_deb.py", line 34, in run
    self.run_command('sdist_dsc')
  File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/oracle/wifibroadcast/env/lib/python3.8/site-packages/stdeb/command/sdist_dsc.py", line 137, in run
    build_dsc(debinfo,
[... cut ...]
  File "/home/oracle/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 585, in dpkg_buildpackage
    process_command(args, cwd=cwd)
  File "/home/oracle/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 226, in process_command
    check_call(args, cwd=cwd)
  File "/home/oracle/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 59, in check_call
    raise CalledProcessError(retcode)
stdeb.util.CalledProcessError: 3
make: *** [Makefile:53: deb] Error 1
```

svpcom, 2021-03-13:
`sudo apt install python3-setuptools python3-all`

Abu-Abdullah, 2021-03-13:
many thanks, it seems i missed to install v3 of these libs. it works now. i have installed  `dh-python` as well to avoid one error

thank you @svpcom @lucasgrelaud

closed 2021-03-13 by Abu-Abdullah

Abu-Abdullah, 2021-03-14:
@svpcom 
another change i did is in file `/lib/systemd/system/wifibroadcast@.service` and change `/usr/bin/python' to `/usr/bin/python3'

i tried first to have an alias `alias python=python3` but it didn't work. not sure if it can be done in a better way

svpcom, 2021-03-14:
@Abu-Abdullah Try this fix

Abu-Abdullah, 2021-03-14:
working fine. thank you @svpcom